heishamon/HeishaMon #1: Further protocol

ariep, 2019-09-25:
Thanks for this project! I don't have an Aquarea yet, but am considering it, and one of the factors is the possibility to control it programmatically.

Do you happen to have done more reverse engineering of the protocol? Do you know what features the UART connection supports? I would be particularly interested in:
* controlling basic functions (on/off, heating/cooling mode);
* modifying the setpoint of the leaving temperature;
* reading various sensors (temperatures, flow, etc.).

BTW, your `README.md` is not really formatted as markdown, and so some of the newlines are lost in the github rendering. You might want to either use actual markdown syntax, or rename it to `README.txt`.

Egyras, 2019-10-01:
@ariep 
I have decoded basic functions like changing set point from -5 to +5 (setting fixed one isnt possible without powering off), changing heating/cooling,reading all possible sensor values.

I will upload my file later today, problem that I'am not a good code writer and dont have free time to develop futher. Maybe during winter will have more time.

ariep, 2019-10-01:
Nice! That sounds really useful.

> setting fixed one isnt possible without powering off

Does that mean that it's not possible to set an absolute setpoint via the serial interface at all? Or will it indeed change the setpoint given the right instruction, and restart the heat pump?

It's no problem for me that the protocol information isn't polished or put into proper code, anything you can provide is a great help!

Egyras, 2019-10-02:
@ariep all settings which needs powering off heat pump via button on controller could be changed via serial, because with Aquarea Cloud for engineers/installers these settings can be changed.  I just didnt had enough time to check other bits in protocol yet :)  It needs just a proper sequence - power off via controller, change desired setting, power on. So same steps could be done via serial commands.

IgorYbema, 2019-11-03:
I would like to help. Have some ESP8266 skills. Can you start a project in your github so we can have a go on it?

Egyras, 2019-11-03:
> I would like to help. Have some ESP8266 skills. Can you start a project in your github so we can have a go on it?

@IgorYbema I have ESP8266 skills, just need more skills with C++ to develop code. Will try to write a few lines this month.

maertsen, 2019-11-10:
I have started on a Prometheus.io exporter on top of the NodeMCU firmware for ESP8266, based on the information in this repo.

Egyras, 2019-11-11:
@maertsen did you tried to connect NodeMCU? I have tried NodeMCU and wemos d1 mini with small luck only. I believe signal level shifter is requered as Panasonic uses 5V Tx and Rx. As I get only one way communcation.

IgorYbema, 2019-11-11:
I'm still waiting for my aliexpress shipping for the connectors so didn't start yet. However can you share your wemos code already egyras? A levelshifting should not be that complex.

Egyras, 2019-11-11:
@IgorYbema I have only test sketch, will post when get my Wemos D1 mini working. It post data to mqtt, it would be better to format all as json as we have multiple values.  So it needs more re-work at the moment.

IgorYbema, 2019-11-11:
Every babystep is welcome :-)

IgorYbema, 2019-11-11:
What direction was working for you? From Wemos to Panasonic? Or the other way?
Wondering if the wemos is 5v tolerant or the panasonic is 3.3v tolerant.

Egyras, 2019-11-11:
@IgorYbema wemos and all othe esp8266 modules tolerates 3.3V signal level and Panasonic gives 5V signal, so signal shifter should be used according me. Tx accord me was working on wemos,  will check later today.

Egyras, 2019-11-11:
@IgorYbema @maertsen did you guys tried to connect wemos D1 or other ESP8266 controller ? I have uploaded my test sketch which made today, no luck even using logic converter from 5V to 3.3V with Wemos D1 mini. If I emulate sketch uploaded to wemos D1 on Laptop with IO Ninja  all works, with Panasonic no luck yet....
I use simple USB to TTL converter connected to Panasonic and can query with Laptop and get answers.

IgorYbema, 2019-11-11:
No connectors yet. Will take a while from aliexpress.

Egyras, 2019-11-12:
Ok, found my spare board https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/overview
and all works using logic level shifter 5V-3.3V on TX as RX on this board already tolerates 5V signal.
Testing sketch gives heat shift value :)

IgorYbema, 2019-11-12:
Great news to read that you had you first communication on a esp8266 board!
In the PDF of that board I read that only the RX (to the board) is level shifting from 5v. The TX on that board is 3.3v and that means that the panasonic is 3.3v tolerant if that works for you.
Anyway, I ordered an extra 4-chan bidirectional level shifter (https://nl.aliexpress.com/item/32309390020.html?spm=a2g0s.9042311.0.0.77894c4d7lrla8) which should help.

Egyras, 2019-11-12:
@IgorYbema I used level shifter on TX pin only, I believe that latest original Wemos D1 mini could have some issues with TX pin. Didnt compared newer ver. with older as other my Wemos boars are fake ones. I will try simple ESP8266 ESP-01 module when get it from friend. 
Next steps to include all most interesting values in the code, implement wifimanager library and maybe some "fancy" web page on ESP8266 for value lookup and some minimal control. If somone could "iron" code a little bit would be great help.  You can emulate heat pump work with such software as IO ninja or simillar - just send Panasonic answer to connected ESP module via USB to ttl adaper.

IgorYbema, 2019-11-12:
I can have a go with that. But first need to find out how to use IO ninja :-)

Egyras, 2019-11-14:
[image]
A little progress,  testing latest my code in Openhab. Need to doublecheck some decrypted values, water flow can be wrong.

IgorYbema, 2019-11-14:
Wow nice! I whish I had my connectors already. Playing with IO ninja is not my cup of tea (and don't have a license also). Could you update your latest code?

Egyras, 2019-11-14:
@IgorYbema IO ninja gives free trial and its easy to launch serial port with a few clicks :)  Latest code will be tonight.

IgorYbema, 2019-11-16:
Yesterday I focused on the HTML part (config wifi and mqtt settings instead of hard coded). That is almost done. I hope to add a status page also showing some basic values and log dumps.
I find working with IO ninja not that user friendly (or i'm doing something wrong). But then again the backend code you are doing is already going on fine.
On what kind of h-series are your working with? I'm not seeing tank status/on/off/temps in your protocol decodes so I assume you are not using an attached watertank?

Egyras, 2019-11-16:
@IgorYbema I have tank connected, its not all-in-one unit. For tank data currently I found current tank temperature and set temperature, there is other parameter like Delta which tells pump when to start heat tank. So for tank to start heating we will calulate manually set temperature - Delta :) And tank off heating will be set temperature which we already know. Also we know 3 way valve status, room or tank. Could to know that you are doing some code improvements. I will try to figure out other commands and some values from protocol.

IgorYbema, 2019-11-16:
But I am missing the the tank off and tank on command. How did you decode? Also using IO ninja?

Egyras, 2019-11-16:
@IgorYbema I will decode these commands, basically it will be heat pump modes like Auto, heat+dhw, dhw only and etc.  Currently have only heat pump mode responses. I use my PC with IO ninja in between heat pump and Panasonic Cloud module to capture commands. Issue command via cloud module and capture it.

maertsen, 2019-11-17:
> @maertsen did you tried to connect NodeMCU? I have tried NodeMCU and wemos d1 mini with small luck only. I believe signal level shifter is requered as Panasonic uses 5V Tx and Rx. As I get only one way communcation.

Like just reported in #4, I have wired up a wemos d1 mini using the NodeMCU firmware and a Lua implementation very similar to your Arduino code. I use a [4-channel I2C-safe Bi-directional Logic Level Converter - BSS138 by Adafruit](https://www.adafruit.com/product/757) on both TX and RX.

So far, no response though, so I'm still missing something. :-(

Egyras, 2019-11-17:
Mine original new wemos d1 doest work with logic level converter, tested.... I believe that resistors used on rx and tx line can be issue, what I found in forums "On the original wemos D1 mini pro boards they placed 470 ohm resistors on the TX and RX between the cp2104 and the esp8266. On de latest fake boards they replaced this resistors with 2 ohm resistors. An external serial signal won't be strong enough to drive the RX of the ESP because the cp2104 keeps de pin high. If you replace the resistors on the cp2104 TX line, an external serial signal will override the cp2104."

IgorYbema, 2019-11-17:
Can't you just use serial.swap to remap the serial pins to gpio15 and gpio13? The usb-uart isnt connect to these pins.

maertsen, 2019-11-17:
> Can't you just use serial.swap to remap the serial pins to gpio15 and gpio13? The usb-uart isnt connect to these pins.

Then I seem to run into https://forum.micropython.org/posting.php?mode=quote&f=16&p=20690&sid=090ca03eb97f96ca45de00a1fe79b89e
If these are really hardware problems I should just solve them on my end; they could well be unrelated to this effort. Thanks for thinking along though!

IgorYbema, 2019-11-18:
Can't read that post without login in. But then you can also use software serial libraries. Fast enough for the 9600 baud that is required.
Can't wait to receive my connectors so I can join on the hardware side :-)

IgorYbema, 2019-11-18:
GPIO15 is the TX (after swap) and because the swap is done after boot I don't think this would be of any issue.

Egyras, 2019-11-19:
@ariep @IgorYbema @maertsen Guys, what others commands we could have? I believe most useful and which can be used without powering off heat pump we have. I will try to search working hours and on/off counters values.

IgorYbema, 2019-11-19:
For now it looks ok. Ofcourse there a al lot of other settings which can be done in the service portal to change the internal settings of the heatpump but I don't think we want to focus on that. BTW, can you please merge the new pull request? In you commit this morning there where some merge conflicts which where not resolved.

Egyras, 2019-11-19:
@IgorYbema strange as in my visual studio code all looked fine after yesterdays merge....
All other internal settings can be changes only by powering off heat pump, dont know if it gives any benefits as you setup it once and never touch :)

Egyras, 2019-11-19:
@IgorYbema all merged now.

Egyras, 2019-11-19:
To do list:
1. Figure out Defrost, Operating time, Number of operations, Heat Mode Enegry Generation, Heat Mode Energy Consumption Values. Last two are calulated not very well so maybe not worth investing time in it.
2.  Web page based on websockets for Heat pump standalone controller.
3. Optimize code.

maertsen, 2019-11-19:
I'm primarily interested in the energy and COP stats. What makes you believe those numbers are not accurate?

Egyras, 2019-11-19:
@maertsen because energy values are calulated in 200w steps..., COP calulated from these values. I have compared Energy consumption values with real ones as I use dedicated meter and difference is a hudge.... In my 5Kw case it should be around +30% if I compare with real meter.  And for thermal energy calculation all depends on temperature sensor calibration used on unit. So these values doesnt give much sense for me.

konyee, 2019-11-19:
At first, thanks for this repo. I'm waiting for connector and cables to try it. I have a T-CAP 9kW split heatpump, and I want to integrate into Home Assistant with as many functions as possible.

IgorYbema, 2019-11-19:
> To do list:
>
> 1. Figure out Defrost, Operating time, Number of operations, Heat Mode Enegry Generation, Heat Mode Energy Consumption Values. Last two are calulated not very well so maybe not worth [...]

Let me focus on the websockets and some code optimization (we need to put things in .h files like those static byte strings).

Egyras, 2019-11-19:
@IgorYbema how optimal is to use "if else" conditions? I always try to avoid them in code if possible, how we send control commands when all  things will be in. h file? I mean how we format condition? Tomorrow will try to debug more usefull values from heat pump and if have more time test your merged code.

IgorYbema, 2019-11-19:
Using case (https://www.arduino.cc/reference/en/language/structure/control-structure/switchcase/) instead of the if statement is better. For example on the mode_state, quiet_mode_state and valve_state (valve, not walve :-). 

For the byte_commands which are static we can define them as constants in the .h files and just refer to them. Saves a lot of human unreadable code in the main .info file.

Just focus on the protocol decoding first is ok :-) that is the most important now I guess.

maertsen, 2019-11-19:
> [..] I have wired up a wemos d1 mini using the NodeMCU firmware and a Lua implementation very similar to your Arduino code.

I have just made [my lua implementation](https://github.com/maertsen/Aquarea-H-prometheus-exporter) available in a separate repo. I will try to contribute to protocol decoding here.

tsybolek, 2019-11-20:
Hi guys,
nice project! I also own a 12 kW aquarea H-Series and will try to connect and control it by NodeMCU. Will share my experience here, of course.
@maertsen : so have you resolved your problems with communicating between NodeMCU and HP as posted 3 days ago?

Egyras, 2019-11-20:
> Hi guys,
> nice project! I also own a 12 kW aquarea H-Series and will try to connect and control it by NodeMCU. Will share my experience here, of course.
> @maertsen : so have you resolved your problems with communicating between NodeMCU and HP as posted 3 days ago?

check closed issue, there is a solution

IgorYbema, 2019-11-25:
Hi,
Still waiting for my connectors. But in the mean time I was wondering if I could get into the CZ-TAW1 wifi adapter. Happen to find four unsoldered pins on the pcb which I thought could just be 3.3V TTL. After some testing with my multimeter I was pretty sure this was the case. Not a minute later I hooked on (or better, tried to stick on with my hand) a GND and a RX wire to my terminal and immediatly got a serial connection (on 115200 baud)! I was presented with the same hex dumps you discovered (as a serial debugging thing on the device). And next I rebooted the device and got a nice u-boot and linux (openwrt) booting!

[cz-taw1-console-dump-.zip](https://github.com/Egyras/Panasonic-H-Aquarea/files/3888975/cz-taw1-console-dump-.zip)
[image]

Didn't attach my TX yet so couldn't enter anything but I saw message like 'press enter to activate console' and 'press any key to enter u-boot'. So that looks promising! Maybe we can run some custom code on the CZ-TAW1 itself to spit out the messages not only to the aquarea-smart portal but to our own servers also :-)

Egyras, 2019-11-26:
@IgorYbema nice find :)   I saw these pads before but didnt had time to test as at the begining I was checking TCP packet which was sent from CZ-TAW1 to Azure IOT (I believe this endpoit was). But as it was encypted didnt had a chance to decrypt, so maybe we can figure out certificate keys which shoould be inside CZ-TAW1. Anyway I believe that Panasonic sends plain hex data and all decoding should be done on "server" side or maybe Iam wrong.

MiG-41, 2019-11-26:
I don't want to start new isue... In onnection details,Pin-outs there is in the brackets 250mV . I belive it shoul be 250**mA**

Egyras, 2019-11-26:
@MiG-41 good catch 👍

IgorYbema, 2019-11-26:
Just now I managed to solder some cables to the connectors. After booting I now could interact with the console. Just pressing enter allowed me to have access, without any password!

Created a filesystem dump. Still have to check if there are no privacy components in it before I post it here :-)

I already searched for certificates and there is indeed a client certificate on it for the device to authenticate to the webserver. Ofcourse server side certificates are also used for encryption.

The main process (called a2wmain) is a binary which spits out all those hexdumps. If killed it will initiate a kernel oops and the watchdog will reset the device. Further, I didn't find any other process running which could be used to gain access remotely. Dropbear/ssh/telnet isn't installed. HTTP isn't running and can't be started. As it seems to use basic openwrt atheros binaries probably those can be installed, didn't try yet.

Also the u-boot isn't password protected. At first glance it seems to have a backup tftp config in it so the device can start from a tftp server if the internal flash is corrupt. Have to dig into this more.

Egyras, 2019-11-26:
@IgorYbema could you share certificate files with me, I would like to check my tcp dumps. It would be clear what playload is sent,  just hex dumps or some nice json.

IgorYbema, 2019-11-26:
Don't think you could decrypt your tcpdump's with my certificates.
But if you see my dump from yesterday you see hexdumps with messages like "PCF" or something like that (instead of CN-CNT". Those are the messages sent and received from there IOT server (https://ldcws.pcpf.panasonic.com). So no json there also.

Egyras, 2019-11-26:
@IgorYbema seems my theory was good, it sends RAW data to IOT server. In such way device can be more universal and used for newer products.

Egyras, 2019-11-27:
Will decode more commands as I need to know some values which depends on other controls.  Seems we wont get full protocol in easy way. Also I dont have optional PCB connected, so these values wont be possible to decode for me.

MiG-41, 2019-11-29:
Do you think it is possible to cut A2WMAIN aplication and pass it to another router ?
Or perhaps on that CZ-TAW1 run a MQTT client or even a server ?

IgorYbema, 2019-11-29:
a2wmain just reads from /dev/ttyUSB0 (that is where the the heatpump is connected on) and it should be relative simpel to put something between that. Running mqtt client or server should be possible also because these are just regular openwrt ar71xx packages.
But then again, this project is to replace the whole CZ-TAW1 so you don't need it anymore. Still looking if running custom code on the CZ-TAW1 is to any use.

Egyras, 2019-11-29:
> Do you think it is possible to cut A2WMAIN aplication and pass it to another router ?
> Or perhaps on that CZ-TAW1 run a MQTT client or even a server ?

Its better get rid of CZ-TAW1 :)

IgorYbema, 2019-11-29:
@Egyras got my aliexpress connectors today. Only to find out I bought PH connectors instead of PA ...
Just order new ones at the sites you mentioned. A few days later I hope to connect finally :-)

Egyras, 2019-11-29:
@IgorYbema  where are you located, I can send my spare one as I have 10 of them :)

IgorYbema, 2019-11-29:
Already ordered 10 my self 😀 happy to help others in NL with my spares later on

[196 more replies]