heishamon/HeishaMon #4: Does the magic query string require prior initilialization?

maertsen, 2019-11-17:
Hi @Egyras,

I have an implementation very similar to yours, but in Lua on the nodemcu firmware. It's based on your specs in the README. My setup is very similar, though I use a level shifter on both TX and RX directions.

Somehow, my heatpump does not react to the query string at all. Does the cloud unit send any initialization commands prior to the query that might be required for the query to trigger a response?

Egyras, 2019-11-17:
@maertsen heat pump should respond to query, no any additional commands prior it required. Did you tried to swap rx with tx? I have nodemcu, but didnt tried yet. Also you can check if query which you send is correct, just connect it to your PC and lauch software to read serial data.

maertsen, 2019-11-17:
> @maertsen heat pump should respond to query, no any additional commands prior it required. Did you tried to swap rx with tx? I have nodemcu, but didnt tried yet. 

I've just swapped rx and tx, which yields the same result, so that's not it.

> Also you can check if query which you send is correct, just connect it to your PC and lauch software to read serial data.

Yeah, I can confirm my code sends the query correctly. I have tested it with a USB TTL dongle.

Egyras, 2019-11-17:
https://github.com/Egyras/Panasonic-H-Aquarea/issues/1#issuecomment-554787956

Egyras, 2019-11-17:
Will check my both nodemcu and wemos boards tomorrow, we need to make them work :) Soldering can be required, but I have skills.

Egyras, 2019-11-17:
@maertsen one more idea, just try simple voltage divider from 220 ohms and 470 on RX pin and conect it directly to heat pumps TX.

maertsen, 2019-11-17:
> [#1 (comment)](https://github.com/Egyras/Panasonic-H-Aquarea/issues/1#issuecomment-554787956)

This may very well be my problem; I have the 150mA regulator discussed there. I'll try to work around it later this week and report back. Thanks for thinking along!

Egyras, 2019-11-18:
@maertsen please check my voltage divider, I dont have such value resistors at home so will need to visit a shop :)
[image]

maertsen, 2019-11-19:
> @maertsen please check my voltage divider

I've just replicate it using 2K and 3.3K ohm resistors. No luck. Was there any particular reason you suggested using smaller ohm values?

Egyras, 2019-11-19:
> @maertsen please check my voltage divider
> 
> I've just replicate it using 2K and 3.3K ohm resistors. No luck. Was there any particular reason you suggested using smaller ohm values?

Because it will have stronger pull-up ;)  Try my suggested values.

maertsen, 2019-11-19:
> Try my suggested values.

Thanks, that was it, I'm getting a response now. I'll close this ticket.

closed 2019-11-19 by maertsen