heishamon/HeishaMon #13: Decode heat pump model

Egyras, 2019-12-11:
Please post your answers from heat pump sending magic packet to panasonic, write model as well.

MiG-41, 2019-12-11:
I can put part of dump file from my CZ-TAW1:
[A2W CN-CNT request       12:11:13](len:0x6F)
71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 12
[A2W CN-CNT reply         12:11:14](len:0xCB)
71 C8 01 10 56 95 62 49 00 55 00 00 00 00 00 00 00 00 00 00 19 15 11 55 16 9E 55 05 09 00 00 00
00 00 00 00 00 00 7F 8A 80 8A AD 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 3D 8A 85 85
D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A1 99 71 8A B7 A3 7B 8F 8F 87 8F 8A 8A 94 9E 8A 8A 94 9E 85 8F
8A 15 3D 7B B7 3D 00 00 00 00 00 00 00 00 55 F5 55 21 A2 15 59 05 19 12 69 00 00 00 00 00 00 00
00 E2 CF 0D 86 05 12 D0 0C 95 05 97 00 AE 85 98 97 32 32 9A AD 32 32 32 7F 9A 98 9F 9A 98 84 89
8F 90 8A 2B 01 01 01 00 00 8B 14 1F 51 01 01 79 79 01 01 C8 05 00 BC 08 00 03 00 00 0D 00 00 0D
02 01 01 01 01 01 01 00 00 00 10
[A2W Device Info          12:11:14](len:0x65)
24 F0 01 01 F1 01 01 F2 01 01 F3 01 01 F4 01 01 C0 01 01 C1 01 02 C2 01 02 E6 01 02 E7 01 01 F9
01 02 C3 01 02 C4 01 01 D0 00 D1 00 D2 00 D3 00 8F 00 80 01 02 E0 01 02 E1 01 02 E3 01 01 E4 01
01 C5 01 01 C6 01 51 FE 01 01 E2 01 01 62 01 7F 63 01 8A 64 01 80 65 01 8A 60 01 02 61 01 01 00
00 71 00 89 00

I don't know from where this Device Info comes... But hope that Answer is readable.
This is from my 3-phase -T-CAP 12kW KIT-WXC12H9E8 ,IDU:WH-SXC12H9E8  ODU:WH-UX12HE8

konyee, 2019-12-26:
71 c8 01 10 56 55 52 49 00 55 00 00 00 00 00 00 00 00 00 00 19 15 11 55 16 a5 55 05 06 00 00 00 00 00 00 00 00 00 ad 80 a3 80 b1 71 71 61 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8a 85 85 d0 7b 78 1f 7e 1f 1f 79 79 8d 8d ad a3 7b 8f a3 a3 7b 8f 98 85 80 8f 8a 94 9e 8f 8a 94 9e 85 8f 8a 11 3d 78 c1 0b 00 00 00 00 00 00 00 00 55 75 55 21 73 15 59 05 14 12 65 00 00 00 00 00 00 00 00 e2 cf 0d 77 09 12 d0 0b 05 11 af 00 aa 88 aa af 32 32 ad a3 32 32 32 ad ad af b7 9b b0 82 88 83 99 96 91 01 2e 22 00 00 d2 13 2c 61 31 35 4c 01 01 01 93 04 00 7b 02 00 01 00 00 01 00 00 0a 02 0a 1f 01 01 01 01 00 00 00 95

This is my reply for "info" packet. My device is a single phase T-CAP 9kW H series

Egyras, 2019-12-27:
Who has 7 Kw unit ? :)

MiG-41, 2019-12-27:
I.E. KIT-WXC09H3E5 is a singlephase T-CAP... But let's wait for @konyee

IgorYbema, 2019-12-28:
KIT-AXC09HE5  should be the 1-phase 9kw tcap

konyee, 2019-12-29:
Outdoor: WH-UX09HE5
Indoor: WH-SXC09H3E5

Egyras, 2019-12-30:
I believe that found suspected bytes for model but still need 7 kw unit to verify....

Egyras, 2020-01-13:
7kw monoblock
Received 203 bytes data
data: 71 C8 01 10 56 55 52 61 00 55 00 00 00 00 00 00 
data: 00 00 00 00 19 15 11 55 15 9D 55 05 09 00 00 00 
data: 00 00 00 00 00 00 80 8A 80 8A B4 71 71 97 99 00 
data: 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A4 A0 74 83 B7 
data: A3 7B 8F 91 85 80 8A 8A 94 9E 8A 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 95 55 
data: 55 21 78 15 59 05 13 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 14 33 42 D1 0B 17 33 A1 00 00 86 9D 
data: A1 32 32 A0 AC 32 32 32 80 A0 A1 B0 93 9F 81 61 
data: 83 61 95 66 01 0D 16 00 00 29 0A 2F 6B 20 01 30 
data: 79 01 01 50 01 00 5D 02 00 02 00 00 01 00 00 08 
data: 01 03 0D 01 01 01 01 00 00 00 57 
Checksum received ok!

gregor-samosir, 2020-01-13:
Pana Monoblock 5H

Received 203 bytes data
data: 71 C8 01 10 56 55 52 4C 00 56 00 00 00 00 00 00 
data: 00 00 00 00 2A 15 11 99 15 95 55 0A 0A 00 00 00
data: 00 00 00 00 00 00 9F 94 A3 8A C1 80 71 91 99 00
data: 00 00 00 00 00 00 00 00 00 00 80 80 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 65 79 65 5B 9F A3 7B 8F A3 
data: A3 7B 8F 9A 81 82 94 94 99 9E 8A 8A 94 9E 81 94
data: 8F 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 65 55 
data: 55 21 73 15 59 05 22 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 9C 00 00 86 99 
data: 9C 9B 32 9F A3 32 32 32 9F 9F 9C B2 90 99 80 61 
data: 81 61 8F 61 01 15 44 00 00 64 12 4D 91 36 01 44 
data: 79 01 01 2C 02 00 32 0B 00 1E 00 00 01 00 00 06 
data: 01 05 17 01 01 01 01 00 00 00 20 
Checksum received ok!

Egyras, 2020-01-14:
seems  191 byte represents unit power

gregor-samosir, 2020-01-14:
`External PCB, Heisha 5kW

PCB, Testing SG-Ready

0,0 = Normal
1,0  = Heatpump + Heater Off
0,1  = Config 1 on PCB
1,1   = Config 2 on PCB

State 1,0 Heatpump + Heater Off

Received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 56 00 00 00 00 00 00 
data: 00 00 00 00 2A 15 11 99 15 95 55 0A 0A 00 00 00 
data: 00 00 00 00 00 00 9E 94 A0 94 C1 80 71 91 99 00 
data: 00 00 00 00 00 00 00 00 00 00 80 80 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 65 79 65 5B 9E A3 7B 8F A0 
data: A3 7B 8F 9A 81 82 94 94 99 9E 94 8A 94 9E 81 94 
data: 8F 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 
data: 55 21 73 15 59 05 0C 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 97 00 00 8A 96 
data: 97 98 32 9E A0 32 32 32 9E 9E 97 88 90 90 87 61
data: 89 61 8C 2F 01 01 01 00 00 22 00 01 01 01 01 79 
data: 79 01 01 2D 02 00 37 0B 00 1E 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 48 
Checksum received ok!

State 1,1 Config 2

Received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 56 00 00 00 00 00 00 
data: 00 00 00 00 2A 15 11 99 15 95 55 0A 0A 00 00 00 
data: 00 00 00 00 00 00 9E 94 A0 94 C1 80 71 91 99 00 
data: 00 00 00 00 00 00 00 00 00 00 80 80 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 65 79 65 5B 9E A3 7B 8F A0 
data: A3 7B 8F 9A 81 82 94 94 99 9E 94 8A 94 9E 81 94 
data: 8F 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 
data: 55 21 73 15 59 05 0A 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 97 00 00 8A 96 
data: 97 98 32 9B 9C 32 32 32 9E 9B 97 88 90 90 86 61 
data: 89 61 8C 2F 01 01 01 00 00 4B 07 43 7F 01 01 79 
data: 79 01 01 2D 02 00 37 0B 00 1E 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 65 
Checksum received ok!

State 0,1 Config 1

Received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 56 00 00 00 00 00 00 
data: 00 00 00 00 2A 15 11 99 15 95 55 0A 0A 00 00 00
data: 00 00 00 00 00 00 9E 94 A0 94 C1 80 71 91 99 00
data: 00 00 00 00 00 00 00 00 00 00 80 80 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 65 79 65 5B 9E A3 7B 8F A0 
data: A3 7B 8F 9A 81 82 94 94 99 9E 94 8A 94 9E 81 94 
data: 8F 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 
data: 55 21 73 15 59 05 14 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 97 00 00 89 97 
data: 97 95 32 A4 A6 32 32 32 9E A4 94 88 90 90 86 61 
data: 89 61 8C 2F 01 01 01 00 00 39 06 42 7D 01 01 79 
data: 79 01 01 2D 02 00 37 0B 00 1E 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 5B 
Checksum received ok!

State 0,0 Normal

Received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 56 00 00 00 00 00 00 
data: 00 00 00 00 2A 15 11 99 15 95 55 0A 0A 00 00 00 
data: 00 00 00 00 00 00 9E 94 A0 94 C1 80 71 91 99 00
data: 00 00 00 00 00 00 00 00 00 00 80 80 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 65 79 65 5B 9E A3 7B 8F A0 
data: A3 7B 8F 9A 81 82 94 94 99 9E 94 8A 94 9E 81 94 
data: 8F 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 
data: 55 21 73 15 59 05 0B 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 98 00 00 89 98
data: 98 96 32 9E A0 32 32 32 9E 9E 98 88 90 91 87 61 
data: 89 61 8C 2F 01 01 01 00 00 4D 0C 38 72 01 01 79 
data: 79 01 01 2D 02 00 37 0B 00 1E 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 67 

`

AppieM, 2020-01-18:
I have a 5Kw Monblock WH-MDC05H3E5
Only used for heating. No DHW Buffer or extra pump. Floor-heating downstairs and radiators on bedrooms. Heating according to heating curve.

Received 203 bytes data
data: 71 C8 01 10 56 55 52 61 00 55 00 00 00 00 00 00 
data: 00 00 00 00 19 15 11 55 15 9D 55 05 09 00 00 00 
data: 00 00 00 00 00 00 80 94 80 8A B4 79 71 97 99 00 
data: 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A2 97 7B 8D B7 
data: A3 7B 8F 99 85 80 94 8A 94 9E 8A 8A 94 9E 82 97 
data: 92 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 95 55 
data: 55 21 73 15 59 05 21 12 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 9D 00 00 82 9C 
data: 9D 32 32 9E AF 32 32 32 80 9E 9F AD 97 9C 7A 61 
data: 7A 61 8A 62 01 0C 27 00 00 CD 09 1F 53 1F 01 23 
data: 79 01 01 5C 03 00 E1 0C 00 38 00 00 01 00 00 06 
data: 01 03 07 01 01 01 01 00 00 00 12 
Checksum received ok!

mi-hol, 2024-09-06:
@Egyras should this be closed in favor of a new issue to "add a new type of heat pump" (like in #489)?

I'd be happy to provide an issue template for it

mi-hol, 2024-09-06:
same topic in https://github.com/Egyras/HeishaMon/issues/274

mi-hol, 2024-09-12:
> favor of a new issue to "add a new type of heat pump" (like in https://github.com/Egyras/HeishaMon/issues/489)? I'd be happy to provide an issue template for this purpose

@Egyras thanks for closing, I'd appreciate a comment regarding this suggestion