heishamon/HeishaMon #14: Adding extra features

IgorYbema, 2019-12-14:
I currently already have a wemos connected to my panasonic for other functions.
Just leaving this issue here so we can discuss if we want these functions also in the main release as extra options.
What I am using is:
- a wemos relay shield controlled from a web-url (could be a mqtt topic also) which connects to the 'remote thermostat' input of the panasonic aquarea. I use a thermostat of my choosing to control this relay over wifi. I am not impressed by the roomthermostat of the aquarea itself so I am using this.
- a 1-wire sensor network which measures all the piping around the panasonic and publishes it to my influxdb

The main project could have this as extra functions. I think the esp8266 is fast enough to do the main work and these extra options all together.

Egyras, 2019-12-14:
@IgorYbema I also controlled panasonic with wemos and relay with some resistors to imitate thermistor.  Now I do with current code and if have remote controller thermistor value 26C (not calibrated) openhab2 sends - 5 for heat curve shift. When 25C sends 0. But we could make some lightweight webpage on wemos and make some logic for control there. Also serial read we must make not blocking with some smart logic that during read command we wont have write command.

IgorYbema, 2019-12-21:
Another great idea I just got from someone else is to add pulse sensor support to read pulses from one or two kWh meters connected to the heat pump. I already have that code for electric kWh meters.
This all will make this project a complete 'monitoring and controlling' platform for the heatpump environemt.

Egyras, 2019-12-21:
@IgorYbema  we could add some ds18b20 temperature sensors which cloud help to calibrate existing Panasonic ones for outlet and inlet.

MiG-41, 2019-12-23:
I still thinking ,that CZ-TAW1 should be present if possible ,that is why this device should be in the middle ,also to control what kind of informations Panasonic webpage receives. So on the board additional conector for CZ-TAW1.

Egyras, 2019-12-23:
@MiG-41 no need for CZ-TAW1 if we will have own webpage with all info :)

IgorYbema, 2019-12-23:
Yes the main goal is to forget the panasonic site. We don't need a company like panasonic to have our big data.

IgorYbema, 2019-12-23:
Nice work @Egyras if you can share your mqtt to grafana stuff it will spare me some time to do the same :)

Egyras, 2019-12-23:
@IgorYbema I'am using Openhab2 default persistance service for influxdb, so its "out of box"  I believe your home automation solution should have something similar.

IgorYbema, 2019-12-23:
Ah ok :-) domoticz will do the same indeed.

MiG-41, 2019-12-24:
Ok , another feature : maintenance of errors... Like sending email/sms with code error , additional GPIO with will close ( to send a signal to external alarm ect).

Egyras, 2019-12-24:
@MiG-41 then we need to figure out error bytes :)

MiG-41, 2019-12-24:
I already have some "genereted " by me like:
H91 - just remove the shortcut or thermic protection ,and probably swith the external heater.
H76 - remove the connection beetween the steering panel and mainboard
F45 - remove the CN-TH1 connector from mainboard ( these are NTC resistors for input output and heat exchanger).

These are examples of easy to make errors .

MiG-41, 2019-12-27:
Next feature: When the defrost time comes switch to DHW to make it in Deice mode 2
Conditions : inlet temperature less then 27C , outdoor temperature higher then -10C , tank temperature higher then .. 35C ?

Egyras, 2019-12-27:
> Next feature: When the defrost time comes switch to DHW to make it in Deice mode 2
> Conditions : inlet temperature less then 27C , outdoor temperature higher then -10C , tank temperature higher then .. 35C ?

@MiG-41 this is good one, but we should calculate possible profit by doing this. As to prepare DHW you will have lower COP then heating underfloor after Deice. Other problems by doing this:
1. Its hard to catch exact defrost time. We know when defrost is active. Anyway we have some decoded pipe temperatures which are responsible for deice start, possible to figure out exact start conditions.
2. After switchower to DHW and after deice when we switch back to Room side its possible that Heating will be stopped. Because it will meet compressor off conditions.

MiG-41, 2019-12-27:
The couse is that i have limited time ( from 22:00 till 6:00) when the power is really cheap for me ,so every minute count in that period :)
Ad.1. Acc service manual there is a curve showing Outdoor pipe line temperature related to outdoor temperature. When Outdoor temperature reach the curve in 3min defrost apear . ( It have to be also more then 35 min from last defrost).
It can be also situation that we predict the defrost a little bit earlier , set it to DHW and make a force defrost ( this CZ-TAW1 can , and maybe you also already decoded that function ,hav't check).

Ad.2 For this i don't have answer , we have to see that during tests.

Egyras, 2019-12-27:
@MiG-41  suggest you to simulate this and just test 2. possible problem. Because I have seen such situations when after switchower from DHW to Room side it doesnt start to heat instantly. 
For defrost COP you are wrong, its high by my calulations as you use water from underfloor.

MiG-41, 2019-12-27:
No , in Deice Mode 2 water pump stops , no exchange of heating is in inner unit. Heat is generated only by compressor.
In Deice mode 1 heat is going from underfloor ( but for this mode inlet temperature should be above 27C ,and unfortunatly not in my house ,im reaching only 23 , max 24C temp of water ).

Egyras, 2019-12-27:
@MiG-41  just forgot that T-Cap has 2 modes for de-ice :)  In simple unit it has only Deice mode 1 by your description and if inlet is below 27C it uses build-in heater (this can be switched off in service menu as I did in my unit). 
For your case maybe it's better to "simulate" inlet temperature (same for outlet to avoid compressor off ) before de-ice instead switching to DHW and back.  Simulation can be done with relay and resistors.

MiG-41, 2019-12-27:
Swithing to DHW during Deice gives me real high temperature for faster defrost.
Playing with resistors is creates more risk since you never know whit kind of misturious depends are in the Panasonic code .In inner unit there are 4 temperature sensors ,perhaps you need to rise them all. Then set temperature ( im' using the temperature curve , not outer thermostat).
On polish forum one user complained ,that in Aquarea F he tried to influence on DHW sensor by relay and resistors , and mainboard sometimes detect it as an error.

gregor-samosir, 2020-01-02:
Feature request:
- setup for an unit ID in the mqtt topic like pana_headpump_%variable%/...
to monitor multible panas in one mqtt server.

tsybolek, 2020-01-20:
Hi guys,
can you tell me how offen can you send the request and read out the data from aquarea without getting many errors - I mean are you able to read out once per second? Or even faster? I was able to do it once per second, but now I have to insert delays, at least 5 sec, otherwise I geht many errors (seither zeros or garbage).

tsybolek, 2020-01-20:
And I don't know what the reason is.

Egyras, 2020-01-20:
@tsybolek  serial is not best option for fast data processing, even CZ-TAW1 makes reads with ~3-4 second delays.

JWGracht, 2021-03-10:
Question to the contributers. How much programming space is still left in the current design? I have some wild ideas te make the heishamon a little smarter and actually take control of some of the heat pump aspects. For example I want to run the heating curve based on average outside temperature of the last day. So a more advance heating curve setting. Currently I control this through Domoticz, but I like it better when Heishamon can operate stand alone and has a few of these options available in the web interface. Could that be feasable and should I try if I can add these kind of features, or is it a no go from the start?

Egyras, 2021-03-10:
@JWGracht  you can try to do this. Just interesting how you do heating curve modification, as such one needs heat pump off and you will interupt heating. If its just curve shifting by -5...+5, thene this can be done without turning off.

JWGracht, 2021-03-10:
@Egyras Thanks for the quick response. Ok I am going to start with it. I have limited amount of time, so maybe it could take some while before I have something working. A prerequirement would be then that the heatpump is set to direct heating. (So make the heatpump dumb) And then you can actually control the full temperature using the Heishamon.

MiG-41, 2021-03-11:
@Egyras @JWGracht now with command SetCurves it is possible to modify  curves on the fly , without heat pump off /stop ( so without interupting heating). Tested and works fine.

JWGracht, 2021-03-11:
@MiG-41 Good to know about the SetCurves. I was not aware of that feature. But I also want to investigate if it is possible to take over the default Panasonic firmware abilities, and actually extent with more advance water temperature control in the HeishaMon. I think that would only be possible when the heatpump is in direct heating, so to be sure the heatpump is not doing things at its own.

MiG-41, 2021-03-12:
@JWGracht ,as i mentioned , in thermostat mode and direct heating SetCurve is also only one way to change target temperature ( besides stop HP and change it on controller).

PhoenixNP, 2023-02-26:
> @MiG-41 no need for CZ-TAW1 if we will have own webpage with all info :) At the moment all data goes to influxdb and viewed in Grafana. And for me its better the Panasonic Cloud.
> 
> [image]

Could you perhaps share your grafana json file? =)

Egyras, 2023-02-26:
> > @MiG-41 no need for CZ-TAW1 if we will have own webpage with all info :) At the moment all data goes to influxdb and viewed in Grafana. And for me its better the Panasonic Cloud.
> > [image]
> 
> Could you perhaps share your grafana json file? =)

here you go :)
 `{
  "annotations": {
    "list": [
      {
        "$$hashKey": "object:136",
        "builtIn": 1,
        "datasource": {
          "type": "datasource",
          "uid": "grafana"
        },
        "enable": true,
        "hide": true,
        "iconColor": "rgba(0, 211, 255, 1)",
        "name": "Annotations & Alerts",
        "target": {
          "limit": 100,
          "matchAny": false,
          "tags": [],
          "type": "dashboard"
        },
        "type": "dashboard"
      }
    ]
  },
  "editable": true,
  "fiscalYearStartMonth": 0,
  "graphTooltip": 0,
  "id": 14,
  "links": [],
  "liveNow": false,
  "panels": [
    {
      "datasource": {
        "type": "influxdb",
        "uid": "000000001"
      },
      "fieldConfig": {
        "defaults": {
          "color": {
            "mode": "thresholds"
          },
          "mappings": [
            {
              "options": {
                "match": "null",
                "result": {
                  "text": "N/A"
                }
              },
              "type": "special"
            }
          ],
          "thresholds": {
            "mode": "absolute",
            "steps": [
              {
                "color": "green",
                "value": null
              },
              {
                "color": "red",
                "value": 80
              }
            ]
[... 3125 lines cut ...]
  },
  "time": {
    "from": "now-6h",
    "to": "now"
  },
  "timepicker": {
    "refresh_intervals": [
      "5s",
      "10s",
      "30s",
      "1m",
      "5m",
      "15m",
      "30m",
      "1h",
      "2h",
      "1d"
    ]
  },
  "timezone": "",
  "title": "Panasonic Aquarea",
  "uid": "kmDMdZWRk",
  "version": 148,
  "weekStart": ""
}`

Egyras, 2023-02-26:
> > @MiG-41 no need for CZ-TAW1 if we will have own webpage with all info :) At the moment all data goes to influxdb and viewed in Grafana. And for me its better the Panasonic Cloud.
> > [image]
> 
> Could you perhaps share your grafana json file? =)

here you go: https://github.com/Egyras/HeishaMon/blob/master/Panasonic%20Aquarea-1677423314424.json

PhoenixNP, 2023-02-26:
Awesome, thanks! =)

salakrzy, 2023-06-20:
Hi,
The **@kamaradclimber** integration uses this same  MQTT topics name so it shoud work.

If there are any differences the MQTT topic name you can change in the topic.cpp file,

#include "Topics.h"

// Can be configured. The mqtt topics will be concatenated with these values
const std::string mqttPrefix = "**panasonic_heat_pump**";
const std::string s = "**/**";
const std::string setTopicPrefix = mqttPrefix + s + "**commands**";
const std::string infTopicPrefix = mqttPrefix + s + "**info**";
const std::string Topics::STATE = mqttPrefix + s + "**main**";
const std::string Topics::LOG = infTopicPrefix + s + "**log**";
const std::string Topics::WILL = infTopicPrefix + s + "**LWT**";

// set topics
const std::string Topics::SET1 = setTopicPrefix + s + "SetHeatpump";
const std::string Topics::SET2 = setTopicPrefix + s + "SetHolidayMode";
const std::string Topics::SET3 = setTopicPrefix + s + "SetQuietMode";

...

// state topics
// Can be configured.
const char *States::TOP0 = "Heatpump_State";
const char *States::TOP1 = "Pump_Flow";
const char *States::TOP2 = "Force_DHW_State";

...

closed 2024-09-12 by Egyras