heishamon/HeishaMon #19: CZ-TAW1 hacking

IgorYbema, 2019-12-23:
It first downloads (after checking that there is an update) a download_file.tgz into /tmp. 
Then it starts /usr/bin/dlupd.sh which will unpack this file, resulting in two images (one for the kernel, one for the rootfs). Then this script starts another script /usr/sbin/fwupdate which will copy these images to the respective MTD partitions (it even does a swap, it always contains an older and newer version).
Blocking it is by editting dlupd.sh, on the end (in the case start) comment out the start_action and add a endless while loop in front:
```
        start)
                while true ; do sleep 10 ; done
#               start_action
        ;;
```

This while loop is necessary because if this scripts finishes the main program a2wmain will restart the unit anyways (resulting in a constant reload after each succesful download).

@Egyras because it is fun :-)

_Originally posted by @IgorYbema in https://github.com/Egyras/Panasonic-H-Aquarea/issues/1#issuecomment-568390393_

IgorYbema, 2019-12-23:
See this comment how to get into the CZ-TAW1: https://github.com/Egyras/Panasonic-H-Aquarea/issues/1#issuecomment-558364184

Egyras, 2020-01-25:
root@OpenWrt:~# strace -t -e open ls /usr/bin/a2wmain
12:51:26 open("/lib/libcrypt.so.0", O_RDONLY) = 3
12:51:26 open("/lib/libm.so.0", O_RDONLY) = 3
12:51:26 open("/lib/libgcc_s.so.1", O_RDONLY) = 3
12:51:26 open("/lib/libc.so.0", O_RDONLY) = 3
12:51:26 open("/lib/libc.so.0", O_RDONLY) = 3
12:51:26 open("/lib/libc.so.0", O_RDONLY) = 3
12:51:26 open("/lib/libc.so.0", O_RDONLY) = 3

Egyras, 2020-01-26:
COMMAND PID USER   FD   TYPE     DEVICE SIZE/OFF NODE NAME
a2wmain 871 root  cwd    DIR       0,13        0  159 /
a2wmain 871 root  rtd    DIR       0,13        0  159 /
a2wmain 871 root  txt    REG       31,2   137320  306 /usr/bin/a2wmain
a2wmain 871 root  mem    REG       31,2    55848  381 /usr/lib/libz.so.1.2.8
a2wmain 871 root  mem    REG       31,2    12504 1146 /lib/libdl-0.9.33.2.so
a2wmain 871 root  mem    REG       31,2   359583 1140 /lib/libuClibc-0.9.33.2.so
a2wmain 871 root  mem    REG       31,2    78728 1145 /lib/libgcc_s.so.1
a2wmain 871 root  mem    REG       31,2  1398268  880 /usr/lib/libcrypto.so.1.0.0
a2wmain 871 root  mem    REG       31,2   320268  882 /usr/lib/libssl.so.1.0.0
a2wmain 871 root  mem    REG       31,2    78652 1108 /lib/libpthread-0.9.33.2.so
a2wmain 871 root  mem    REG       31,2    28968 1141 /lib/ld-uClibc-0.9.33.2.so
a2wmain 871 root    0u   CHR        5,1      0t0   81 /console
a2wmain 871 root    1w   CHR       4,64      0t0  243 /dev/ttyS0
a2wmain 871 root    2w  FIFO        0,6      0t0 1325 pipe
a2wmain 871 root    3u   CHR       50,0      0t0  934 /dev/gpio-led
a2wmain 871 root    4u   CHR       50,0      0t0  934 /dev/gpio-led
a2wmain 871 root    5u   CHR      188,0      0t0  834 /dev/ttyUSB0
a2wmain 871 root    6u   REG       0,10        0 5432 /tmp/ldcc_once_check
a2wmain 871 root    7u  unix 0x82dbf5a0      0t0 5433 socket
a2wmain 871 root    8u  unix 0x82dbfd20      0t0 5434 socket
a2wmain 871 root    9r  FIFO        0,6      0t0 1118 pipe
a2wmain 871 root   10r  FIFO        0,6      0t0 1181 pipe
a2wmain 871 root   11w  FIFO        0,6      0t0 1325 pipe
a2wmain 871 root   12u  sock        0,4      0t0 5535 can't identify protocol
a2wmain 871 root   14u  inet       5572      0t0  TCP D8-AF-F1-80-74-F6:52301->168.63.205.102:https (ESTABLISHED)

MiG-41, 2020-02-07:
To have ssh:
opkg install http://archive.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base/dropbear_2014.63-2_ar71xx.ipk
/etc/init.d/dropbear start
/etc/init.d/dropbear enable
passwd
/etc/init.d/telnet disable

MiG-41, 2020-05-18:
Side1:
setenv bootcmd bootm 0x9fD10000
 
setenv bootargs board=CUS531 console=ttyS0,115200 mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env),6464k(rootfs),6464k(rootfs2),64k(apl),64k(config),1472k(kernel1),1472k(kernel2),64k(art) rootfstype=squashfs,jffs2 noinitrd

Side2:
setenv bootcmd=bootm 0x9fE80000

setenv bootargs board=CUS531 console=ttyS0,115200 mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env),6464k(rootfs1),6464k(rootfs),64k(apl),64k(config),1472k(kernel1),1472k(kernel2),64k(art) rootfstype=squashfs,jffs2 noinitrd

closed 2024-09-12 by Egyras