heishamon/HeishaMon #60: hostname on mqtt connect status "default_hostnameupdate_password"

gregor-samosir, 2020-01-08:
HeishaMon version 8.1.2019
Build with VSCode
Arduino.json

`
{
    "board": "esp8266:esp8266:d1_mini",
    "configuration": "xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200",
    "port": "/dev/cu.wchusbserial1a1230",
    "sketch": "HeishaMon/HeishaMon.ino",
    "output": "../build",
    "programmer": "AVRISP mkII"
}
`

The hostname resolving switch on used hardware. On an Wemos D1 it comes up with "default_hostnameupdate_password".

Testet the same bin an an spare modmcu, the hostname comes up as I expect.

gregor-samosir, 2020-01-08:
Test with release 8.1.2020 20:00

False hostname in the connection to mqtt broker
[image]

MTrab, 2020-01-08:
I see no problems with the current firmware available.

[image]

gregor-samosir, 2020-01-09:
Igor: Found the reason already. In C++ you need to always be aware of overflows. I had the wifi_hostname and ota_password defined as a too small string. While setting it in the hotspot-portal you could easily overflow that string and send the program into doing weird stuff. Fixed now.

The last version of HeishaMon provided by @Igor Ybema 9.1.2020 fixes all issues with hostname and firmware update. Testet USB update, OTA update and firmware update over the webpage.

closed 2020-01-09 by gregor-samosir