heishamon/HeishaMon #118: Additional Topics of CZ-NS4P

seitfa, 2020-01-27:
The extension board CZ-NS4P only connects to the main board using CN-CNT (and some Powersupply-Pins). So there has to be some additional commands like limiting power consumption (which is done by 0-10V Intervace on CZ-NS4P).

It would be awesome if this commands could be implementet ein HeishaMon.

seitfa, 2020-01-27:
I hope CN-CNT-Bus is shared and all Commands of CZ-NS4P can also be sniffed on external CN-CNT-Connection, no need to intercept between CZ-NS4P and main board.

Egyras, 2020-01-27:
@seitfa we need user with cz-ns4p board and connection to pc via ttl-usb

seitfa, 2020-01-28:
If nobody already has cz-ns4p I'll buy one and resell it afterwards.

Are there any spezial requirements for the TTL Interface? I would have bought[ this one](https://www.amazon.de/DSD-TECH-Seriell-Adapter-Kompatibel/dp/B07BBPX8B8/ref=sr_1_1_sspa?keywords=USB+TTL+Serial+Adapter&qid=1580198521&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExTDA0M0RLVURXMUdQJmVuY3J5cHRlZElkPUEwMjAzMjI3MldUTUxPQzc4MzQyOCZlbmNyeXB0ZWRBZElkPUEwMTM3NjkxMzlUN0VUVlZYQ1RNUiZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=) because of the housing and the abilitiy to switch voltage.

Egyras, 2020-01-28:
@seitfa any usb to ttl 5V

MiG-41, 2020-01-29:
@seitfa plese remember about connecting/sniffing points. 
Since you are going to "borrow" CZ-NS4P it will be no good to solder anything on it. 
The easest way will be to use additional CN-CNMODE connector on HP mainboard. ( in readme you can find siutable plug for it).
You can also make a Y cable ,but i belive it is more time consuming.

seitfa, 2020-01-30:
@MiG-41 Any Ideas on running CZ-NS4P without a HP? (i currently don't own one)

Especially the powersupply seem not to be documented, just the connections to the mainboard.

MiG-41, 2020-01-30:
Probably CZ-NS4P will need hex answers from HP. I'm not able to forseen i.e. what happen ,when CZ-NS4P will send a command to HP sending to set some particular temperature (i.e. Pool) , and not receive a proper answer...
Teoretically it could be possible to set a Internet bridge You get command from CZ-NS4P ,and this command is sent to someones HP ....
But that's needs to prepare some solution.

simons4647, 2020-03-16:
Hallo, ich habe eine CZ-NS4P, leider funktioniert die Kommunikation mit meinem Usb-232 TTL Converter nicht. Dieser funktioniert aber mit einen Arduino (softwareserial).
Würde die Daten gerne bereitstellen.

gregor-samosir, 2020-03-16:
> Hallo, ich habe eine CZ-NS4P, leider funktioniert die Kommunikation mit meinem Usb-232 TTL Converter nicht. Dieser funktioniert aber mit einen Arduino (softwareserial).
> Würde die Daten gerne bereitstellen.

Welche Unterstützung brauchst du um die Daten bereit zu stellen?

simons4647, 2020-03-16:
Hallo, habe einen 5kw Monoblock WH-MDC05H3E5. Ich wollte vorerst die Daten von der CN-CNT Schnittstelle mitloggen. Leider bekomme ich keine Daten  von dort. Getestet mit einem USB-232 TTL Converter als auch mit einem Arduino Uno. Angeschlossen Pin 5 GND und Pin 2 TX  CN-CNT so wie in der Beschreibung.

Egyras, 2020-03-16:
> Hallo, habe einen 5kw Monoblock WH-MDC05H3E5. Ich wollte vorerst die Daten von der CN-CNT Schnittstelle mitloggen. Leider bekomme ich keine Daten von dort. Getestet mit einem USB-232 TTL Converter [...]

Please read all information!

sagaardu, 2020-07-01:
Hello,

Has progress been made on this matter?

At the moment I understand that the power of the machine cannot be modulated in function of a photovoltaic solar installation, correct?

Regards

MiG-41, 2020-07-03:
To be honest also CZ-NS4P is not doing this well.... 
In my opinion better to wrote some smart flow in node-red  using frequency /power input signal and steering with temperature. 
Optional PCB is doing it finally in this way.

To find out all CZ-NS4P features somone needs to have it and make a full logs in all state of work.

sagaardu, 2020-07-07:
> To be honest also CZ-NS4P is not doing this well....
> In my opinion better to wrote some smart flow in node-red using frequency /power input signal and steering with temperature.
> Optional PCB is [...]

Hello,

And with the 0-10 volt input that the machine has, you cannot modulate the nominal power?

For example, using an arduino-type microcontroller so that the voltage value varies according to photovoltaic surplus through a pwm output?

MiG-41, 2020-08-10:
Yes you can , but you will be suprised , how this works .  You give 0V ,and expecting that HP slow down to minimum (20Hz) ,but it doesn't , it is still around 40Hz , when the water temperature is lower then set. Maybe @gregor-samosir will describe it better , since that his opinion ( i was only able to test it - simulate CZ-NS4P via command , and observe the same).

So summarizing , still better to wrote smart integration and increse presset temperature when PV generates surplus.

kinycobain, 2021-01-04:
Hello, I have the CZ-NS4P board, is it possible to read the data from the sensors connected to it?

Regards.

gregor-samosir, 2021-01-04:
If you have the sensors connected to the PCB you have this temps in Heishamon. eg. TOP47, TOP48

HighlyCompressedAir, 2024-09-11:
Implemented in #215.
This issue can be closed.

closed 2024-09-11 by Egyras