heishamon/HeishaMon #119: Pump flow not in l/min

knielen, 2020-01-27:
Hi,

Pump flow is not in l/min at the moment, but in l/sec

gregor-samosir, 2020-01-27:
the error is not known to us, please give more information.

knielen, 2020-01-27:
Heisha Monitor has as description for Flow l/min (liters per minute), but the actual value from the Panasonic is in l/sec (liters per sec). For example, my flow at this moment is 17 (which is in liters per second). So not really a bug, but just a 'wrong' description.

gregor-samosir, 2020-01-27:
17 l/min * 60min = 1.020 l/hour, and this is what you can expect from your HP.

17 l/sec * 3600sec = 61.200 l/hour, that would probably blow up your heating pipes.

Everything is fine, both with your heat pump and with HeishaMon.

knielen, 2020-01-27:
Aaargh, you are right, my bad. Sorry for the confusion!

closed 2020-01-27 by Egyras