heishamon/HeishaMon #122: Name giving in Node-Red for Domoticz flow not consistent, resulting in errors

knielen, 2020-01-30:
A few names of topics are wrong in node-red for Domoticz, resulting in not being recognized and ignored.
For example:

Quite mode TOP18: 
Name in Heisha: Quite_Mode_Level
Name in Node-Red: Quitemode_Level

MiG-41, 2020-01-30:
Yes, that was decided ,to recheck all names. Sorry for that.

Dylantje, 2020-01-30:
I have the correct node Red Names almost working

Egyras, 2020-01-31:
> I have the correct node Red Names almost working

@Dylantje make a pull request for fix

Dylantje, 2020-01-31:
In my github a Alfa version, i do not have the time @ the moment, to make all working... 
https://github.com/Dylantje/Heisha-mon-Global-setup/blob/master/Globalsetup

knielen, 2020-01-31:
I have a new list, which should be correct
[code][{"id":"5a304122.7f39e8","type":"function","z":"6f1bd5f9.4cdd9c","name":"global setup","func":"if (!context.global.heishamon) {\n  context.global.heishamon = {};\n}\n\n// write down the Home Automation applications used\n// Possible values: Domoticz, InfluxDB, openHAB, HomeAssistant\n// comma seperated and case sensitive\ncontext.global.heishamon.HAapplication = \"Domoticz, InfluxDB\";\n\n\n//This is the overview of sensors which has to be sent to the Home Automation system\ncontext.global.heishamon.SensorMapping = [\n        // [\"name of topic\", \"IDx in Domoticz/Name in Home Assistant/ ?? openHAB ??\"]\n            [\"Heatpump_State\", 1703, \"Switch\"],\n            [\"Pump_Flow\", 1609],\n            [\"Force_DHW_State\", null, \"Selector Switch\"], //DHW status (20=off, 10=on 0=unknown)\n            [\"Quiet_Mode_Schedule\", null, \"Switch\"],\n            [\"Operating_Mode_State\", null, \"Selector Switch\"], //7 levels - (0=Heat only, 10=Cool only, 20=Auto, 30=DHW only, 40=Heat+DHW, 50=Cool+DHW, 60=Auto+DHW)\n            [\"Main_Inlet_Temp\", 1605],\n            [\"Main_Outlet_Temp\", 1606], \n            [\"Main_Target_Temp\", null],\n            [\"Compressor_Freq\", 1608],\n            [\"DHW_Target_Temp\", null],\n            [\"DHW_Temp\", null],\n            [\"Operations_Hours\", 1611],\n            [\"Operations_Counter\", 1612],\n            [\"Main_Schedule_State\", null, \"Switch\"],\n            [\"Outside_Temp\", 1610],\n            [\"Heat_Energy_Production\", null],\n            [\"Heat_Energy_Consumption\", null],\n            [\"Powerful_Mode_Time\", null, \"Selector Switch\"], //  4 levels -- 0= off - 10= 30 Minute - 20= 60 Minute - 30= 90 Minute //0= off, 30 = level 3\n            [\"Quiet_Mode_Level\", 1704, \"Selector Switch\"], // 4 levels -- 0= off - 10= Silent 1 - 20= Silent 2 - 30= Silent 3\n            [\"Holiday_Mode_State\", null, \"Switch\"],\n            [\"ThreeWay_Valve_State\", null, \"Selector Switch\"], // 2 levels -- 0=Room, 10=DHW\n            [\"Outside_Pipe_Temp\", null],\n            [\"DHW_Heat_Delta\", null],\n            [\"Heat_Delta\", null],\n            [\"Cool_Delta\", null],\n            [\"DHW_Holiday_Shift_Temp\", null],\n            [\"Defrosting_State\", null, \"Switch\"],\n            [\"Z1_Heat_Request_Temp\", 1702, \"Thermostat\"],\n            [\"Z1_Cool_Request_Temp\", null],\n            [\"Z1_Heat_Curve_Target_High_Temp\", null],\n            [\"Z1_Heat_Curve_Target_Low_Temp\", null],\n            [\"Z1_Heat_Curve_Outside_High_Temp\", null],\n            [\"Z1_Heat_Curve_Outside_Low_Temp\", null],\n            [\"Room_Thermostat_Temp\", null],\n            [\"Z2_Heat_Request_Temp\", null],\n            [\"Z2_Cool_Request_Temp\", null],\n            [\"Z1_Water_Temp\", null],\n            [\"Z2_Water_Temp\", null],\n            [\"Cool_Energy_Production\", null],\n            [\"Cool_Energy_Consumption\", null],\n            [\"DHW_Energy_Production\", null],\n            [\"DHW_Energy_Consumption\", null],\n            [\"Z1_Water_Target_Temp\", null],\n            [\"Z2_Water_Target_Temp\", null],\n            [\"Error\", null],\n            [\"Room_Holiday_Shift_Temp\", null],\n            [\"Buffer_Temp\", null],\n            [\"Solar_Temp\", null],\n            [\"Pool_Temp\", null],\n            [\"Main_Hex_Outlet_Temp\", null],\n            [\"Discharge_Temp\", null],\n            [\"Inside_Pipe_Temp\", null],\n            [\"Defrost_Temp\", null],\n            [\"Eva_Outlet_Temp\", null],\n            [\"Bypass_Outlet_Temp\", null],\n            [\"Ipm_Temp\", null],\n            [\"Z1_Temp\", 1702],\n            [\"Z2_Temp\", null],\n            [\"DHW_Heater_State\", null],\n            [\"Room_Heater_State\", null],\n            [\"Internal_Heater_State\", 3095],\n            [\"External_Heater_State\", null],\n            [\"Fan1_Motor_Speed\", null],\n            [\"Fan2_Motor_Speed\", null],\n            [\"High_Pressure\", null],\n            [\"Pump_Speed\", null],\n            [\"Low_Pressure\", null],\n            [\"Compressor_Current\", null],\n            [\"Force_Heater_State\", null, \"Switch\"],\n            [\"Sterilization_State\", null, \"Switch\"],\n            [\"Sterilization_Temp\", null],\n            [\"Sterilization_Max_Time\", null],\n            [\"Z1_Cool_Curve_Target_High_Temp\", null],\n            [\"Z1_Cool_Curve_Target_Low_Temp\", null],\n            [\"Z1_Cool_Curve_Outside_High_Temp\", null],\n            [\"Z1_Cool_Curve_Outside_Low_Temp\", null],\n            [\"Heating_Mode\", null],\n            [\"Heating_Off_Outdoor_Temp\", null],\n            [\"Heater_On_Outdoor_Temp\", null],\n            [\"Heat_To_Cool_Temp\", null],\n            [\"Cool_To_Heat_Temp\", null],\n            [\"Cooling_Mode\", null],\n            [\"Heat_To_Cool_Temp\", null],\n            [\"Z2_Heat_Curve_Target_High_Temp\", null],\n            [\"Z2_Heat_Curve_Target_Low_Temp\", null],\n            [\"Z2_Heat_Curve_Outside_High_Temp\", null],\n            [\"Z2_Heat_Curve_Outside_Low_Temp\", null],\n            [\"Z2_Cool_Curve_Target_High_Temp\", null],\n            [\"Z2_Cool_Curve_Target_Low_Temp\", null],\n            [\"Z2_Cool_Curve_Outside_High_Temp\", null],\n            [\"Z2_Cool_Curve_Outside_Low_Temp\", null],\n            [\"Room_Heater_Operations_Hours\", null],\n            [\"DHW_Heater_Operations_Hours\", null]\n    ];\n    \ncontext.global.heis [... cut ...] uestTemperature\", \"Thermostat\", null],\n    [\"SetOperationMode\", \"Selector Switch\", null],\n    [\"SetForceDHW\", \"Switch\", null],\n    [\"SetDHWTemp\", \"Thermostat\", null],\n    [\"SetCoolTemp\", \"Thermostat\", null],\n    [\"SetForceDefrost\", \"Switch\", null],\n    [\"SetForceSterilization\", \"Switch\", null]\n];\nreturn msg;","outputs":1,"noerr":0,"x":460,"y":120,"wires":[[]]}][/code]

I tried to create a pull request, but this is new for me and I'm afraid something goes wrong.

closed 2024-09-12 by Egyras