heishamon/HeishaMon #134: Explaining parameters

matulekpl, 2020-03-26:
Hey,
I have few questions about parameters of this heat pump.

TOP33 | sdc/Room_Thermostat_Temp | Remote control thermostat temp (°C)
What temp. it is? Couse this is not temerature of my room thermostat. My room thermostat only send signal to turn on or off my heat pump.

TOP4 | sdc/Operating_Mode_State
The are only 6 states. But earlier I had readings from service cloud and there Was additional state readings. Auto mode was recognized as Auto(heat) or Auto(cooling). Can it be done now in HeishaMon?

TOP21 | sdc/Outside_Pipe_Temp
What this is??

TOP49 | sdc/Main_Hex_Outlet_Temp
Where exactly it is?

TOP68 | sdc/Force_Heater_State
internal or external??

TOP90 | sdc/Room_Heater_Operations_Hours
internal?

TOP91 | sdc/DHW_Heater_Operations_Hours
internal??

matulekpl, 2020-03-26:
And what is Eva Temperature??

emign, 2020-03-26:
0-6 is 7

Dylantje, 2020-03-26:
Perhaps some explane:
[https://github.com/Egyras/HeishaMon/blob/master/heatpump-internal-temperatures.png](url)

matulekpl, 2020-03-26:
> Perhaps some explane:
> https://github.com/Egyras/HeishaMon/blob/master/heatpump-internal-temperatures.png

I have seen that.
Maybe I said it wrong. For what this two temperatures is used to?
I mean TOP21 and TOP49?

MiG-41, 2020-03-27:
First ,we are not Heat Pomps designers , and only they know for what their project is like this :)
Such data provides HP ,and such data are also present in Cloud.
I can only assume that TOP49 is a water temp , beetween TOP5 and TOP6 to know how much heat generates compresor and how much heat generates heater ( if turn ON).
TOP49 in T-CAP also controls water temperature when defrost is deice mode 1. When water outgoing from heat exchanger will be lower then 22 deg it change mode to 2.

TOP21 is one of the deciders for starting defrost ( and this is not done by TOP52 as the name suggest). Please see Service Manual (chapter 13.2.2. Deice Operation ) as it is long explanation.

Regarding your first questions:
TOP33 is the temperature from controler (with can be taken also to room). Cloud is calling it Room Thermostat. 
TOP68 it is showing the state ,when HP is turn on without compresor (so only heaters generating heat). It is either "emergency " ,or ou can also turn on HP in such mode from quick menu ( top right icon in quick menu). 
TOP90 and TOP91 this values are identical to the one inside controler. 
So yes Room_Heater_Operations_Hours it is always internal,  but DHW_Heater_Operations_Hours can be internal , external or mix of them ( depending if you change in options from intrnal to external).

matulekpl, 2020-03-27:
> TOP33 is the temperature from controler (with can be taken also to room). Cloud is calling it Room Thermostat.

Which part is called "controller"?? 
[image]
This??

MiG-41, 2020-03-27:
Yes :) 
In bottom right corner of This yuo have a termistor.

matulekpl, 2020-03-27:
WTF?! Nobody told me that I can install this in room...
I understand that it has to be wire-connected to be in room?

MiG-41, 2020-03-27:
Yes ,it can be moved to room. 
You need two wires beetween that controller and HP mainboard.

matulekpl, 2020-03-27:
Any power needed?

MiG-41, 2020-03-27:
No , only this two wires. ( power and data are in them).

MiG-41, 2020-03-27:
But if you have external thermostat maybe it is better , since this controler has also resolution 1 deg  so at least 1 deg have to be temp.lower to turn on HP. Sometimes it is too much.

closed 2024-09-12 by Egyras