heishamon/HeishaMon #153: HeishaMon v3 communication PCB shipping to Australia

hotswapster, 2020-07-22:
Good evening.
I like your project and PCB and would like to use it in my house. I tried to purchase `HeishaMon v3 communication PCB` from the Tindie store but it will not ship to Australia. Is there another way I can purchase it?

I would like to help and provide support for the split system air conditioners (e.g. CS-E24LKR and CS-E12JKR). Using your hardware interface and finding common items in your protocol map is a good start I think.

Regards,

Jaidan

MiG-41, 2020-07-22:
Hello , 
Please join the Slack channel ( link at the top of ReadMe.md) , there are more active colleques there to help in booth cases :)

hotswapster, 2020-07-22:
Thanks @MiG-41   I completely missed that link!

closed 2020-07-22 by hotswapster