heishamon/HeishaMon #158: heishamon.yaml - #### SENSORS BELOW ARE NOT IN FIRMWARE ####, not working

bse4792, 2020-09-13:
heishamon.yaml
sensors are giving unknown see picture
i have not changed anythin in yaml file all others sensor/switches works perfectly
[image]

trsqr, 2020-09-13:
You are using Home Assistant, right? I recently committed a change that should have fixed those. Did you pull the master recently or which version are you using (this would be helpful to mention in a bug report)?

bse4792, 2020-09-13:
Hi. Yeah updated firmware and reset settings, pulled latest yaml file yesterday.

trsqr, 2020-09-13:
Hi, currently on mobile, but try this:

Under   #Energy use/production look at places where it says sensor.heishamon_blahblah and replace those with sensor.heishamon_aquarea_blahblah. So basically just add the aquarea into the variable names for the power consumption and production.

MiG-41, 2020-09-14:
@trsqr , could you in the same time change pattern of MQTT topics , from "heishamon_heat_pump/"  to "panasonic_heat_pump/" ?
"panasonic_heat_pump" is still standard one , with heishamon gives ,and since it is not changed till that time it will probably stay longer....

bse4792, 2020-09-14:
@trsqr i could not get it to work last night, could not find correct sensor.bla_bla_bla.
also i agree with @MiG-41 on topics

trsqr, 2020-09-14:
@bse4792 can you please test the yaml in: https://github.com/Egyras/HeishaMon/pull/159

geduxas, 2020-10-29:
Also there was a problem, that COP is calculated before it get's energy values. Fixed that in future release.

https://github.com/Egyras/HeishaMon/commit/e0bf1e34d510de5b1e8166a065608f40e322c8de

For my HA with that YAML everything works fine now.

trsqr, 2020-10-29:
@bse4792, please test and close this if ok.

bse4792, 2021-01-30:
All fixed in 1.0

closed 2021-01-30 by bse4792