heishamon/HeishaMon #164: S0 kWh meter ID & Topics

stdioGangsta, 2020-09-30:
- Connecting a S0 kWh meter to GND and the port marked '1' on the PCB makes it show up in software as Port '2'

- MQTT topics for S0 connected on Port 2 is described in Readme as:
panasonic_heat_pump/s0/2, but the status is actually reported on:

panasonic_heat_pump/s0/Watt/2 
panasonic_heat_pump/s0/Watthour/2

IgorYbema, 2020-10-01:
@stdioGangsta which PCB are you using? The heishamon v3 pcb has s0_1 connected to GPIO12 and s0_2 to GPIO14. Those match the default settings for port 1 and port 2.  So I am puzzled for this.

I'll change the readme for the topics

stdioGangsta, 2020-10-01:
[It's a v2 PCB,](https://www.tindie.com/products/thehognl/heishamon-v2-communication-pcb-with-wemos-d1-mini/)
According to the schematic it is connected to GPIO12. Still, the heishamon webportal as well as the MQTT topics report the S0_1 port as port 2.

IgorYbema, 2020-10-01:
@stdioGangsta and you have gpio12 set as the gpio for port 1 or port 2 in the settings?
[image]

IgorYbema, 2020-10-01:
@stdioGangsta 
Just noticed a design error in the v2 heishamon pcb. Thanks for that! 👍 
S0_1 is wired to gpio14 and S0_2 is wired to gpoi12. You can change that in the settings if you like (so change the settings in above screenshot to have S0 port 1 GPIO to 14 and S0 port 2 GPIO to 12)
[image]
[image]

stdioGangsta, 2020-10-01:
@IgorYbema yes, I have exactly the same settings as shown by you. (Well, not exactly, because I use a 1000 imp/kWh counter)

stdioGangsta, 2020-10-01:
Great, thanks for figuring it out!

IgorYbema, 2020-10-01:
@Egyras request close of this issue

Egyras, 2020-10-01:
As requested

closed 2020-10-01 by Egyras