heishamon/HeishaMon #166: Service Mode

serav993, 2020-10-01:
Hey guys,

very nice project. I worked a lot with Panasonic & other Ac- oder ventilationdevices, but almost on mechanical site.
My company was one of the biggest main supporter of panasonic in northern germany. I have a lot of "hardware" knowledge, but my programming skills are only a very little bit better then my english language skills :D

Is it possible to set HP in servicemode with your programm ?
For example. Put the waterpump  "On" in service mode and set flow to a selectable value. Like you do on normal wired remote for testing or bleeding. I think with the normal panasonic cloud modul it is possible.
I build a system with Panasonic Heatpump and water-bearing wood stove as bivalent heating system.
When the stove isn't buring, the Heatpump work as a single Device.
When the wood stove burns the "indoor"modul of HP just works as a normal circulation pump. Water bearing wood stove is primary circuit and HP is secondary. system is hydraulic cuted by big buffer tank. Don't worry flow temps are limited by seperated 3-way valves. 
The Problem is, the HP waterpump tries to modulate flow by pipe resitance and heat demand. For my special system it is 
counterproductive and inaugural parents always turn around the thermostats. 
For now i solved that, like to often, with hydraulic tricks and some self-sufficient 3 way values.
Befor I build that system, i thought the additional board solve my problem. It doen't, there was a translation mistake in the german service technican manual.
This heating system is not a normal one, but it works very efficent and cost much less then a normal bivalent heating setup.
If you can help me, i would present a pipediagram and try to explain the hole setup and its working method.
If this problem is solvable with your technics, it will be very great. That updates real "smart" home a lot.
Did you found out the byte queue commands. I don't find anything in your data.
Sorry, my english is not so good....

best regards

MiG-41, 2020-10-02:
Generaly service mode is normaly not needed to use remotly , you just set it from controller and it is enough for normal user.
Second thing ,is that is you change anything there ,HP have to be switched off - stand by ( controller also don't let you in there ,if HP is on). Service Cloud via CZ-TAW1 is doing it in this way ,that first is sending switch off command ( i belive even twice , and also wait for confirmation) and then sends the service changing command.
Generally it is possible to send it now by HeishaMon ,since there where implemented SETCommand topic ,where it was possible to put anything ,and Heishamon just send it to HP. 
Also a command it self is a question of making logs from CZ-TAW1 when sending this command from SmartCloud.( the easiest way). 
So yes , something is possible, you have to publish first SetHeatpump=0 ,to be sure that HP is off , and then logged command.

IgorYbema, 2020-10-02:
Yes I think this is possible.
In the controller you can select service mode, max pump speed and then select the pump speed and turn the pump on or off.
I guess this is possible from the heishamon also. Just need to find out the correct commands now. I'll take a look into this today.

But isn't your setup easier to just have a small buffer (hydraulic neutral) between your installation and the heatpump and have a second pump which you can control yourself after this buffer to pump at your desired speed? 

And yes please share your pipediagram.

MiG-41, 2020-10-02:
Just checked Servicecloud , and it is not possible to change the speed of circulating pomp , and turn it on/off from there....
There is only status ,but not active action.... ( i was pretty sure that you can. Maybe it was possible before some actualization ? ).  
 So it is not possible to log the command from CZ-TAW1.... Any ideas how to find the command ?

IgorYbema, 2020-10-02:
I hope to find the commands by setting the service mode and pump speed on the controller. Probably the hexdump will show the values needed for this.

MiG-41, 2020-10-02:
Hope this works , but i belive controller have another way of communications , so in hexdump you will see only the result in answers, not the steering command it self. 
Maybe a blind shoot to send something to Byte #45...

IgorYbema, 2020-10-02:
> Hope this works , but i belive controller have another way of communications , so in hexdump you will see only the result in answers, not the steering command it self.
> Maybe a blind shoot to send something to Byte 45...

I think it is indeed byte 45 together with some bits on byte 4.

serav993, 2020-10-02:
Thanks a lot. I'm going to report my progress on this.
In the next week I get some cracked boards from Panasonic Aquarea, they should work. Only optically damaged, I hope.
With resistances and some controllers I try to simulate a normal Device and try your Tipps.

IgorYbema, 2020-10-02:
I got it working!
Setting the correct bits on byte 4 allows to enter pump on in service control (you even see the controller switch to that screen :)). It even allows to do this with the pump still on ...
And changing byte 45 does indeed change the pump speed then :)

Soon in the new firmware! Already working in my system now :)

Dylantje, 2020-10-02:
Perhaps more install settings possible? [ change pumpspeed etc ]

MiG-41, 2020-10-02:
I'm not sure , if turning on/off waterflow pomp during HP is on is a good idea in real life.( for test is ok , since you are not doing it for longer time ).
Perhaps that is why this options disapear from Smartcloud...

IgorYbema, 2020-10-02:
That is why you should turn off the heatpump first I think. So the logic for the heating is disabled.

MiG-41, 2020-10-02:
Yes , but  leaving this to user can be dangerous. So maybe with on and especially off function also in byte 04 a bit to turn off HP...

IgorYbema, 2020-10-02:
Turning off the pump in service menu does not turn the pump off if it is running under normal conditions.
And I can't find a situation when turning the pump on can break anything.

Changing the max pump speed while the pump is running under normal conditions will only give a error and the system will turn off if you put the speed to low. So I don't see a reason to hide this or the block the user from doing bad stuff (which I think is impossible).

serav993, 2020-10-02:
If u turn off waterpump in normal mode, high pressure alert of cooling circuit goes on.
The systems are build very safety

MiG-41, 2020-10-03:
Yes , in Tschernobyl they also thought , that when they press the button whole reaction will stop :)

But looks like it is possible only to start waterpump , It will not stop when system will need it.

serav993, 2020-10-04:
Important is to start pump with static flow.

MiG-41, 2020-10-04:
Just tested newest @IgorYbema test version , and that works very well. In his repository it is aslo how this SET command works. 
If you wont to test it just go to https://github.com/IgorYbema/HeishaMon 
Sooner or later pullrequest will be made .

gregor-samosir, 2020-10-04:
Tested the set topics from @IgorYbema this evening. 

- SetPump 0 / 1
If you change to 1 the service menu opens (on remconn) The Water Pump ist running with max speed. Thats what we expected. But if I change back to 0, the heat pump stays in the service menu und does go back to normal operations. Only press the left-top button (back) on the remconn, the HP operates in normal mode.

- SetPumpSpeed ??-200
It looks like we can change the max speed on the running heat pump without an stop of the HP, That's fine

What I am missing is the point to come back from SetPump from 1 -> 0 to start normal operation without need of the button press on the remconn.

But it looks like an very nice feature, very usefull on my serial system.

MiG-41, 2020-10-05:
@gregor-samosir did you tried to work "normally" after SetPump from 1 -> 0 ,even that on controller display is this config window ? I believe this inconvenience is not blocking this, and you can operate normally,  only it exist on display all the time. 
Off course it will be good to find a way to quit from that,but till now it is ok . And if you want to see the disply you have to press go back button first...
Regarding SetPumpSpeed ,lowest possible value is 65 (with corresponse to 0x41 ) ,highest 255 (0xFE), and default is 151 (0x96).

gregor-samosir, 2020-10-05:
@MiG-41 Will do more tests on the point SetPump 1 -> 0 to check the normal funktion.

closed 2020-11-27 by serav993

serav993, 2021-05-31:
Your device works like it should!!
Very Nice work thanks a lot