heishamon/HeishaMon #173: Is 3.3V enough TX level for the Panasonic Heatpump ?

Happyboy0980, 2020-10-20:
In the schematic there is no level converter for the TX part.
This means that the heatpump only gets 3.3V as TX level.

Is this in all cases high enough for a stable connection ?

Egyras, 2020-10-20:
@Happyboy0980 

You have issues with stable connection? When we connect device to cn-cnt its fine to have ~3.3V on TX level.

MiG-41, 2020-10-21:
Also CZ-TAW1 (at least one i have) use PL2303SA chip ,with acc to Datashet:

"Flexible signal level requirement on the RS232-like
serial port side also allows PL2303SA to connect directly to any 3.3V~1.8V range devices."

And Panasonic connected CZ-TAW1 directly to CN-CNT with 5V level....

IgorYbema, 2020-10-21:
Which schematic are you talking about? The mostly used heishamon v2 and v3 do have level shifting on rx and tx

IgorYbema, 2020-10-21:
All the schematic match level shifting design of page 12 of https://www.espressif.com/sites/default/files/documentation/esp8266_hardware_design_guidelines_en.pdf

You probably are missing the 5v pullup on the panasonic side but that is on the panasonic board already. See the schematic of the panasonic in the service manual.

Happyboy0980, 2020-10-21:
This schematic was my rev: 
https://easyeda.com/igor_6537/heishamon-esp8266ex

Looking to the schematic and the picture of the pcba, and the text in the schematic, the CNT connector goes to the heatpump.
The level converter is at the extra IO lines (?)
I check the heatpump manual but couldn't find the pull up resistor but already expected it to be there.

Egyras, 2020-10-21:
@Happyboy0980 here is heatpump schematic - https://github.com/Egyras/HeishaMon/issues/1#issuecomment-562920493

Happyboy0980, 2020-10-21:
did a short deep dive, level converting is there, but it is depening on the pull up resistor and C of the heatpump if it works or not.

Text in the schematic should be different (now it is 2 times the same/incorrect..) 

**TX towards the panasonic** propose **TX HeishaMon -> RX Panasonic** 
**RX from the panasonic** propose **RX HeishaMon <- TX Panasonic**

closed 2020-10-21 by Happyboy0980