heishamon/HeishaMon #181: Incorrect sequence

geduxas, 2020-10-28:
https://github.com/Egyras/HeishaMon/blob/c08377209c3e0c61fd99201f560fd62b46797edf/Integrations/Home%20Assistant/heishamon.yaml#L576-L613

Please fix that, COP calculation should be after power calculations.

MiG-41, 2020-10-28:
But there arn't any power calculations ....

geduxas, 2020-10-28:
COP calculated from sensor.heishamon_w_production and sensor.heishamon_w_consumption this is calculated later in YAML

MiG-41, 2020-10-28:
sensor.heishamon_w_production and sensor.heishamon_w_consumption are not calculated , they have value received from mqtt.

geduxas, 2020-10-28:
no, from MQTT you get sensor.aquarea_dhw_power_produced, sensor.aquarea_power_produced, sensor.aquarea_dhw_power_consumed, sensor.aquarea_power_consumed and from that 

https://github.com/Egyras/HeishaMon/blob/a05b4448be822b98b7c46d727073fa66d413d0b4/Integrations/Home%20Assistant/heishamon.yaml#L597-L617

you get 'sensor.heishamon_w_production and sensor.heishamon_w_consumption. Because it is after COP calculation, HA always displays 0

geduxas, 2020-10-28:
I have committed pull request 
https://github.com/Egyras/HeishaMon/pull/183

MiG-41, 2020-10-28:
At my side HA calculate COP, and it is not 0 when HP runs , but ok , now i see the point.

geduxas, 2020-10-28:
maybe yours's yaml configuration is different? I didn't get anything than 0 from first try. I am new here, maybe 2 month's :) Only switching orders COP/Energy solved my problems.

MiG-41, 2020-10-28:
I had identical to actual one . Just now changed to your proposal to see if this calculation still be ok .
But there were complains from others , that this COP is not calculating at all. So hopfully this solve that problem on some users.
Also in case of HA i'm rookie , and can't tell why at my place his worked....

trsqr, 2020-10-29:
Same here, I see COP calculation despite the order (my heishamon.yaml is pretty customized though), but I had other people in another discussion forum that were not able to get it working. Let's see if this fixes it.

MiG-41, 2020-10-29:
I changed order , and it is ok also.
From my opinion actual also should work , since this four sensor.heishamon.... should have some values from previous read . So COP could not be as actual , but still should be calculated...
But agree ,this change should do it better.

closed 2020-10-29 by geduxas