heishamon/HeishaMon #189: Model decode

geduxas, 2020-11-02:
I starting to see some similarities in model numbers

| Value | IDU code       | ODU code       | IDU            | ODU        | KIT           | Power [kW] | 1ph/3ph | T-CAP/HP           |
|-------|----------------|----------------|----------------|------------|---------------|------------|-------- |--------------------|
|     2 | E2 CF 0D 77 09 | 12 D0 0B 05 11 | WH-SXC09H3E5   | WH-UX09HE5 | KIT-WXC09H3E5 |          9 |     1ph | T-CAP              |
|     4 | E2 CF 0D 85 05 | 12 D0 0C 94 05 | WH-SXC09H3E8   | WH-UX09HE8	| KIT-WXC09H9E8 |          9 |     3ph | T-CAP              |
|     5 | E2 CF 0D 86 05 | 12 D0 0C 95 05 | WH-SXC12H9E8   | WH-UX12HE8	| KIT-WXC12H9E8 |         12 |     3ph | T-CAP              |
|    13 | E2 CF 0C 74 09 | 12 D0 0D 95 05 | WH-ADC0916H9E8 | WH-UX12HE8	| KIT-AXC12HE8  |         12 |     3ph | T-CAP - All-In-One |
|     6 | E2 CF 0D 87 05 | 12 D0 0C 96 05 | WH-SXC16H9E8   | WH-UX16HE8 | KIT-WXC16H9E8 |         16 |     3ph | T-CAP              |

we have 2 devices with same ODU, `WH-UX12HE8`, for one code is `12 D0 0C 95 05` second one `12 D0 0D 95 05` almost identical. only `0C` and `0D` maybe it it would be some date code, or manufactured location.. mine is with `0D`, and date in sticker 2020.01

second we have different power values but same model. 9, 12 and 16 kW with t-cap and without
|Code| Model | Type|
|---|---|---|
|12 D0 0C 94 05 |WH-UX09HE8| T-Cap|
|12 D0 0C 91 05 | WH-UQ09HE8 | T-CAP - Super Quiet |
|12 D0 0B 97 05 | WH-UD09HE8| HP |
|12 D0 0C 95 05 |WH-UX12HE8| T-Cap|
|12 D0 0C 96 05 |WH-UX16HE8| T-Cap|

I think all that data is for internal and external data units, or maybe main boards for hydro module and indoor unit. 

May be it will help in decoding data.

MiG-41, 2020-11-03:
For sure byte #131 with 0B means monoblock cover . Alsu it maches to Super Quiet unit , where ODU cover is monoblock.

trsqr, 2020-11-03:
Yeah, I was last week trying to write this open bit-by-bit to find similarities. Separation between IDU and ODU codes is one possibilty indeed. I also have here one J-gen 5kW split unit that I'll get up and running soon - have already a HeishaMon ready for that too.

MiG-41, 2020-11-03:
> `0C` and `0D` maybe it it would be some date code, or manufactured location.. mine is with `0D`, and date in sticker 2020.01

One of these WH-UX12HE8 is mine , and it is manufactured in 2018...

geduxas, 2021-01-24:
> manufactured location ... Malaysia

Yesterday we have discussed about H generation. And have found that latest H series have SMART DHW support, as J series. So I still think that it could be revision number, or software version :)

IgorYbema, 2021-01-31:
New version of code just compares all 10 bytes to produce an ID.

geduxas, 2021-11-17:
interesting findings, just by accident found secret meniu:
[image]

and by that there is interesting addresses: 

[image]

this maps almost same values as we use as model numbers :) maybe there is more interesting stuff there ? 

E2 CF | 0C 74 09 | 12 D0 | 0D 95 05

but somehow 12 0D apperes as 0x0D11, and 0c 74 09 as 0x09740B also in address 93 i have 0xCFE1 (but in heisha e2 cf )

trsqr, 2021-11-17:
Interesting.

My model string is:
62 D2 --- 0B 41 54 --- 32 D2 --- 0C 45 55

Address 93: 0xD2 61
Address 94: 0x54 41 0A
Address 95: 0xD2 31
Address 96: 0x55 45 0B

Byte order seems to be different in these fields (big-endian vs. little-endian). In addition, the last number in the Address 93-96 is smaller by one than in the model string.

MiG-41, 2021-11-18:
My model:
E2 CF --- 0D 86 05 --- 12 D0 --- 0C 95 05

Address 93: 0xCF E1
Address 94: 0x05 86 0C
Address 95: 0xD0 11
Address 96: 0x05 95 0B

So yes, looks like you can read this string also from control panel , good find !

IgorYbema, 2021-11-18:
I believe all values are stored in those 'addresses' but some addresses contain only one byte some two bytes and some more bytes. That's why they only have 96 addresses and we have 203 bytes. They are also not in the same order.

MiG-41, 2021-11-19:
Maybe not All , but  for sure some of them could be find , most of them in hex values .

geduxas, 2021-11-19:
> Maybe not All , but for sure some of them could be find , most of them in hex values . ( At least now we know what is in 93-96 adresses ;) )

Also 12 :) as manual shows it's flow meter
It seams that there is all data remote controler gets.. it could be useful data to determinate unrecognized bytes and structure..

MiG-41, 2021-11-20:
It is only usefull , when you do something near HP... 
Some time ago tried to figure out some of them:
https://forum.muratordom.pl/showthread.php?192315-Panasonic-Aquarea&p=7762492#post7762492
 For me most usful is adress 8 ( with gives Target / Outlet Temp) and adress 9 (  Heater Exchenger / Inlet Temp) .

closed 2024-09-12 by Egyras