heishamon/HeishaMon #190: Remote does not work

power-dodge, 2020-11-07:
for me the control no longer works with 0.9b. But with 0.7 everything still works. 
Had just been looking for the update for a long time. The simple on / off did not work. Everything else doesn't work. 

MQTT settings were not changed, just updated. The 0.7b is played back and the control goes immediately. When I updated to 0.9b I didn't see any changes in the MQTT settings.

greeting

geduxas, 2020-11-07:
Update MQTT setting, becouse topic names have been changed i think.

power-dodge, 2020-11-07:
Hi, thanks for the reply. 
The MQTT connection exists, I get all the values ​​from the Panasonic, but I can't control them. 
Just tried it again, MQTT settings are the same, I also briefly activated expemertal PBC but nothing.

With 0.7 I can also change the hot water target temperature, with 0.9 nothing can be controlled.

MiG-41, 2020-11-07:
So what topic in MQTT is used in your system to change hot water target temperature ? Becouse when you changing this ,one particular topic is also changing (it's value)

geduxas, 2020-11-07:
Which system do you use? Personally i am using home assistant, everything works fine.

power-dodge, 2020-11-07:
Helo,

i using the PCB: https://www.tindie.com/products/thehognl/heishamon-v31-communication-pcb/
SmartHome Software: Symcon
MQTT Topic: Mqtt topic base: Waermepumpe

MQTT DHT: Waermepumpe/SetDHWTemp      (Integer, Retain (off))

With 0.7 i can send command and receive
With 0.9 only receive

back to 0.7 i can again send and receive

geduxas, 2020-11-07:
It's wrong. Topic should be :

panasonic_heat_pump/SetDHWTemp

MiG-41, 2020-11-07:
Not necceserly , you can change the Mqtt topic base in Settings web page of heishamon.

Please chesk if it is "Waermepumpe" instead "panasonic_heat_pump"

geduxas, 2020-11-07:
Ups i was wrong thought mqtt topic is bin.

power-dodge, 2020-11-07:
hello, 
thank you very much for the answers! 
I have now found out the problem, there seems to be a small bug in the Heishamon. 

It does not allow the topic to be changed. Waermepumpe to waermepumpe is only receiving. Waermepumpe to panasonic_heat_pump is only receiving. It can only be controlled if panasonic_heat_pump is sent to panasonic_heat_pump.

Thank you for the quick help! greeting

MiG-41, 2020-11-07:
I'm not able to see a problem. MQTT topics and directories are case sensitive.

IgorYbema, 2020-11-10:
Just tested. If I change the mqtt base topic I receive on the new topic. And if I send command to the new topic it works also.
So probably you forgot to change the command topic?
I don't understand why it works on 0.7 then for you

power-dodge, 2020-11-10:
Hello, 
thanks for trying it out. 
Yeah, I don't understand either. As I said with 0.9 and other topics on both sides, only data was received. With 0.7 both send and receive. 
I use 0.9 and now it works and only if on both sides in the setting: Mqtt topic base: panasonic_heat_pump 
was entered, now have this Topic in everywhere.

MiG-41, 2020-11-11:
Also tied with "Waermepumpe" and "waermepumpe" as Mqtt topic base. In booth cases works as it should.

Egyras, 2021-01-21:
seems all is solved.

Egyras, 2021-01-21:
solved

Egyras, 2021-01-22:
a

CurlyMoo, 2021-05-26:
Time to close the issue 👍

serav993, 2022-08-12:
Sorry for posting on old issue. I have the same problem with version 2.0, there are no set topics in iobroker mqtt broker. Do i need to trigger them witch mosquitto or stuff like that ?

IgorYbema, 2022-08-12:
You just send the set commands when you need them. It is not necessary for them to exist already, trigger it, or something like that.

serav993, 2022-08-13:
thx, the problem was me. Everything works great.

CurlyMoo, 2023-10-22:
@Egyras @power-dodge can you close this issue?

closed 2023-10-23 by Egyras