heishamon/HeishaMon #193: Solarpump

BazemanKM, 2020-12-16:
Possibility to start/stop the solarpump who is connected to expansion board.

Start/stop is automatic in Panasonic when solar temp is higher than SWW sensor temp.

MiG-41, 2020-12-18:
Probably there is no direct way to do it. 
Perhaps it would be possible to switch off solar system on the fly....
Please join slack ,so we could see posibilities.

IgorYbema, 2020-12-18:
> Possibility to start/stop the solarpump who is connected to expansion board.
> 
> Start/stop is automatic in Panasonic when solar temp is higher than SWW sensor temp.

Can you please explain a bit more your use-case?

BazemanKM, 2020-12-20:
> Can you please explain a bit more your use-case?

The temp. sensor is on 2/3 of SWW tank. When temp. sensor is < solar.temp the solar pump starts pumping.

I have also a sensor at 1/3 of SWW tank, combined withe HeishaMon.

I want to start solar pump if 1/3 sensor temp is < solar. temp.

When this is possible. it should also be possible to start solar pump for buffer CV tank.

IgorYbema, 2020-12-20:
These pumps are installed on the optional PCB I think? There is no way to control the pumps. The panasonic main pcb talks to the optional pcb and requests a pump start depending on its logic.
We can simulate to be a optional pcb but only if the real optional pcb isn't installed. Then we receive the 'start pump' command just like the real optional pcb does. This will allow you to control a pump yourself but requires additional hardware (like a relay).

In your setup, I understand that you want to override the solar logic from the heatpump. Why not remove the solar configuration completly from the panasonic and control the pumps with your own logic. This again requires some relays (like sonoff or something like that) but that is not that hard to implement.

MiG-41, 2020-12-20:
Maybe you could change solar destination to buffer ,and then influance on buffer sensor ( without enabling bufer tank it self). I don't know what consequeces will have this on HP behavior it self ( like high temperature in buffer , but buffer it self not enabled). 

Easiest way is to disable solar logic in HP and do it externaly ,like @IgorYbema mentioned. And beter use SSR relays instead traditional ,like sonoff have , since inductive low current from pump can kill traditional realay after some time.

BazemanKM, 2020-12-20:
> These pumps are installed on the optional PCB I think? There is no way to control the pumps. The panasonic main pcb talks to the optional pcb and requests a pump start depending on its logic.
> We can [...]

Yes, pump is installed on and controlled by optional PCB. If it was possible and easy to override the solar logic from the heatpump that was a great option.

The bennefit when the setup is changed doesn't weight up to the costs i think.

IgorYbema, 2020-12-20:
So yes this is not possible. Can you close the case?

closed 2020-12-20 by BazemanKM