heishamon/HeishaMon #195: H76 error - communication error

Ferforfaen, 2020-12-24:
Hello

Really a great system!

I have my system integrated with home assistant. The first day i had H76 Communication Error. Solved by disconnecting the Heishamon reconnecting it. Then reset error on the Panasonic Panel. 

I have had no H76 the last 14 days. But today i woke up to a cold house. The H76 came at 02:00. 

I have connected the Heishamon with the 5 meter cable. Could a shorter cable solve this? Or is it a known bug?

Regards Tommy Fer

geduxas, 2020-12-24:
No issues for me, do you have additional board?

MiG-41, 2020-12-24:
H76 error have nothing to do with Serial connection , it is a problem with communication beetwen main PCB and controller with disply ( it is connected with 2 wires to main PCB to terminals called REMOTE CONTROLLER). Check this connection.

Ferforfaen, 2020-12-25:
> H76 error have nothing to do with Serial connection , it is a problem with communication between main PCB and controller with disply ( it is connected with 2 wires to main PCB to terminals called REMOTE CONTROLLER). Check this connection.

Ok, thank you. Just weird it never happened before Heishamon :) But sometime weird things happen :)

MiG-41, 2020-12-25:
You have a monoblock ?(where you have to perform  conection beetwen main PCB to disply panel)
Or did you moved display panel/controller from inner unit to the room?
Check that connection ,maybe screws are not tight....

Ferforfaen, 2020-12-26:
I have a 9KW monoblock. J series. 

I have made det connection between heatpump and display panel. I checked the connection in the HP today, will check junctionbox. and connection in display. 

Thank you!

MiG-41, 2021-01-15:
@Ferforfaen did you solve a problem ? Can you close the issue ?

closed 2021-01-22 by Egyras