heishamon/HeishaMon #196: Is 1.0 working well??

sermayoral, 2020-12-24:
Hi friends. I have using heishamon since 0.6 perfectly. The correct received data were always above 95%.

After upgrading to 1.0, I have a correct received data of 36.73%. 

Are you experiencing something similar??

geduxas, 2020-12-24:
Correct received data: 99.99% updated yesterday

MiG-41, 2020-12-24:
Almost sure , that it is just coincident ( To be sure you can go back to 0,6 and check if that possible for you ofcourse).

@sermayoral what kind of board you have ?  

I had some isue with Correct received data , with sudenly also droped this parameter to 55% level. Tried to exchange ESP12 on that PCB , but it was even worse. But when i exchange to oginal again worked good.... Looks like i made a reflow of ESP8266EX chip and that helpd.

sermayoral, 2020-12-24:
Thank you guys.

I have a Wemos D1 Mini. I am in a trip, so I dont want to test anything right now, i will try in a few days.

Maybe the flash went wrong, I had a strange thing. When i have updated to 1.0, I lost my wifi and mqtt credentials, so I had to connect to de Heishamon AP in order to enter these data again. I have never had to introduce credencials again after an update. Maybe something went wrong in this last update procedure

MiG-41, 2020-12-25:
There is a warning in Readme that after flashing to v.1 all credentials will gone (becouse of changing file system) ,so nothing surprising here.
But if you can , wipe your Wemos flash (even twice) ,and flash again with v.1 directly.
Becouse I had observed this isues with lover Correct Received Data also during testing betas ,when file system was changed.... Maybe it is coincident ,or maybe have something common....

sermayoral, 2020-12-25:
@MiG-41 thanks. I didnt see the Readme, my fault. So credentials is normal to be reset.

How can i wipe my wemos flash? Do you mean doing a hard reset with an USB cable and flashing a clean firmware?

MiG-41, 2020-12-25:
To be honest , i don't know solution to bring back good Correct Received Data.
Very first thing could be go back to v0.6 ,and see if there is ok.
Then Hard reset via USB ( some flashing programs like Tasmotizer has Erase option before flashing) , with flashing v.1.0.

Then recheck all connections ,especially this ,with is connected to TX and RX of serial communication. ( But also power supply is important).

What i think , with helped me was a heat gun and reflowing of ESP8266EX chip under metal case. But this is only last solution , with i'm not sure.

geduxas, 2020-12-25:
Usually ESP flashers have that functionality, to clean whole flash.

sermayoral, 2020-12-25:
Thanks guys, when i come back home from holidays I will check all your advices ☺️.

sermayoral, 2021-01-07:
Guys, First of all I have done a Factory Reset maintaining the same version (1.0) without success. After some hours, the correct received data fell to 30%.

Second test, I have reflashed the Wemos D1 by OTA with the 1.0 version again. Same result :-(

Tomorrow i will try to flash by OTA the 0.9b version to see what happen. I keep you informed

Thanks!!

geduxas, 2021-01-07:
[image]

geduxas, 2021-01-08:
Do you have any resultsy with 0.9b?

sermayoral, 2021-01-10:
> 
> 
> Do you have any resultsy with 0.9b?

@geduxas sorry, I have been busy. It has snowed a lot in Madrid... 

I have just flashed the 0.8b by OTA. At the moment 100 % of correct received data:

[image]

I think in the 1.0, it would have already dropped below 40. However, I will leave it a few hours to see its behavior, and I inform you

sermayoral, 2021-01-10:
Guys, after 3 hours, correct received data is above 95% (as usual before 1.0 in my case):

[image]

What do you think?

sermayoral, 2021-01-10:
[image]

MiG-41, 2021-01-10:
Generally it should not have aany influence on this ,communication is not changed. 
Uou can try to upgrade firmware using a "bridge soft", like tasmota-minimal (from that one it is posible to upgrade any firmware ,from "normal" tasmota sometimes it is not posible).  So first tasmota-minimal , then once again HeishaMon v.1. Maybe this flush somhow flash...

sermayoral, 2021-01-10:
> Generally it should not have aany influence on this ,communication is not changed.
> Uou can try to upgrade firmware using a "bridge soft", like tasmota-minimal (from that one it is posible to upgrade [...]

Thanks @MiG-41.

I have flashed with tasmota-minimal, and an AP "ESP-XXXXXX" has appeared. I can connect to this AP, but i cannot resolve the IP 192.168.4.1, so i don't know how to proceed. Any idea?

MiG-41, 2021-01-10:
Looks like everything is going bad .... Maybe you are able to switch power off and then on ( as it is described there). 
Finally you need to climb to your HP...

sermayoral, 2021-01-10:
> Looks like everything is going bad .... Maybe you are able to switch power off and then on ( as it is described there).
> Finally you need to climb to your HP...

I'm climbing right now... 

[image]

sermayoral, 2021-01-10:
Hi guys.

After erasing the firmware twice and flashing the 1.0 version using an USB cable, i have better results:

[image]

In the previous attemp, I'm sure i would have 30%. I don't know why i don't have more than 95% as usual in previous versions, but i think above 90% is acceptable.

MiG-41, 2021-01-11:
Good that helps now , but i'm almost sure ,that it is not the case of HeishaMon version , but some hardware issues ( like cold solderings , not touching connector , or even esp12 on the wemos.  Observe it .

kenzodeluxe, 2021-01-11:
I've been running v1.0 since its release and it has been absolutely as stable as the versions before: 
[image]

sermayoral, 2021-01-11:
Thanks @kenzodeluxe 

It could be my board. It can have cold solderings or something like that, i'm not an expert soldering. 

The stranger thing was the 30's %. After a full erase and flashing again, i get this data:

Wifi signal: 66%
Memory free: 67%
Correct received data: 92.21%
Uptime: 1 day 2 hours 20 minutes 23 seconds

I'm seeing all data are right, so I assume its confidable, valid for me :-)

MiG-41, 2021-01-12:
> > Generally it should not have aany influence on this ,communication is not changed.
> > Uou can try to upgrade firmware using a "bridge soft", like tasmota-minimal (from that one it is posible to upgrade any firmware ,from "normal" tasmota sometimes it is not posible). So first tasmota-minimal , then once again HeishaMon v.1. Maybe this flush somhow flash...
> 
> Thanks @MiG-41.
> 
> I have flashed with tasmota-minimal, and an AP "ESP-XXXXXX" has appeared. I can connect to this AP, but i cannot resolve the IP 192.168.4.1, so i don't know how to proceed. Any idea?
> 
> EDIT: I think i have got bad luck: [arendst/Tasmota#5743](https://github.com/arendst/Tasmota/issues/5743)

Should be tasmota-litle , minimal has no setup page.... Sorry for that.

sermayoral, 2021-02-08:
Everything is working well after the second flash will full erase. Thanks!!

closed 2021-02-08 by sermayoral