heishamon/HeishaMon #198: Possible to run in standalone mode

chmadkk, 2021-01-04:
Hi
I was wondering if IT could be possible to implement an function to run the full webinterface in hotspot mode ? 

IT would be a great fault and service tool.

IgorYbema, 2021-01-04:
Hi @chmadkk,
A wonderfull idea to use heishamon as a service tool for the Panasonic H and J-series.

There are two possibilities. Indeed, we could adjust the code to add hotspot function (somewhat like it already does for the initial config). You then could connect any phone or laptop to the heishamon hotspot and read values directly from the topics on the webpage of heishamon. However this code change is not easy currently.

Another way which already works is to enable the wifi hotspot on your phone or laptop. This is mostly one or two clicks away. Then just connect heishamon to that wifispot (and only need to do that once, it will remember it for the next time you need it as a service tool). Then you can browse to the topics values on the Heishamon webpage from your phone or laptop. Also, this way you would be able to install a mqtt broker and fancy mqtt dashboards on your phone or dashboard allow even more user friendly service information.

Then again, a real panasonic aquarea service worker should be having such service tools already I think. If I am not mistakes, the CN-CNT on the panasonic board is just for such a official tool. And it would be bad if heishamon shows more usuable values than the official service tool :)

chmadkk, 2021-01-04:
Yeah i see the point. 
The official tool is a dumb cable you need a laptop for. 

IT was actually my the Guy who serviced my unit that was interested. Because the need to always Carry the laptop with you. 
Everything Gets downsized 😁

MiG-41, 2021-01-04:
Maybe you have this software with service have ?

CurlyMoo, 2021-05-26:
This feature has been implemented in the fork by @IgorYbema

rpannekoek, 2022-02-28:
I was about to report a security issue when I noticed this related feature request.
But back to my security issue: I used the HeishaMon AP mode to configure my home WiFi network.  After successful connection to my home network, it drops the HeishaMon Setup network, as expected.
However, I have my home WiFi configured to switch off between 23:00 and 6:00.
It turns out that Heishamon re-actives its AP mode as soon as it looses connection to the home WiFi network (in my case: at night).
However, the HeishaMon Setup is an unsecured WiFi network, which makes it extremely easy for hackers to monitor and control my heatpump now!
So, although I can see why this may be a convenient feature, it is very undesirable from a security perspective.
IF the HeishaMon Setup is automatically re-activated after initial setup, it should at least be securied.

IgorYbema, 2022-03-01:
I understand your concerns but I see no reason to change this behaviour in the public available version of heishamon.

If a hacker is interessested in controling your heatpump it most then be close the the heishamon and therefore be very close to the perimiter of your house. If he is that committed,  a WPA2 protected wifi is almost as easy to hack as a open-wifi network.. Probably the reason why you close down your wifi at night? Only a wired ethernet heishamon will be protected to cover this issue but that is not on the roadmap for now.

You are free to compile your own heishamon code for your own use with a password in the hotspot configuration if you deem that necessary.

rpannekoek, 2022-03-01:
That's an interesting reasoning... you're basically saying that WPA2 protection is useless?
The fact that devices like solar inverters offer poorly protected WiFi networks resulting in security issues has been in the news. In those cases, it's often about people forgetting to change the default credentials, but in case of Heishamon there are no credentials needed whatsoever (other than credentials needed to upgrade the firmware).

I'm not nearly as paranoid as many security experts, but for me this issue is serious enough to compile my own Heishamon code, indeed.
I just wanted to make you aware of this significant security issue in an otherwise excellent piece of code.

IgorYbema, 2022-03-01:
Yes WPA2 security is useless for an experienced hacker. But I don't see any reason for such a experienced hacker to destroy a heatpump through heishamon hotspot, only if he is your neighbor who is complaining about the noise it is making :-)

Yes, I am aware of this and currently I see the usability in favor of security for this feature.

The inverter issue you are pointing at is missing the real point. This is what you often see if people try to pick up a hot topic security issue but don't really understand the real reason. In some cases some chinese inverters where so badly secured that you could easily disable all inverters in one region with one button, through the cloud api interface too which those inverters are connected. That is a real security threat, not the single inverter for one single household having a default password installed.

rpannekoek, 2022-03-01:
Okay, let me do one more attempt to scare the hell out of you...
When Heishamon is in Access Point mode, anybody close to my house can connect to the Heishamon-Setup WiFi network, open the Settings page, view the HTML source and see the SSID and password of my home WiFi network (unencrypted!).
So now they can log into my home WiFi network and now let's hope they can't hack my router...
If you don't consider that a big security leak, then I surrender. :-)

IgorYbema, 2022-03-01:
Again, WPA2 is faulty. Yes true, you can easily steal the wifi password using the hotspot on heishamon. But if anyone is even remotely interested in your wifi network he/she doesn't need a security issue on the heishamon to gain access your wifi network.

If you really need a secure wifi go with wpa2-enterprise or higher.

Maybe think about a better way to protect the wifi in the code while keeping the usuability for users who have changed theire wifi or other reasons that heishamon needs to restart the hotspot. Then we could implement that. Remember it is all open source and community driven so while talking about security issue, you could better spend time in making it better.

CurlyMoo, 2022-03-01:
> Remember it is all open source and community driven so while talking about security issue, you could better spend time in making it better.

I think we can settle this issue with the inventation. I've done various PR's on the HeishaMon and have had great discussions with @IgorYbema implementing them. So, it would be great if you could implemented the changes you feel comfortable with for the better of all of us users.

rpannekoek, 2022-03-01:
I'm willing to contribute, but then we should settle on an acceptable solution.  Indeed, there is tension between usability and security (as usual).
Personally, I would expect the Heishamon to not automatically fallback to Access Point mode at all.  If a user decides to change his WiFi password (I wouldn't consider doing that; I havve way too many connected devices that would have to be reconfigured), he can use the double reset to reconfigure the Heishamon.  Seems good enough to me.
Alternatively, we could let the Heishamon fallback to Access Point mode  (admittedly, it is convenient that I can still monitor my heatpump after my WiFi shuts down :-),  but ensure the WiFi network is WPA2 protected (not sure if you can do that with an ESP8266) or put a login form in.

CurlyMoo, 2022-03-01:
> Personally, I would expect the Heishamon to not automatically fallback to Access Point mode at all.

I love that feature. But it could be more safe by letting you configure a hotspot password.

> he can use the double reset to reconfigure the Heishamon.

In my case that would mean opening up the heatpump which i'd rather not do when it's running.

> but ensure the WiFi network is WPA2 protected (not sure if you can do that with an ESP8266)

Exactly, i believe you can

IgorYbema, 2022-03-01:
Another option to start a hotspot with the previous SSID (padded with a heishamon identifier) and the previous password.

IgorYbema, 2022-03-01:
Or just for now add the configurable option to 'not start hotspot when wifi not available' and look for a good solution later

rpannekoek, 2022-03-01:
Okay, what about a mix of these?
A configurable option to:
* Not start Access Point Mode at all
* Start Access Point mode with "HeishaMon-Setup" SSID and same password as home WiFi
* Start Access Point unsecured

rpannekoek, 2022-03-04:
https://github.com/Egyras/HeishaMon/pull/244