heishamon/HeishaMon #199: HeishaMon via Ethernet?

michalwoz, 2021-01-14:
I would like to talk to my Aquarea via Ethernot, not WIFI.
Is there any ready to use solution?

MiG-41, 2021-01-15:
You could try GoHeishaMon :https://github.com/lsochanowski/GoHeishaMon
Forseen to run on orginal Panasonic CZ-TAW1 module ,but you can start it on any machine (like RaspberryPi , OpenWrt router , PC with linux or Windows...), you have to made only serial conection to HP.

markuspulch, 2022-11-18:
Is it possible to run the code on esp32? Then it would be possible to use an WT32-ETH0 which has on board ethernet. I have the same issue and would like to avoid wireless connection.

IgorYbema, 2022-11-18:
Not without some code changes

markuspulch, 2022-11-18:
I was somehow expecting this ;-(
Are there many ESP8266 specific quirks in there, or should it be straightforward to adapt it to ESP32? I guess, libraries needs to be exchanged, ports needs to be adapted... But otherwise it should work, right?
Is there anything planned in this direction? Depending on wifi is somehow sub optimal if you have wifi issues. I already added an external antenna to your board, installed an access-point in the same room (ubiquiti) but still fighting with re-connect issues. A simple LAN connection would just solve the issue...

geduxas, 2022-11-18:
It's writen in arduino.. i think it should run fine on different hardware.. and yes ethernet part will be missing..  just try to compile in arduino for esp32 and you will find out :)

ddokupil, 2023-01-23:
Has anyone tried it already? ESP8266 is very unreliable, at least in my network infrastructure. I always had connectivity problems regardless of if it was ESPHome, Adrduino project or HeishaMon. Can't tell if it's due to AiMesh, Asus WRT or something else. My mesh node is like 1m away from my ESP8266 and still breaks connection every once in a while. I have no such problems with ESP32, it doesn't even need to be connected via ethernet.

ddokupil, 2023-01-26:
First of all it needs esp8266wifi.h  changed to wifi.h everywhere (would need to use directives), than there is DoubleResetDetect which may not work as expected on esp32, #include ESP8266WiFiGratuitous.h needs to be taken care of (not sure what it does and is it needed on ESP32) and that's just a beggining. I ended up with a wall of errors for now. Porting it to ESPHome would be awesome and it might turn out to be euqally troublesome. There is also a GoHeishaMon project that does the same but it's abandoned and lacking some of less common functionalities.

geduxas, 2023-01-26:
Also there is code which modifies Panasonic taw adapter to run heishamon.. so it's possible to have wired Heishamon..

ddokupil, 2023-01-26:
> Also there is code which modifies Panasonic taw adapter to run heishamon.. so it's possible to have wired Heishamon..

GoHeishaMon

MiG-41, 2023-01-27:
@ddokupil  ,you can ty fork of GOHeishaMon by rondoval : https://github.com/rondoval/GoHeishaMon
He is doing update frequently following Heishamon project , since @lsochanowski don't have Aquarea anymore , and is lest interested to make updates...

eldrik, 2023-10-20:
I fiddled the last days getting a w5100 Ethernet shield connected to an esp8266 and after some code changes (mostly on changing the serial switch to gpio 13 & 15 because Ethernet shield needs 12 13 14 & 16 for spi communication), it is running fine with Ethernet connected the Webserver is reachable and Mqtt connections working.

The wifi ap is also working thus I did not tested it if both interfaces connected to the home network will cause any trouble or high cpu. 

There are already the needed lwip expansions in esp 3.02 included for w5500, w5100 & encj.. Ethernet interfaces) so the changes are not really that big and mostly done in Heishamon.ino

on the other side because of the needed extra gpios some Heishamon features can‘t be used openthern for example.

SirIcy, 2023-11-09:
> I fiddled the last days getting a w5100 Ethernet shield connected to an esp8266 and after some code changes (mostly on changing the serial switch to gpio 13 & 15 because Ethernet shield needs 12 13 [...]

Could you please share your solution. Do you have a fork?

eldrik, 2023-11-09:
yeah I created a fork https://github.com/eldrik/HeishaMonEth with binaries for W5100 & W5500 Shields. 

Regard the readme when it comes to available gpios & functions.

HighlyCompressedAir, 2024-09-11:
Implemented in #501 
This issue can be closed.

closed 2024-09-12 by Egyras