heishamon/HeishaMon #201: Cannot find thermostat state

supersjimmie, 2021-01-20:
I am using my Panasonic J-series Heatpump with external thermostat, connected to the special Thermostat connector on the main PCB.  In the installer settings it is set to "External Thermostat". This works fine for us.

In the Aquarea Service Cloud, I can see the state of the Thermostat, under "System Status" there is not only the "Operation" (set to "On"), but also "Thermo" which shows the current state of the thermostat. (On/Off).

In heishamon I am unable to find the Thermostat state.

MiG-41, 2021-01-20:
Looks like this bytes are not decoded.... 
https://github.com/Egyras/HeishaMon/blob/master/ProtocolByteDecrypt.md
Maybe via log from HeisHamon are you able to find with byte correcponds to that ?

Egyras, 2021-01-20:
We didnt decoded this value yet

supersjimmie, 2021-01-21:
> We didnt decoded this value yet

Do you know which bit/byte/piece contains this value?
Currently my compiler is broken, but I can rebuild that, put in some code for that bit/byte and try?

MiG-41, 2021-01-21:
It is not a case of compile , it is a case to find a bit in packet , with HP sends ,and decode it . Just like in file i give a link in my previous post.
I belive it is a byte #116 , but it will be good , if you tell what is the value HP give ,when thermostat is of , and what is the value ,when on.

supersjimmie, 2021-01-21:
I am pretty new to heishamon, is it possible to get the hexdump over network? The HP with the heishamon is far away from where my computer is. (so getting it over Serial is complicated for me)

MiG-41, 2021-01-21:
yes, you can do a log of mqtt topic called  log :) 
Bu first , loging and hexloging have to be enabled in heishamon settings , then topic in mqtt apear.

supersjimmie, 2021-01-21:
ah yes.
Now I get multiple lines with "data:". 6 lines with 32 bytes and one with 11 bytes. Then it confirms 203 bytes received.
Should I look at the 20th byte of the 4th set, and compare that between thermostat on/off?

4th set with thermostat ON:
```
data: 94 07 3D 7B B8 10 7E 7C 3D 78 7C 00 00 00 55 55 55 21 87 15 59 05 1A 12 69 00 00 00 00 00 00 00
```
Again with thermostat ON but HP stopped:
```
data: 94 07 3D 7B B8 10 7E 7C 3D 78 7C 00 00 00 55 55 55 21 87 15 59 05 12 12 69 00 00 00 00 00 00 00
```

and finally with thermostat OFF:
```
data: 94 07 3D 7B B8 10 7E 7C 3D 78 7C 00 00 00 55 55 55 21 87 15 59 05 22 12 69 00 00 00 00 00 00 00
```

MiG-41, 2021-01-21:
byte116 is the one , with have value is 59 all the time .... So either it is not this byte , or you have to wait longer... 

Could be byte 118 ,why not

supersjimmie, 2021-01-21:
Yes, byte 118 changed 2 bits.
Waited longer:
```
data: 94 07 3D 7B B8 10 7E 7C 3D 78 7C 00 00 00 55 55 55 21 87 15 59 05 19 12 69 00 00 00 00 00 00 00 
```
I see byte 118 changed again. Now its 0x19 = 0001 1001

MiG-41, 2021-01-21:
So nothing fitting to it.... In this byte 118 is just some value.
I have emulation of optionalPCB with two zones and i'm using external thermostats for that , and ther i get in byte 116 something like this:

HeatMode:
SetExternalThermostat1State-2 and SetExternalThermostat2State-2      -    9A
SetExternalThermostat1State-2 and SetExternalThermostat2State-0      -    69
SetExternalThermostat1State-0 and SetExternalThermostat2State-2      -    A9
SetExternalThermostat1State-0 and SetExternalThermostat2State-0      -    59
But also saw value 99
CoolMode:
SetExternalThermostat1State-1 and SetExternalThermostat2State-1      -    65
SetExternalThermostat1State-1 and SetExternalThermostat2State-0      -    65
SetExternalThermostat1State-0 and SetExternalThermostat2State-1      -    55
SetExternalThermostat1State-0 and SetExternalThermostat2State-0      -    55

Thought , that it will be the same bytes.
Maybe you can look more wide on all 203 bytes ?

MiG-41, 2021-01-21:
BTW , i can see such information in Service cloud in Status ( about thermostat states) , is it really there ? Is this state changes even when HP is turned off ?

supersjimmie, 2021-01-21:
[image]

Egyras, 2021-01-21:
This is different state, thermo off shloud be no heating required.

MiG-41, 2021-01-21:
In Polish it is translated as Thermometer :)  🌡️  Anyway in Zone 2 thermostat on , in seervice cloud there is still off.

MiG-41, 2021-01-21:
@Egyras , no heating required , becouse thermostat signal is off ?

supersjimmie, 2021-01-21:
Two different things.
I control this via the special connector for thermostat. When switching this, the rest of the HP keeps working, including things like the DHW control and the remote controller. There is also another connector that many other people use as a thermostat but if you turn that off, you also turn off the DHW and remote controller and things like that.

So when using the official connector for the thermostat, it really works as expected from a thermostat because it only stops the HP from warming the house, without interfering with the Domestic Hot Water (boiler) system. And it also does not stop the remote controller from working.

The "Operation: On" will be switched on/off when I would use that other connector.
By using the connector for the thermostat, I must also set the settings to "External thermostat" and then the "Thermo On/Off" wil function. And then the "Operation" will stay "On" because it can still heat the boiler/DHW.

To make it a bit more complex, when the HP is unable to get it's minimal power into the house, it will stop heating. So then the Thermotat is still on but the HP will stop adding more heat for a couple of minutes. I guess in English this is "commuting" (in Dutch it is Pendelen).
So at some times, Operation=On, Thermo=On but still the HP is not heating for a couple of minutes.

Do you know some way how to easilly compare those lines of 32 byte data?
When I turn the thermostat on/off a couple of times I see 3 of the lines keep changing.
How can I easilly see which bytes are different?

supersjimmie, 2021-01-21:
Putting them in Excel made it pretty easy.
The only byte that I see change which is not for things like power or pump is again that byte 118.

Egyras, 2021-01-21:
> @Egyras , no heating required , becouse thermostat signal is off ?

Because pump doesnt heat, question if it will have same state if pump gets signal from external thermostat connected to heat pump.

supersjimmie, 2021-01-21:
If it is vissible in the Service Cloud, it must be vissible in the data.
It is a diffent bit than Operation, pump or things like that. Specific bit for Thermo, otherwise the Service Cloud would not know.

MiG-41, 2021-01-21:
But now during DHW i have this Thermo ON. ( And all thermostats are off)....

supersjimmie, 2021-01-21:
Oke, I took a closer look.

first test:
My Thermostat On: Thermo On
My Thermostat Off: Thermo Off

second test:
My thermostat On again within a few seconds:  Thermo Off for a few minutes and de HP does not start yet. Then after a few minutes the HP starts and Thermo On.

third test:
My Thermostat Off: Thermo Off
Start DHW On: Thermo Off
After a short delay (HP starts later): Thermo On

So yes, Thermo is also on for DHW and does not switch 1:1 with the thermostat.
It looks more like Thermo switches to On when the HP is really producing heat.

But actually that is fine for me, that still gives me information that I can use for my automations and logging.
So now I still would like to know which byte/bit can be used to see this state.

MiG-41, 2021-01-21:
You schuld be fine taking into consideration compressor frequency different from 0.
Or tell us , with byte you think it is .Should be no problem to veryfy and implement it as a new TOP.

supersjimmie, 2021-01-26:
I am unable to find the correct bit, So I agree to use something like freq.

closed 2021-01-26 by supersjimmie