heishamon/HeishaMon #204: Correct received data drops from 100% to <2% when in read-write mode

supersjimmie, 2021-01-26:
As long as I have the "Listen only mode" enabled, I receive 100% correct data.
In MQTT log, I see Total Reads and Total Good Reads with same values.

When I disable Listen Only and reboot, data gets dropped and the good percentage drops under 2%.
(even without sending anything)
Strange is that after reboot within less than a minute, the log says:
```Total reads : 162 and total good reads : 3 (1.85 %)```
And also a lot of lines:
```Received bad header. Ignoring this data!```
As if it expected 162 logs in the first minute?

MiG-41, 2021-01-26:
Since "Listen only mode" worked , you have connected CZ-TAW1 in parallel ?
Probably now there is a "war" beetween CZ-TAW1 ,and Heishamon , since also Heishamon want to send a Magic Packets. 
Perhabs you also turn on OptionalPCB eulation ? in that case , there willbe a lot of packet , with HeishaMOn sends...
(OptionalPCB packets are send as much as possible , not like regular packet for data ,with is send every 5s minimum.)

supersjimmie, 2021-01-26:
Optional PCB is disabled.

I do have the CZ-TAW1 in parallel, but since I only change the flag without letting heishamon sending anything, what would be different?

What I mean:
Situation 1: CZ-TAW1 + Heishamon in listen-only no problem
Situation 2: CZ-TAW1 + Heishamon normal but not sending anything is losing a lot of incoming data.

I would expect that there is no serious difference between setting the listen-only flag or not sending anything?
(when not sending, it is the same as only listening)

Or do you mean that heishamon starts sending that "magic packet" every few seconds when not in listen-only, and then also expects more data back?

supersjimmie, 2021-01-26:
Confirmed, I have now tested with the CZ-TAW1 disconnected and now it stays at 100% correct.

closed 2021-01-26 by supersjimmie

MiG-41, 2021-01-26:
Listen only is the flag , with tells HeishaMon not to send Magic Packet ( and probably commands as well , @IgorYbema correct me ,if i'm wrong). So in this state Heishamon don't send anything ,it anly listen , for what HP will send.
HP will not send anything without sending to it A Magic Packet . And this was done by CZ-TAW1 in your case.
Now ,when you disable Listen only Mode , HeishaMon is also sending Magic Packets , and fight with CZ-TAW1.
In Service Cloud , in charts you will also notice , that some unrealistic numbers started apear.