heishamon/HeishaMon #205: how SO watthour counter should be configured?

jaripetteri, 2021-01-30:
I'm using two kwh meter with pulse output connected to heishamon. How this is suppused to function? On v0.9b it added kwh values on previous value and I think it was showing meaningful values. Now on v1.0 it has changed and I remember same behavion was also earlier version before v0.9b. There is value divider both on heishamon settings and also on heishamon.yml template for Home Assistant and output is more or less nonsence to me now. Has old version somehow come back or what is behind this?

MiG-41, 2021-01-31:
Can you show some graphs  , or at least more information , why you think it is nonsence now ?
This functions is continiously improving by @IgorYbema , and no one said that is perfect now. 
And i can't see any divider in heishamon.yml ....

jaripetteri, 2021-01-31:
Here is how it was ln v0.9. Pulses were counted together.
[image]

Here is go it is now on v1.0. So it shows every counted pulse, or two, or three, or half?
[image]

Dividers are here
[image]

and here
[image]

IgorYbema, 2021-01-31:
I am not sure where it got changed in which version. But now you have WattHour and WattHourTotal.
The first one is per mqtt message so you need to add them together yourself. And the second is the total.

IgorYbema, 2021-01-31:
So you either change the topic in the yml file from WattHour to WattHourTotal or create another measurement to add watthourtotal.

IgorYbema, 2021-01-31:
And I notice I already changed it in my repo https://github.com/IgorYbema/HeishaMon/blob/main/Integrations/Home%20Assistant/heishamon.yaml

The main repo (from egyras) will have this corrected soon.

jaripetteri, 2021-01-31:
I have actually commented out value template on my heishamon.xlm. so received values are now like 0,003 or so on that screenshot. With value template those would be 0,000003 or so.

Does it still count some pulses together and send those in one message so that all pulses are counted?
I need to try that WattHourTotal, didn't see that at all and still using WhattHour.

Watts are showing fine, or I understand the concept there.
[image]

IgorYbema, 2021-01-31:
I thought I was clear enough in my earlier message. WattHour only counts watthour between mqtt messages. So each received mqtt message should be added together by the receiving system. You can use this for example if you want to split watthour between cool, dhw and heat runs. You then create a script which adds the watthour mqtt message together based on the current run (cool/dhw/heat).

For your situation, just change the WattHour to WattHourTotal and get the value_template back.

IgorYbema, 2021-01-31:
And ofcourse please close the issue if you are satisfied :)

jaripetteri, 2021-01-31:
Sure, thanks for the clarification.

closed 2021-01-31 by jaripetteri