heishamon/HeishaMon #206: GPIO15 - Wemos D1 unable to boot

DamianBroj, 2021-01-31:
Hi, i have just flashed heishamon on my wemos D1, plugged a level shifting to match voltages with heat pump and got a problem with GPIO15. Having it connected to level shifting part making boot not possible because of its state. Is there any way how to prevent GPIO15 not active on boot and after a while to turn it on? Any ideas, workarounds?

gregor-samosir, 2021-01-31:
Have you looked at the pcb design page?
https://github.com/Egyras/HeishaMon/blob/master/PCB_Designs.md

DamianBroj, 2021-01-31:
Yes, i did. However I built my own using external level shifting module like that: https://wiki.liutyi.info/plugins/servlet/mobile?contentId=30507610#content/view/30507610

5v, 3.3v and gnd are taken from Wemos. LV side to Wemos GPIO15/13, HV to cnt socket TX/RX. 

Any idea how to make it working?

By the way- why the built in TX/RX Wemos pins are not used but there is a serial switch done?

MiG-41, 2021-01-31:
GPIO15 during boot have to be low. Otherwise it searching for SD card.  So GPIO5  during boot gives low to GPIO5  ,and after starts up connect it to RX of HP.

IgorYbema, 2021-01-31:
Reason why original tx/rx isn't used is because that will put out some data on the tx during boot and that can 'break' the heatpump.
So after boot we swap to GPIO15 and GPIO13 and enable the levelshifters with GPIO5.

IgorYbema, 2021-01-31:
Another reason was because some wemos (cheap chinese ones) have the usb connector connected with wrong resitor causing errors on the tx/rx to panasonic.

IgorYbema, 2021-01-31:
Check https://www.espressif.com/sites/default/files/documentation/esp8266_hardware_design_guidelines_en.pdf at page 12 and see if you can change your levelshifter to match that configuration

IgorYbema, 2021-01-31:
And if you want, just remove the lines https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/HeishaMon.ino#L409 line 409 to 412. Check if it works for you with the normal tx/rx

DamianBroj, 2021-01-31:
> GPIO15 during boot have to be low. Otherwise it searching for SD card. So GPIO5 during boot gives low to GPIO15 ,and after starts up connect GPIO15 to RX of HP.

Looking at your reply and the esp documentation i understand that it would be the solution to connect GND-GPIO5-GPIO15 and that would pass gnd line for boot giving GPIO15 low state and after boot will close connection of GPIO15 to gnd letting it to work normally? If so then is that a built in behaviour of GPIO5 or need to be coded?

IgorYbema, 2021-01-31:
I am not sure if I understand your explanation. But the GPIO5 solution must be in the code (and it is in the heishamon code, see https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/HeishaMon.ino#L418).
Before/during boot GPIO5 is low and that blocks the mosfet/levelshifter. So GPIO15 stays low. After boot GPIO5 is pulled high, which closes the mosfet and allowing GPIO15 to talk to the heatpump (which pulls it high).

You can see a simple wemos shield design for this: https://easyeda.com/igor_6537/panasonic-cn-cnt-to-wemos

supersjimmie, 2021-02-01:
Are you sure those schematich are correct?
I see that for both directions, the Source is connected to the ESP-GPIO and the Drain is connected to the HP?
I mean, since they communicate in different direction, shouldn't one be in the other direction?

IgorYbema, 2021-02-01:
Correct :)

IgorYbema, 2021-02-01:
See a nice video about it https://www.youtube.com/watch?v=FwdwzMTmcsU

DamianBroj, 2021-02-01:
@IgorYbema - i missed the circuit between GPIO5 that controlls mosfet . Basically i have followed the first PCB design (the one with through-hole soldering) and changed the external lever shifter with the whole part of the board designed by you there.

@supersjimmie - the shifter is bi directional so no issue here. it just would need one additional circuit that would not let GPIO15 to go high on boot, as @IgorYbema described well in his schemas.

Is there any particular order of moves to connect HM to HP? should the pump be switched off -> connecting HM -> switching  HP on -> switching HM on?

Another thing is that when i tried to trigger the HomeAssistant switches it prompted me that the command is not correct, any experience with the similar?

IgorYbema, 2021-02-01:
Heishamon can be plug in at any time. Actually I almost do that daily on new heishamon tests while heatpump is running :)

About HomeAssisant, I got no clue for you. I am not a HA user myself. But you can always join our slack channel and ask it there. See slack channel details on top of readme of github page.

supersjimmie, 2021-02-01:
I'm currently running heishamon on a breadboard and plugging it in/out many times while the HP is running.
The only thing to keep in mind is that there might be some setting waiting in your mqtt that is sent to the HP as soon as you connect heishamon (gave me some strange behaviour last time).

I am using Home Assistant and used many fixes from this https://pastecode.io/s/96F1nsvMj4

DamianBroj, 2021-02-01:
Great! will report here effect once i'll find few mins to test it then. Thanks guys for your support!

@IgorYbema as you are here the main designer of different PCBs- here's an idea for all of those who would like to build HM even easier not going into too much soldering: based on my case the only one missing element to complete HM even on the prototype breadboard is a circuit between GPIO5, NPN and GPIO15. the rest can be just bought for few $ and plugged with pin cables. So that would be a nice additional way of building HM without even a soldering iron :)

IgorYbema, 2021-02-01:
@DamianBroj How do you think I got started :-) But eventually running on a breadboard still gave me too much errors (due to breadboard resistance/capacitance buildup etc).
I still have some wemos shields (with extra connectors for s0 and 1wire) laying around. But I noticed people just wanted the complete new board so I am not selling them anymore. They where damn cheap (only shipping price was higher) so people didn't want to make it themselves.

DamianBroj, 2021-02-01:
the case is that 20$/EUR is about nothing and that is the price for the complete build up. But shipping even across europe makes it doubled and this becomes nearly 50% of the original Panasonic device- not attractive at least for me :) if i would not build the PCB i would go for the Panasonic TAW

IgorYbema, 2021-02-01:
And that's why it is open source so you can make it yourself :)

MiG-41, 2021-02-01:
Still this one is the simplest way : Wemos Panasonic Aquarea CN-CNT shield through whole.png
Worked some months on spider soldered directly on NodeMCU

gregor-samosir, 2021-02-01:
Ordering and buying the finished circuit board from one of the developers is also something of a recognition of the work done on this project.

IgorYbema, 2021-02-05:
Can you close the issue?

closed 2021-02-06 by DamianBroj