heishamon/HeishaMon #207: Cunfusion around 'SetPumpSpeed'

supersjimmie, 2021-02-02:
My HP has a Max Pump Duty setting in the standard remote controller.

When I set it very low to e.g. 0x5A (which is 90 decimal), the pump will run at a maximum of around 1950 rpm.
This RPM can be found in TOP65 Pump_Speed = 1950.
The value that I have set, the Duty, is seen in TOP93 Pump_Duty = 90.

When I use SetPumpSpeed, I do not set the Pump_Speed but the Pump_Duty.
So Speed and Duty are confusing here.

Even more, when I set it to 90 and the HP stops, the value in TOP93 goes to 0 while in my remote controller it is still 0x5A.

So how can I see the real setting 0x5A (as seen in the remote controller) when the HP is not running?
I would like to use that value as the feedback to the Home Assistant slider that I want to use to control the pump.
(just like many others have both read value and set value for a slider)
I cannot find any TOPxx that represents the 0x5A of 90 value in the remote controller when the HP is not active.

A am able to Set the value, but I cannot always retrieve it from the HP.

geduxas, 2021-02-02:
I think dutty  that is showing in TOP are calculated values from HP, in controller you set maximum value in which it could go. So it's two separate values.

MiG-41, 2021-02-02:
> When I use SetPumpSpeed, I do not set the Pump_Speed but the Pump_Duty.

True, name could be better. But in English langauage in controller option this is named : "Pump maximum speed"

> Even more, when I set it to 90 and the HP stops, the value in TOP93 goes to 0 while in my remote controller it is still 0x5A.
> 
> So how can I see the real setting 0x5A (as seen in the remote controller) when the HP is not running?

TOP93 is actual duty , not Max Pump Duty. HP can modulate water pump below Max Pump Duty to keep DeltaT.
For Max Pump Duty there is a byte 45 ,but no one ask for TOP from it. 
In HA for feedback you can use the same topic with you send command , and belive ,that Heishamon really set it :)

supersjimmie, 2021-02-03:
Thanks, so I would need to modify heishamon to create a TOPxx for that byte 45?
About the feedback, yes sure I can believe it. But what if it is changed on the remote controller, like many other things can be changed at more points than only in HA? That is why I would like the TOPxx as feedback of the current setting.

IgorYbema, 2021-02-03:
Do you also need to set the byte 45 or only read?

IgorYbema, 2021-02-03:
My bad. SET15 | SetPumpSpeed already sets byte 45 but the name is wrong. It should be SetMaxPumpDuty .. i'll change that also in next beta release.

IgorYbema, 2021-02-03:
You can try this version: https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v1.0-iy-8.bin

It containts topic95 for maxduty and changes the SetPumpSpeed command to SetMaxPumpDuty

gregor-samosir, 2021-02-03:
Igor, you are right.

But I think, Supers is also asking for an additional TOP to have the aktual setting on the HP

TOP | 45 | 97 | (hex) 96 = 97 | Maximum set pump speed

IgorYbema, 2021-02-03:
Yes: 

TOP95 | main/Max_Pump_Duty | Max pump duty configured
-- | -- | --

supersjimmie, 2021-02-05:
Thanks Igor!!!
are there any other changes in that bin that I should be aware off?
(It's going to be very cold next week and I don't want to risk malfunction due to something that changed while I didn't know)

IgorYbema, 2021-02-05:
You can check the commit lines (for example in https://github.com/IgorYbema/HeishaMon/tree/main/binaries). They are a bit technical. But the changes are minor and only improvements.

IgorYbema, 2021-02-05:
please close the issue if you are happy with the result

supersjimmie, 2021-02-05:
I have checked it and yes TOP95 is good!

closed 2021-02-05 by supersjimmie