heishamon/HeishaMon #214: Communication with the heatpump stops working after 50 days

ebogaard, 2021-06-14:
I've experienced this twice now: after 50 days of uptime, the Heishamon stops communicating with the heatpump.
At that time I can still login to the web interface and see that the uptime is 50days and a bit.
But:
* The Heishamon doesn't receive/gets new values from the heatpump
* It can't change settings on the heatpump anymore
So it seems all communication to the heatpump is lost at that time.

If I reboot using the option in the web interface, all starts working again.

MiG-41, 2021-06-21:
@ebogaard  what version are you using , and try https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v1.0-iy-cm-14.bin ( you maybe have to set ssid again becouse of new aproach).

ebogaard, 2021-06-22:
I was using "v1.0.a-jw", but if I remember correctly, this issue also happened with older v0.9-versions.

I just flashed "v1.0-iy-cm-14", but can only say for sure if this has helped in 50 days at minimum (if I don't have to restart the heishamon in between).

geduxas, 2021-06-22:
Strange, i didn't noticed any issues. Unfortunately can't confirm more 50 day's uptime for now, but i didn't got any  issues leading to restart or hanged device.. for now i have 34 days of uptime. 
[image]

PGnl, 2021-06-22:
Ik heb nog 7 dagen te gaan tot 50 dagen.
Ga het eens in de gaten houden.
Draai 1.0-iy-cm-07

IgorYbema, 2021-06-29:
This is a known issue. Something with the internal timers which are overflowing.
Hard to debug if you have to wait 50 days always :)
If someone hits it, please go to 'console' tab in the newer version and copy paste some lines from there to here in the issue.

IgorYbema, 2021-06-29:
> Ik heb nog 7 dagen te gaan tot 50 dagen.
> Ga het eens in de gaten houden.
> Draai 1.0-iy-cm-07

are you at 50 days now?

geduxas, 2021-06-29:
> > Ik heb nog 7 dagen te gaan tot 50 dagen.
> > Ga het eens in de gaten houden.
> > Draai 1.0-iy-cm-07
> 
> are you at 50 days now?

I am on 41day.. but got notification from energy operatior about power line maintenance in 02 of July... Really long term for debugging :D

PGnl, 2021-06-29:
49 days 22 hours 25 minutes 28 seconds.
Keeping an eye on what happens.
When it crashes, I'll see if I can access the access point mode and copy from the console.
Just hoping it is the wifi or something similar. Otherwise there is also no way of accessing it.

IgorYbema, 2021-06-29:
> 49 days 22 hours 25 minutes 28 seconds.
> Keeping an eye on what happens.
> When it crashes, I'll see if I can access the access point mode and copy from the console.
> Just hoping it is the wifi or something similar. Otherwise there is also no way of accessing it.

You can also get serial console output on gpio2 (only need to connect gnd and a rx from a ttl serial connection to gpio2, 115200 baud)

PGnl, 2021-06-29:
Don't have that lying around (regretfully)

IgorYbema, 2021-06-29:
Mine is at 35 days currently so we will see if your output is not clear enough I hope mine will

PGnl, 2021-06-29:
Now at 50 days and 8 minutes. Since about 25 minutes, the site seems sluggish. Sometime freezes temporarily. Then after some time it goes again, couple of minutes later freezes again.

IgorYbema, 2021-06-29:
Can you see the console output and paste some?

PGnl, 2021-06-29:
I can, but there is nothing special in it. 
Main data still fine as well.
Wifi signal: 76%
Memory free: 71%
Correct received data: 99.98%
MQTT reconnects: 12
Uptime: 50 days 0 hours 21 minutes 33 seconds

IgorYbema, 2021-06-29:
Can you enable 'toggle hexdump' ? This will create some more output. Not necessary for debugging this but maybe it helps to show more log lines.
And are the messages repeating each 5 secs?

PGnl, 2021-06-29:
Already on:

Requesting new 1wire temperatures
Heishamon stats: Uptime: 50 days 0 hours 24 minutes 1 second ## Free memory: 75% 27240 bytes ## Wifi: 74% ## Mqtt reconnects: 12 ## Correct data: 99.98%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 53 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
data: 00 00 00 00 00 00 94 94 94 93 AD 71 71 84 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9F 9B 80 8B 9F 9B 80 8B 98 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 55 21 79 15 55 05 24 1A 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 97 00 00 91 90 90 94 94 8F 8F 32 32 32 94 9B 91 90 98 91 91 61 
data: 92 61 95 3E 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 18 
Checksum and header received ok!
Requesting new 1wire temperatures
Heishamon stats: Uptime: 50 days 0 hours 24 minutes 6 seconds ## Free memory: 75% 27240 bytes ## Wifi: 74% ## Mqtt reconnects: 12 ## Correct data: 99.98%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 53 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
data: 00 00 00 00 00 00 94 94 94 93 AD 71 71 84 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9F 9B 80 8B 9F 9B 80 8B 98 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 55 21 79 15 55 05 24 1A 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 97 00 00 91 90 90 94 94 8F 8F 32 32 32 94 9B 91 90 98 91 91 61 
data: 92 61 95 3E 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 18 
Checksum and header received ok!
received TOP0 Heatpump_State: 0
received TOP1 Pump_Flow: 0.13
received TOP2 Force_DHW_State: 0
received TOP3 Quiet_Mode_Schedule: 0
received TOP4 Operating_Mode_State: 1
received TOP5 Main_Inlet_Temp: 16
received TOP6 Main_Outlet_Temp: 16
received TOP7 Main_Target_Temp: 27
received TOP8 Compressor_Freq: 0
received TOP9 DHW_Target_Temp: 45
received TOP10 DHW_Temp: -128
received TOP11 Operations_Hours: 4660
received TOP12 Operations_Counter: 1190
received TOP13 Main_Schedule_State: 0
received TOP14 Outside_Temp: 17
received TOP15 Heat_Energy_Production: 0
received TOP16 Heat_Energy_Consumption: 0
received TOP17 Powerful_Mode_Time: 0
received TOP18 Quiet_Mode_Level: 0
received TOP19 Holiday_Mode_State: 0
received TOP20 ThreeWay_Valve_State: 0
received TOP21 Outside_Pipe_Temp: 17
received TOP22 DHW_Heat_Delta: -8
received TOP23 Heat_Delta: 4
received TOP24 Cool_Delta: 5
received TOP25 DHW_Holiday_Shift_Temp: -15
received TOP26 Defrosting_State: 0
received TOP27 Z1_Heat_Request_Temp: 20
received TOP28 Z1_Cool_Request_Temp: 20
received TOP29 Z1_Heat_Curve_Target_High_Temp: 31
received TOP30 Z1_Heat_Curve_Target_Low_Temp: 27
received TOP31 Z1_Heat_Curve_Outside_High_Temp: 11
received TOP32 Z1_Heat_Curve_Outside_Low_Temp: 0
received TOP33 Room_Thermostat_Temp: 24
[... 3470 lines cut ...]
received TOP42 Z1_Water_Target_Temp: 15
received TOP43 Z2_Water_Target_Temp: 15
received TOP44 Error: No error
received TOP45 Room_Holiday_Shift_Temp: -15
received TOP46 Buffer_Temp: -78
received TOP47 Solar_Temp: -78
received TOP48 Pool_Temp: -78
received TOP49 Main_Hex_Outlet_Temp: 17
received TOP50 Discharge_Temp: 16
received TOP51 Inside_Pipe_Temp: 17
received TOP52 Defrost_Temp: -31
received TOP53 Eva_Outlet_Temp: 18
received TOP54 Bypass_Outlet_Temp: -31
received TOP55 Ipm_Temp: 20
received TOP56 Z1_Temp: 23
received TOP57 Z2_Temp: -128
received TOP58 DHW_Heater_State: 0
received TOP59 Room_Heater_State: 0
received TOP60 Internal_Heater_State: 0
received TOP61 External_Heater_State: 0
received TOP62 Fan1_Motor_Speed: 0
received TOP63 Fan2_Motor_Speed: 0
received TOP64 High_Pressure: 12.2
received TOP65 Pump_Speed: 0
received TOP66 Low_Press

PGnl, 2021-06-29:
Oops, had to erase some stuff before hitting send

IgorYbema, 2021-06-29:
Soo seems to work indeed. How about mqtt send and receive?

PGnl, 2021-06-29:
All still working. 
Now at 50 days 1 hour 48 minutes 20 seconds
Will keep track of it and let you know if anything changes.

PGnl, 2021-06-29:
By the way. The sluggish response on the website now seems to be gone. 
Weird. Never had it before. Was there for over an hour around the 50 daymark, and now seems gone again.

ebogaard, 2021-06-29:
I should have taken some more inventory when this issue happened at my heishamon... Only 43 days before I get that chance again ;-)

As the issue arises at around 50 days, it might be an unisgned integer overflowing:
* unsigned integer max value: 4294967295
* 50 days in milliseconds: 4320000000

That's too close to be a coincidence, I think ;-)

So the original issue In experienced was not in the web interface or wifi connection, but more in the communication with the heatpump: the numbers wouldn't get updated anymore and commands wouldn't reach the heatpump.

IgorYbema, 2021-06-29:
Yes we know that. But there are a lot of timers in heishamon and we don't know which is doing wrong. Most say that mqtt fails and it seems you are describing the same.

PGnl, 2021-07-01:
51 days 12 hours 58 minutes 52 seconds
still going strong

IgorYbema, 2021-07-01:
Thnx. I'm checking code commits if we might have fixed this while fixing something else.

IgorYbema, 2021-07-02:
@PGnl still working? also mqtt?

PGnl, 2021-07-03:
MQQT stopped working, but it seems that the heisha doesn't see it. No additional MQQT reconnects (was at 12, still at 12 after Uptime: 53 days 11 hours 25 minutes 24 seconds)
So somewhere there is no MQQT connect, Heiasha does see it, but seems to not perform a reconnect
Stopped somweher between 51 days and 12 hours and 52 23 hours (had seen it yesterday, but missed the time to respond)
MQQT-Explorer shows no issues and my devices (shelly and tasmota) all still work via MQQT.

PGnl, 2021-07-03:
Requesting new 1wire temperatures
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 26 seconds ## Free memory: 78% 28280 bytes ## Wifi: 68% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 D3 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
data: 00 00 00 00 00 00 94 94 94 93 AD 71 71 84 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9F 9B 80 8B 9F 9B 80 8B 98 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 55 21 79 15 55 07 1B 19 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 95 94 00 91 8F 8F 93 92 8F 8F 32 32 32 94 9B 8F 8E 96 90 8F 61 
data: 91 61 94 3D 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 1F 
Checksum and header received ok!
Requesting new 1wire temperatures
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 31 seconds ## Free memory: 78% 28280 bytes ## Wifi: 68% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 D3 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
data: 00 00 00 00 00 00 94 94 94 93 AD 71 71 84 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9F 9B 80 8B 9F 9B 80 8B 98 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 55 21 79 15 55 07 1B 19 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 95 94 00 91 8F 8F 93 92 8F 8F 32 32 32 94 9B 8F 8E 96 90 8F 61 
data: 91 61 94 3D 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 1F 
Checksum and header received ok!
Requesting new 1wire temperatures
Received 1wire sensor temperature (288587890c000074): 15.69
Measured Watthour on S0 port 2: 0.00
Measured total Watthour on S0 port 2: 0.50
Calculated Watt on S0 port 2: 0
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 36 seconds ## Free memory: 77% 27832 bytes ## Wifi: 68% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 D3 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
data: 00 00 00 00 00 00 94 94 94 93 AD 71 71 84 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9F 9B 80 8B 9F 9B 80 8B 98 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 55 55 21 79 15 55 07 1B 19 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 95 94 00 91 8F 8F 93 92 8F 8F 32 32 32 94 9B 8F 8E 96 90 8F 61 
data: 91 61 94 3D 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 1F 
Checksum and header received ok!
Requesting new 1wire temperatures
Received 1wire sensor temperature (288587890c000074): 15.62
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 41 seconds ## Free memory: 78% 28328 bytes ## Wifi: 68% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 203 bytes data
data: 71 C8 01 10 55 55 D3 49 00 55 00 00 00 00 00 00 00 00 00 00 1A 16 44 55 15 95 55 05 09 00 00 00 
[... 129 lines cut ...]
received TOP104 Solar_Frost_Protection: 5
received TOP105 Solar_High_Limit: 80
Requesting new 1wire temperatures
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 51 seconds ## Free memory: 77% 27872 bytes ## Wifi: 68% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
[... cut ...]
data: 00 E2 CF 0B 13 33 32 D1 0C 16 33 95 94 00 91 8F 8F 93 92 8F 8F 32 32 32 94 9B 8F 8E 96 90 8F 61 
data: 91 61 94 3D 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 A7 04 00 35 12 00 73 00 00 01 00 00 06 
data: 01 01 01 01 01 01 01 00 00 00 1F 
Checksum and header received ok!
Requesting new 1wire temperatures
Lost MQTT connection!
Heishamon stats: Uptime: 53 days 11 hours 25 minutes 56 seconds ## Free memory: 76% 27448 bytes ## Wifi: 70% ## Mqtt reconnects: 12 ## Correct data: 99.97%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18

PGnl, 2021-07-03:
Ik laat hem zo even staan (geen reboot), dus als je wil dat ik iets check. laat maar weten. Pana hoeft de komende dagen toch niets te doen.
Ben na deze ochtend niet thuis dit weekend, dus of vanochtend of na het weekend.

IgorYbema, 2021-07-03:
You can reboot. I found it: 
https://github.com/IgorYbema/HeishaMon/blob/c6157f63103100120b2948211adb7054a38443c8/HeishaMon/HeishaMon.ino#L180
If 'now' is back to zero again (due to roll over) it will never reconnect because the timeout isn't over yet (maybe only after another 50 days).
I already fixed it in my test setup as it seems in the code but It takes another 25 days for my test setup to reach that.

You can reboot yours.

PGnl, 2021-07-03:
Super that you found it, will then reboot.
50 days for a reconnect attemt seems indeed a bit long ;)
Regards,

PGnl, 2021-07-10:
Is there a version available that has the fix?
Visiting me father in law tomorrow and want to update his heishamon.

IgorYbema, 2021-07-10:
This version is what I am testing for a month now and still waiting for the 50 days to confirm the fix. https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v1.0-iy-cm-15.bin

IgorYbema, 2021-07-14:
My first test device with 1.0-iy-cm-14 (confirmed to be bad) also had mqtt issue after 50 days (depends on when it needs to reconnect to mqtt after 50 days so not always exactly at that point).

The test device with 1.0-iy-cm-15 which should have the fix still needs another two weeks for the 50 days test.

IgorYbema, 2021-07-28:
My test device with 1.0-iy-cm-15 is past 50 days and it still works even after reconnects of mqtt! So it seems this bug is resolved!

IgorYbema, 2021-08-25:
@ebogaard can you close this issue if you are happy with the result?

ebogaard, 2021-08-25:
Will do. As I installed the v2.0-version 12 days ago, we need to wait at least 38 days before I can make sure this is fixed.

HighlyCompressedAir, 2024-09-11:
Fixed by https://github.com/Egyras/HeishaMon/issues/214#issuecomment-873364859
This issue can be closed.

closed 2024-09-12 by Egyras