heishamon/HeishaMon #218: MQTT commands not working

trsqr, 2021-09-05:
I just updated my HeishaMon 0.7 to 2.1-iy-6 (and reconfigured everything) and now it seems that any command I send from HomeAssistant to HeishaMon does not work.

When I subscribe to the MQTT topic, I can see that the HeishaMon is publishing its' stuff correctly as expected under homeauto/aquarea and also the HomeAssistant is sending the command SetDHWTemp correctly there.

```
homeauto/aquarea/stats {"uptime":68626175,"voltage":4.06,"free memory":75,"wifi":100,"mqtt reconnects":1,"total reads":4597,"good reads":4597,"bad crc reads":0,"bad header reads":0,"too short reads":0,"too long reads":0,"timeout reads":0}
homeauto/aquarea/LWT Online
homeauto/aquarea/main/Pump_Speed 1500
homeauto/aquarea/stats {"uptime":68641176,"voltage":4.06,"free memory":75,"wifi":100,"mqtt reconnects":1,"total reads":4598,"good reads":4598,"bad crc reads":0,"bad header reads":0,"too short reads":0,"too long reads":0,"timeout reads":0}
homeauto/aquarea/LWT Online
homeauto/aquarea/main/Main_Outlet_Temp 25
homeauto/aquarea/main/Z1_Temp 25
homeauto/aquarea/main/Pump_Speed 1450
homeauto/aquarea/commands/SetDHWTemp 52
homeauto/aquarea/stats {"uptime":68656177,"voltage":4.06,"free memory":74,"wifi":100,"mqtt reconnects":1,"total reads":4599,"good reads":4599,"bad crc reads":0,"bad header reads":0,"too short reads":0,"too long reads":0,"timeout reads":0}
homeauto/aquarea/LWT Online
homeauto/aquarea/main/Main_Inlet_Temp 26
homeauto/aquarea/main/Main_Outlet_Temp 26
homeauto/aquarea/main/Z1_Temp 26
```

However, on the HeishaMon nothing happens and I can see nothing on the console either.

I tried enabling the debug to MQTT, but that doesn't show anything either when I change the DHW temperature:

```
homeauto/aquarea/log Requesting new panasonic data
homeauto/aquarea/log sent bytes: 111 including checksum value: 18
homeauto/aquarea/LWT Online
homeauto/aquarea/log Received 203 bytes data
homeauto/aquarea/log Checksum and header received ok!
homeauto/aquarea/commands/SetDHWTemp 54
homeauto/aquarea/log Heishamon stats: Uptime: 0 days 19 hours 9 minutes 16 seconds ## Free memory: 74% 26496 bytes ## Wifi: 100% (RSSI: -37) ## Mqtt reconnects: 1 ## Correct data: 100.00%
homeauto/aquarea/stats {"uptime":68956386,"voltage":4.06,"free memory":70,"wifi":100,"mqtt reconnects":1,"total reads":4619,"good reads":4619,"bad crc reads":0,"bad header reads":0,"too short reads":0,"too long reads":0,"timeout reads":0}
homeauto/aquarea/log Requesting new panasonic data
homeauto/aquarea/log sent bytes: 111 including checksum value: 18
homeauto/aquarea/LWT Online
```

trsqr, 2021-09-05:
A note here, if I use the API with a request such as http://heishamon.local/command?SetDHWTemp=54 it works just fine and the temperature is changed and I can see in the console a line regarding the temperature change.

sermayoral, 2021-09-05:
Mqtt topics changed in 1.0: https://github.com/Egyras/HeishaMon/blob/master/MQTT-Topics.md

trsqr, 2021-09-05:
Yes, that's true, however, I updated the HomeAssistant configuration files as well.

The page you linked states:

> Command Topics:
> 
> These topics are commands through heishamon to set modes and values on the heatpump and they can be set by either using:
> 
> MQTT: send mqtt message to base_topic/commands/SetTopic (e.g.: panasonic_heat_pump/commands/SetHeatpump)

My base_topic is set to homeauto/aquarea and I'm sending the command to homeauto/aquarea/commands/SetDHWTemp so that should be correct as per my understanding.

trsqr, 2021-09-24:
I just found out that this might be connected to the issue #223 that I opened. If I send a command from HomeAssistant right after the "LWT Online" message it actually does work.

Here two SetZ1HeatRequestTemperature commands, only the second one seems to work:

```
homeauto/aquarea/LWT Offline
homeauto/aquarea/commands/SetZ1HeatRequestTemperature 0
homeauto/aquarea/LWT Online
homeauto/aquarea/ip 192.168.101.112
homeauto/aquarea/log Requesting new panasonic data
homeauto/aquarea/log sent bytes: 111 including checksum value: 18
homeauto/aquarea/LWT Online
homeauto/aquarea/log Received 203 bytes data
homeauto/aquarea/log Checksum and header received ok!
homeauto/aquarea/log Total reads : 13 and total good reads : 13 (100.00 %)
homeauto/aquarea/log received TOP62 Fan1_Motor_Speed: 410
homeauto/aquarea/main/Fan1_Motor_Speed 410
homeauto/aquarea/commands/SetZ1HeatRequestTemperature 1
homeauto/aquarea/log set z1 heat request temperature to 1
homeauto/aquarea/log sent bytes: 111 including checksum value: 17
homeauto/aquarea/log Received 203 bytes data
homeauto/aquarea/log Checksum and header received ok!
homeauto/aquarea/log Total reads : 14 and total good reads : 14 (100.00 %)
homeauto/aquarea/log received TOP1 Pump_Flow: 15.17
homeauto/aquarea/main/Pump_Flow 15.17
homeauto/aquarea/log received TOP8 Compressor_Freq: 21
homeauto/aquarea/main/Compressor_Freq 21
homeauto/aquarea/log received TOP14 Outside_Temp: 9
homeauto/aquarea/main/Outside_Temp 9
homeauto/aquarea/log received TOP62 Fan1_Motor_Speed: 420
homeauto/aquarea/main/Fan1_Motor_Speed 420
homeauto/aquarea/LWT Offline
```

I'm closing this because it seems my true issue is that the MQTT doesn't stay connected.

closed 2021-09-24 by trsqr