heishamon/HeishaMon #219: Energy measurement (kWh)

skortela, 2021-09-08:
README.md has statement "The software is also able to measure Watt on a S0 port of two kWh meters".
What that means? Some external kWh meters are supported? Any more info?

Panasonic original software is able to provide energy usage without external meters. Can HeishaMon?
I need total kWh energy used via mqtt.

trsqr, 2021-09-08:
Moro Sami,

You can get the same energy usage out from HeishaMon as you can get from the original Panasonic unit. The S0 connection is for an external meter that has S0 pulse output. That can be used to monitor the heat pump or any thing else.

MiG-41, 2021-09-09:
Just to be clear , Energy usage is not given by HP. HP is given only actual Power usage (and also with accuracy of 0,2kW , and with voltage assumption , generally it is not good enough ,and it is lies , sometimes more sometimes less. 
From Power ou can calculate Energy ( generally calculating avarage Power and multipling this by time) , and that what the driver with display  do , and what the Cloud do ( they do it separatly and independently from each other , that is why sometimes Energy from them are vary). 
HeishaMon is not calculating Energy from Power given by HP at all. It can only calculate Power and Energy from exernal energy meter ,with have so called S0 connection ( impulse every particular Wh).

trsqr, 2021-09-09:
Oh yes, you're correct. I'm counting the energy usage per day in Grafana on my system. I have an external M-Bus capable energy meter in front of my heat pump as well as energy meter connected to the inlet and outlet water pipes. The figures match quite well:
[image]

bse4792, 2021-11-21:
I'm still confused about where to connect S0 is this on the board on HP or?

As far as I know the HP has energy and power meter build in, so why is this data not available via TOP?

geduxas, 2021-11-21:
S0 is only for external power meter. Panasonic have it's own power calculation.l but often it's fare from accurate..

closed 2024-09-12 by Egyras