heishamon/HeishaMon #221: problems with version 1.0 and 2.0

bse4792, 2021-09-23:
Hi, i'm facing some problems with data/connection to mqtt server

i have tried to upgrade the firmware to both 1.0 and 2.0, but i always had to revert back to 0.9, as many data does not get send to mqtt and also loses connection to wifi

also "Correct received data" never reaches 100% as on 0.9

one thing i found was that model # was not shown in 2.0, but it's looking nice in 0.9

i dont know if reset would fix the issue or not, but could you please look into this?

MiG-41, 2021-09-23:
For me it is the same topic: https://github.com/Egyras/HeishaMon/issues/196
What hardware you use ? Perhaps ou can easly "erase flash" and share if that helps.

bse4792, 2021-10-06:
Hi Hardware is made by @MTrab, i think its a WeMos D1, he has done the programming, and i'm not that familiar with this kind of work, but i have flashed D1's before

mi-hol, 2024-09-06:
@Egyras guess its time to close this issue

closed 2024-09-12 by Egyras