heishamon/HeishaMon #222: Add possibility to send commands from the web interface

trsqr, 2021-09-24:
The web interface provides a nice list of all topics as well as the console. How about enhancing that with the possibility to send commands to the heat pump as well?

MiG-41, 2021-09-24:
@trsqr , you can ask ( or dream) for a lot of posibilities , starting from sendin commands , drawing charts with analises ect :) 
But orginally this was the automation's task , next thing ESP8266 has limited power ( for sure for steering is enough , but anyay) , and finally  limited resources of progrmmers :)

trsqr, 2021-09-24:
Yes, I know. :)

But we can already send the commands by doing a call to an URL such as:
http://heishamon.local/command?SetDHWTemp=54

Basically what I'm suggesting is one more page in the web UI (or a modification of the current UI) where there would be a box for entering that number and a button to make the call to that URL. Should not be very power intensive I think.

IgorYbema, 2021-09-24:
The biggest cullprit is designing a user friendly interface. Domoticz, home assistant etc are already having wonderfull intefaces and creating another interface on the heishamon itself would require good knowledge about designing a user interface.
If a frontend webdeveloper wants to step in, be my guest. The backend is already done for it with the json output and api inputs.

closed 2024-09-12 by Egyras