heishamon/HeishaMon #224: Feature request: Provide signal for deicing active

trsqr, 2021-09-26:
The J-generation pumps have support for providing a signal when the deicing is active. This is handy if you have to implement extra heating cables for keeping the drain pipes unfrozen. With the deicing signal it's possible to heat the drain pipes only during and a bit after when deicing is active instead of keeping the heating on constantly when it's cold.

H-generation pumps do not have this. Would it be possible to implement such functionality in HeishaMon? For example, if one of the S0 interfaces is not used, maybe one could be configured for this purpose?

PGnl, 2021-09-26:
Why include it. If you notice a deicing, use a smart switch via domoticz/HA or any other to activate the heating element/cables. 
This can easily be scripted. Myself would prefer not to include it in heisha. That is what HA/Domoticz is for.

MiG-41, 2021-09-26:
@trsqr in regards of heating cable ,in booth H an J series there is a function Base Pan Heater , and corelated with this connector on Outdoor unit PCB , with switches 230V acc to function Type A or B.
Next thing ,as @PGnl mention , there is a TOP26 Defrosting State , with can be easly inplement for devices as Sonoff or Shelly 1 to turn on/off.

trsqr, 2021-09-26:
@MiG-41 I'm aware of the base pan heater and I have one in my unit (it's more or less mandatory here in the north). However, to lead out the water from the unit to the sewer that's well below the ground (frost depth is something like 1.2 meters here) I need to use a pipe that has some heating - otherwise it freezes. Panasonic specifically says not to connect anything more than the base pan heater to the connector (risk of pulling too much current). In J generation units the Boiler Contact serves also as Deice Signal (if configured in the menus), in H generation this does not exist (I own both a H and J gen units).

J-gen install manual:
[image]

I know that I could implement that using a wifi socket and HA. The issue is that then my system will require that HeishaMon is up and running, my home server is up and running, that the WiFi actually works and that the WiFi socket is up and running. 4 devices and failure of any of them will result in no defrosting. If HeishaMon would provide the signal to a simple relay/contactor I would only have 2 devices and the relays I consider very reliable.

I know that another option is to feed the relay from the base pan heater connector inside the outside unit, but I'd prefer not to touch the insides of the machine while it's still under warranty...

PGnl, 2021-09-26:
Connecting a relay to the base pan heater connector should not void the warranty.
That would be far cleaner than the heisha doing it. Then only one failure point.
Much cleaner.

closed 2023-11-28 by Egyras