heishamon/HeishaMon #225: Question: Knowledge about which parameters end up in lifetime-limited non-volatile storage in heatpump?

NerdyProjects, 2021-10-02:
Hey there,

thanks a lot for HeishaMon! I am considering using this or at least the decoded protocol information for my own aquarea automation/monitoring solution.

Thinking about automation, especially maximising solar usage, I wonder which parameters are safe to be controlled every few minutes?

A lot of the parameters should be stored in non-volatile storage in the heatpump (e.g. EEPROM or flash) that has limited write cycles (for example 10000 to 100000 for an EEPROM) - If I do one write every hour, the eeprom cell would be broken after a bit more than 10 years (for 100000 writes - if it only lasts 10k, it's just over 1 year).

I am wondering especially about everything that changes heating curves (as is done in the "smart controller" every few minutes) or temperature offsets - I wonder if not the SG signals should be used for that instead.

I guess there is also no long-term experience, e.g. somebody who already did more than 10000 writes to any of those parameters?

IgorYbema, 2021-10-03:
I guess no one can give you an answer about this. Interesting question. Would make a special support case at panasonic as long you have warranty :)

michalwoz, 2021-10-03:
@NerdyProjects I'm at the same point as you are. Have you considered using demand control? Originally it is 0-10V, but everything says it can be emulated. I think that it should be designed to allow changing demand with not limited frequency. At least from potential flash/EEPROM point of view. Of course changing demand 0%,100%,0%,100% etc, doesn't make any sense and will be very bad for compressor, but I can imagine some algorithm which makes +5%,-3%, +2% adjustments every 15 minutes.

MiG-41, 2021-10-04:
@michalwoz ,Demand Control only sounds good , but in reality you will be not able to conrol anything . It work in special way , what we find out ,it has influence on Compressor current. In my case minimum set 5% limiting it to 3A. But 3A it is a lot in normal conditions ( let say around 0) , something like 40Hz in my case.  In many situations you will find , that this Demand Control do nothing....  Limiting Demand Control to 5% behaving more/less like to set Quiet Mode 3.

NerdyProjects, 2021-10-04:
In addition to demand control there is the SG signals. As I don't have the heatpump yet, I don't know exactly how they will behave and how quick the heatpump will react - I understood it so you can set two temperature factors (SG1 and SG2) from 0.5 .. 1.5 that will be multiplied with the currently requested temperatures. This _should_ be enough to minimize/maximize power consumption, no? In my opinion, this should be a good enough start to e.g. maximize heating power from lunch till afternoon while there is enough power or minimize power in summer for hot water operation while there is peaks of other demand.

Maybe even the (limited, I know) demand control is good enough for that. Very nice that the parameter set is known to heishamon now. I wonder, if I should cancel my optional PCB/buffer sensor order as I think it is much more useful to do it with the optional PCB emulation :-)

MiG-41, 2021-10-04:
Correct ,SG works in this way. But having Heishamon it is easier to set temperatures in direct way , not puting signals and let the HP to do the mathematic recalculation . 
Ofcourse writting a lot of changing temperatures can influence on eeprom , if you do it to often...

NerdyProjects, 2021-10-04:
Well, you can set SG with the optional pcb emulation, so it stays all digital :-)

NerdyProjects, 2021-10-11:
Although this is not an issue, I think it is something users should be aware about. Would be sad to have the heatpump fail due to irresponsible writes by some home automation software... Where would it fit better? I guess by default only everything that happens through the optional PCB can be seen as "safe"...

CurlyMoo, 2021-12-11:
I don't quite see what controller values should be stored in an EEPROM or Flash. The statistics of the device itself maybe, and the settings are probably stored in an EEPROM or Flash, but why should the controlling parameters be stored there?

There is absolutely no use for it. Every decision of the heatpump is based on a sensor measurement. These sensor values can be measured in realtime when the device turns back on after being shutdown. The same counts for all decisions based on those measurements. So, i really don't think there is an issue here.

MiG-41, 2021-12-14:
By Set Commands users can change parameters like:
-Max Water pump flow
-Parameters each of 4 Heating and Cooling Curves
- Mode 
- Temperatures for zones 
- temperature for DHW
- And generally all parameters with can be presset/changes.
Personally i hae automaions for two first parameters , with automation sets when some conitions apear.
Having in mind limit of operations for EEPROM , it is good to amount of writes  would be not higher then 50 per day ( HP for lifetime 20 years). 
That is why ,from one side good to be responsible and know ,that there is a limit in set commands , from other it woul be greatto find out how this parameters are being storaged ( becouse they are no changed after energy failure).
Maybe Panasonic have protection for it ,and there is nothing to worry , but how to be sure ?

Celphor, 2023-12-09:
I once noticed a strange behavior, when operating cycles and hours went back to older values. I don't rememer what I've done, maybe a reset because of frequent cyling. So it seems not every set value is written to the eeprom immediately. Maybe only after a few minutes from RAM to EEPROM. Look at the graph 
[image]

MiG-41, 2023-12-09:
...or first sympthoms of broken EEPROM ...

Celphor, 2023-12-09:
I don't think so, this happened 2 months ago and never showed up again

Celphor, 2023-12-09:
Does anybody know the model of the used flash chip? On my indoor unit i can't locate one on the frontside. Don't know if the board is double sided. If the model is known the spec should provide info about number of writes per cell.

MiG-41, 2023-12-09:
Had a picture of my PCB on back side ( yes it is doublesided) , main CPU and surrounded chip (could be EEPROM) is wihout markings , and also glued with some siliconlike thing.

Celphor, 2023-12-09:
If it really fails: The boards are not that expensive, <400€. Maybe I should keep one in stock ^^

edterbak, 2024-09-03:
I just read an issue report that my node red flow will in fact as a certainty kill the eeprom on the panasonics...   _(nice report)_

I have developed the node red flow, and in the beginning I probably flooded my pump with set commands. (Imagine 1 cmd per minute for weeks)

Is there a way to get some sort of reading on the health state of the EEPROM? 

Typically the node red flow produces the following amount of set commands per day: 
low: 0-10 
normal: 20-40
high: 80-150*
*The high scenario only applies when a user has enabled all my custom functions and have set it up that it reacts a bit nervous / jumpy.

FYI. 
I personally do not have an issue so far. I have no indication of a problem yet.
My pump is installed in jan-2021.
If I were to make an estimate on how many commands have been sent to my pump, I would guess 250k+ at least.

mi-hol, 2024-09-06:
> I just read an issue report that my node red flow will in fact as a certainty kill the eeprom on the panasonics... _(nice report)_

@edterbak are you referring to https://github.com/edterbak/NodeRed_Heishamon_control/issues/169 ?

From my view this specific issue fails to provide any hard facts and just makes claims with no evidence!

HighlyCompressedAir, 2024-09-07:
If every command is written to EEPROM immediately as the heat pump's behavior suggests (if you change a parameter and turn off the breaker, the change will be there after the power is back on), then it will break the chip eventually, it's just a matter of when. Most EEPROMs can take 100k to 1kk writes and these are hard facts.
So far nobody proved that Panasonic heat pumps work differently, therefore in my opinion we should assume the worst case scenario.
I think that number one priority for these projects should be discovering how it actually works. The fact that this issue is unresolved for years now seems dangerous to me, especially if users are unaware.

edterbak, 2024-09-08:
> "The fact that this issue is unresolved for years now seems dangerous to me, especially if users are unaware."

Maybe reverse psychology is applicable here.
The longer this 'issue' keeps being unresolved, the more it seems to prove it is not a real issue.

If anyone has more data supporting the fact this is a real issue and we all agree on it that it is true... I will instantly take the node red flow offline.

HighlyCompressedAir, 2024-09-08:
Obviously it's an issue that won't manifest itself for years until it's too late. I don't want you to take your project down, just warn users of possible consquences and try to minimize writes as much as possible with every feature until someone figures out how things work.
Panasonic users aren't the first who thought of this problem. Here's a thread from 2019 regarding Nibe heat pumps. They actually got an answer from Nibe representative:
`The EEPROM settings are stored has, according to the datasheet, a limit of 1 million writes per position. That may of course sound a lot but if you write every second that means only 11.5 days until the EEPROM position dies. So to answer the question, when calculating with 15 years of life the maximum would be 182 writes per day per setting`
And Nibe officialy supports third party control via Modbus, Heishamon had to be reverse-engineered. Do you think that Panasonic engineered their pump to take 100+ commands a day when there's not even an interface for it? I doubt that.

mi-hol, 2024-09-08:
> hey actually got an answer from Nibe representative:

this is a real fact

> Do you think that Panasonic engineered their pump to take 100+ commands a day when there's not even an interface for it? I doubt that.

Panasonic is using readily available EEPROM components, therefore above conclusion is pure speculation and not relevant from my view!

IgorYbema, 2024-09-08:
Please stop the discussion. The EEPROM issues could be a thing and as long as we don't know what the limit is, it is better not to overload with commands. 

If anyone wants to try which settings are written to EEPROM it is an easy test. Just send a command, then power cycle the heatpump 10 seconds later and check if the setting is still changed.

And if anyone wants to take a photo of the backside of the mainboard to see what MCU and EEPROM it is using we can take a rough guess.

IgorYbema, 2024-09-08:
https://github.com/IgorYbema/HeishaMon/blob/main/README.md#eeprom-warning

edterbak, 2024-09-08:
> Please stop the discussion. The EEPROM issues could be a thing and as long as we don't know what the limit is, it is better not to overload with commands.
>
> If anyone wants to try which settings are [...]

What do you considder to be a power cycle?
off/on on the wallmounted controller? How much time between off/on?

I can test in upcoming week on a couple of settings...

IgorYbema, 2024-09-08:
> What do you considder to be a power cycle? off/on on the wallmounted controller? How much time between off/on?
>
> I can test in upcoming week on a couple of settings...

This is also unknown :-)
But at least try a full power shutdown (breakers off).
Try SetQuietMode, SetDHWTemp, SetOperationMode first. I believe at least one of them should return to the set value after power cycle.

mi-hol, 2024-09-08:
> Please stop the discussion.

May I suggest an alternative approach to turn this into a non-issue by knowing "how to replace a broken EEPROM"?

HighlyCompressedAir, 2024-09-08:
> May I suggest an alternative approach to turn this into a non-issue by knowing "how to replace a broken EEPROM"?
It's not that simple, because EEPROM most likely stores the whole operating system. You would have to read it first when it still works properly and make a copy of it's contents, then buy the exact same memory chip model, flash it with previously saved data, desolder the broken module and solder the new one.

Afaik every parameter that can be set survives power outage (SetPump may be an exception, but I'm not sure), so only fake sensors make 0 writes (optional pcb emulation).

mi-hol, 2024-09-08:
> It's not that simple, because EEPROM most likely stores the whole operating system. You would have to read it first when it still works properly and make a copy of it's contents, then buy the exact [...]

This outlines basically the process I had in mind.
Next step would be to sort out the details to make this an actionable plan.

1. is a working Panasonic board already available to someone? 
2. is manufacturer & part number of EEPROM readable on this board?
3. can EEPROM content cloned by a tool (i.e. https://fiedka.app)

MiG-41, 2024-09-09:
Yes , had somwhere. No markings on chips , and covered with some layer of silicone ( like in automotive applications).

HighlyCompressedAir, 2024-09-09:
What about the remote controller? It definitely has a memory of some kind (battery powered presumably) as it remembers date and time, weekly timer, holiday timer and the pump itself has no way to access those (Smart Cloud has it's own online weekly timer and you can enable holiday mode, but can't set it's date and time). And when you select "System check" there's a prompt saying "Retrieving data..." for a few seconds. Maybe it's because the controller has to read the logs off the pump's EEPROM and it takes some time, while all the other parameters can be seen immediately, because they're stored in the controller itself so there is no delay. If someone has access to two similar pumps they can swap controllers and see what happens to settings. Or see if it's possible to change a parameter through Heishamon with the controller disconnected (if error H76 won't happen immediately). Or disassemble the controller and look for a battery, remove it for some time and see what happens.

mi-hol, 2024-09-09:
> If it really fails: The boards are not that expensive, <400€. Maybe I should keep one in stock ^^

@Celphor did you perhaps buy a spare?

mi-hol, 2024-09-09:
> Yes , had somwhere.

@MiG-41 would you mind to post them here?

NerdyProjects, 2024-09-09:
> What do you considder to be a power cycle? off/on on the wallmounted controller? How much time between off/on?
>
> I can test in upcoming week on a couple of settings...

It could totally be implemented "properly": Settings are kept in RAM and are only written to EEPROM when power is removed. If you as an engineer see any need to limit the number of EEPROM writes, you would implement it like this. You just need a power gone signal and 10-20ms of time remaining (e.g. through capacitors keeping the MCU and EEPROM alive).
What I want to say: There is not many reliable ways to test this out.
One could be doing writes we expect to hit the EEPROM millions of times to see what happens.
Another one could be wiring the EEPROM up to some circuit and watching writes while changing settings.

geduxas, 2024-09-09:
> What about the remote controller? It definitely has a memory of some kind (battery powered presumably) as it remembers date and time, weekly timer, holiday timer and the pump itself has no way to [...]

Yes, remote controler works as another gateway, it retrieves stored data from heatpump. Also for "Scheduler calendar" cloud and remote controller have it's own logic by which send setting to pump. There is only possibility to turn on/off schedule indicator, to let user know that pump is operating by scheduled task's.

closed 2024-09-12 by Egyras

edterbak, 2024-09-21:
Finally had time to do some testing. This is the result

[image]

IgorYbema, 2024-09-21:
So yeah, these will have been written to eeprom somehow. Could be smart (only writing when power is down, using some rest over condensator power) but probably not.
Should I builtin a limit of about 100 commands per day (which will be way enough for normal practice and will probably never cause any eeprom issue)? And ofcourse an option to change/disable this limit.

HighlyCompressedAir, 2024-09-21:
> So yeah, these will have been written to eeprom somehow. Could be smart (only writing when power is down, using some rest over condensator power) but probably not. Should I builtin a limit of about [...]

I think a warning is enough.
But I remembered something, maybe you can explain it. I have set Pool Temp to 27 a few days ago when I tested optional pcb commands. In the mean time I powered off my pump a few times and the value is still 27. When I connect just the CZ-TAW1, it reads -78. And then when I disconnect it and connect Heishamon again, the reading goes back to 27 which suggests that it is stored somewhere in Heishamon's memory. How is that possible? After an hour without power?

geduxas, 2024-09-21:
@HighlyCompressedAir it's depends how you set commands to heishamon, if it's over mqtt it could be that it has retained flag (which stores value in mqtt server) 
Also heishamon stores it while it's running.. if you somehow not disconnected it from power supply, it could be left in memory

geduxas, 2024-09-21:
> So yeah, these will have been written to eeprom somehow. Could be smart (only writing when power is down, using some rest over condensator power) but probably not. Should I builtin a limit of about [...]

How about other communication methods? It has knx and Modbus interface, and didn't see anywhere about limit's in documentation, or naturally from them will not be any flood?

edterbak, 2024-09-21:
Yeah, review the method how I did this might be wise. 
Is the result trustworthy.

I do see retain flag when I open MQTT explorer on these values. So I dont know what happens when power is restored to the pump. Does Heishamon do anything at that moment, or the broker.. Truly I dont know.

HighlyCompressedAir, 2024-09-21:
With mqtt server powered off it's the same thing. I just switched the breaker to power off my pump, waited 5 minutes and powered it up again - Pool Temp is still 27. Heishamon doesn't have external power connected so it has to hold charge somehow.

geduxas, 2024-09-21:
> With mqtt server powered off it's the same thing. I just switched the breaker to power off my pump, waited 5 minutes and powered it up again - Pool Temp is still 27. Heishamon doesn't have external power connected so it has to hold charge somehow.

Indeed optional PCB stores it's data once in 5min. I have missed that :)

edterbak, 2024-09-21:
> So yeah, these will have been written to eeprom somehow. Could be smart (only writing when power is down, using some rest over condensator power) but probably not. Should I builtin a limit of about [...]

In the node red dashboard, I have a setting like this:
[image]

In the code, I have the default limit to 500 a day. 
This is only reached in case of an programming error. looping stuff. Then the failsafe kicks in.

Maybe you can use this as inspiration

geduxas, 2024-10-26:
Ik we have one user who have cascade controller with Modbus, and he claims that cascade controller updates/changes registers every 20-30 sec. So if it works in official setup, maybe there is not so much problems?

edterbak, 2024-11-02:
> Ik we have one user who have cascade controller with Modbus, and he claims that cascade controller updates/changes registers every 20-30 sec. So if it works in official setup, maybe there is not so much problems?

I hope this is true. That would take away our worry and we can release any restrictions. :) 
For now we dont know for sure yet. So caution should apply of course.

geduxas, 2024-11-02:
https://github.com/Egyras/HeishaMon/discussions/565#discussioncomment-11062501