heishamon/HeishaMon #226: Home Assistant integration outdated

DamianBroj, 2021-10-09:
Hi, there was many things added to heishamon recently and the Home Assistant integration is still not updated (last update from 2020). Is it possible to add all the missing data and functionalities there?

geduxas, 2021-10-18:
Hi, what features are you missing?

johansmitsnl, 2021-10-19:
I'm getting these errors:

```
2021-10-19 15:44:43 ERROR (MainThread) [homeassistant.components.automation.set_heatshift_selector] Set heatshift selector: Error executing script. Invalid data for call_service at pos 1: Invalid value for input_number.heishamon_heatshift: 20.0 (range -5.0 - 5.0)
2021-10-19 15:44:43 ERROR (MainThread) [homeassistant.components.automation.set_heatshift_selector] Error while executing automation automation.set_heatshift_selector: Invalid value for input_number.heishamon_heatshift: 20.0 (range -5.0 - 5.0)
```

and

```
2021-10-13 16:23:56 WARNING (MainThread) [homeassistant.helpers.template] Template warning: 'float' got invalid input 'unknown' when rendering template '{%- if states('sensor.heishamon_w_production') != "Unknown" -%}
  {%- if states('sensor.heishamon_w_consumption') > "0" -%}
    {{ '%0.1f' % ((states('sensor.heishamon_w_production') | float ) / (states('sensor.heishamon_w_consumption') | float)) }}
  {%- else -%}
    0.0
  {%- endif -%}
{%- else -%}
  n/a
{%- endif -%}' but no default was specified. Currently 'float' will return '0', however this template will fail to render in Home Assistant core 2021.12
2021-10-13 16:23:56 WARNING (MainThread) [homeassistant.helpers.template] Template warning: 'float' got invalid input 'unknown' when rendering template '{%- if states('sensor.heishamon_w_production') != "Unknown" -%}
  {%- if states('sensor.heishamon_w_consumption') > "0" -%}
    {{ '%0.1f' % ((states('sensor.heishamon_w_production') | float ) / (states('sensor.heishamon_w_consumption') | float)) }}
  {%- else -%}
    0.0
  {%- endif -%}
{%- else -%}
  n/a
{%- endif -%}' but no default was specified. Currently 'float' will return '0', however this template will fail to render in Home Assistant core 2021.12
2021-10-13 16:23:56 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ZeroDivisionError: division by zero') while processing template 'Template("{%- if states('sensor.heishamon_w_production') != "Unknown" -%}
  {%- if states('sensor.heishamon_w_consumption') > "0" -%}
    {{ '%0.1f' % ((states('sensor.heishamon_w_production') | float ) / (states('sensor.heishamon_w_consumption') | float)) }}
  {%- else -%}
    0.0
  {%- endif -%}
{%- else -%}
  n/a
{%- endif -%}")' for attribute '_attr_native_value' in entity 'sensor.heishamon_cop'

```

DamianBroj, 2021-10-21:
> I'm getting these errors:
>
> ```
> 2021-10-19 15:44:43 ERROR (MainThread) [homeassistant.components.automation.set_heatshift_selector] Set heatshift selector: Error executing script. Invalid data for [...]
First part for sure is related with setting direct temp vs curve and the acceptable values by heat pump in certain modes, as mentioned in the second part of my previous reply

johansmitsnl, 2021-12-17:
Isn't it that the value when running in room temperature mode that the value is not a difference but the desired/set room temperature?

closed 2024-09-12 by Egyras