heishamon/HeishaMon #230: Optional PCB commands not working:  optional PCB is physically installed and activated

Donpepe23, 2021-11-02:
Hi,

I was playing around with the 0-10V demand control, that became quite limited for the J-series (max limitation to apprx 1500W). So, my next step was to try HeishaMon to set the demand control (emulation on and off, optional board activated in the RC), but w/o **any** success. The heat pump is basically ignoring the commands, even within a limitation range that works with the 0-10V interface. 

Are the HeishaMon optional PCB commands only work w/o the optional PCB installed (only emulated)? 

Thanks

Donpepe23, 2021-11-02:
Ok... will rewire tmr and check if the demand controll works better /different form 0-10V interface. 

Would have been a nice backup to have the optional PCB installed and using Heishamon at the same time, in case heishamon loses conection... 

Anyways, thanks.

gregor-samosir, 2021-11-07:
please close this issue.

closed 2021-11-07 by Donpepe23