heishamon/HeishaMon #233: HexDump or Rawdata via MQTT / ??: SendRawValue

Donpepe23, 2021-11-19:
Hi,

tried to get the hexdump via MQTT (panasonic_heat_pump/log) using "**Debug log hexdump enable from start**" setting, but I only get a string saying "parsed json". Only if I addtionally activate "**Debug log to MQTT topic from start**" I get the full log via MQTT, which would work, but it would be really good to only get the heydump via MQTT...

BTW, what porpose has the topic "panasonic_heat_pump/SendRawValue"

Thanks

MiG-41, 2021-11-19:
I belive debug initially goes to serial on GPIO2 ,and that is destination of first setting. Second redirect log also to MQTT.

SendRawValue command is to pass to HP full string with command from mqtt . Generally for testing and advanced users.

Donpepe23, 2021-11-19:
Hi,

what I am looking for is the hexdump as shown in the console output. Reason is, that the water temperatur is only shown rounded, but in service cloud you see at lease: xx,00 / xx,25 / xx,50 / xx,75 °C.  I want to find out, which bytes are the onse definen the value after the decimal point.

Greetings

MiG-41, 2021-11-19:
We belive , that Service cloud predict/ gess/ assume these temperatures. We were not able to find them. And also in controler's menu's ,they are as intigers.  But of course you can try , good luck :)

With receiving only hex answers fom HP , without additional logs without changing the program not possible.
What you could do is to connect parallel to Heishamon RX line  with TTL-USB converter , and use it as a sniffer.

IgorYbema, 2021-11-23:
Indeed, as @MiG-41 says the service cloud just takes an average over multiple messages and calculates the decimals from there.
Panasonic will send 12 messages each minute and if a decimal is shown these 12 message are flipping between two whole numbers.
Easily done within your own tools also. For example in grafana just take the mean value of the time window instead of the last value.

Donpepe23, 2021-11-23:
@MiG-41 , @IgorYbema  Thanks for the feedback...  

It is not about displaying the data.. more about temp. shift function(-5 / +5) / target temp / switching off and restarting...  and just because this stupid power limitation function is not working properly for the J-Series anymore... 

But with this feedback I know what to do... Thanks

closed 2021-11-23 by Donpepe23