heishamon/HeishaMon #234: COP - effective

gilder0, 2021-12-05:
I would like to calculate the effective COP considering that the pump in defrost mode steals energy from the house. According to for me, such energy taken from the house should be included in the "minus" of what was previously produced.

The problem that I have noticed is that the defrost status includes both the time when the pump is still "producing" energy towards the house, but is already slowing down (about 1 minute), and after complete defrosting, when the temperatures in outlet pipe return to the state before start of defrost (also about 1.5-2 minutes).

Do you have any idea how to determine this real period of energy stealing from the house?

MiG-41, 2021-12-06:
Time , when delta ( outlet - inlet temperature)  is on minus ?

gilder0, 2021-12-06:
After looking on many factor I realized that the simplest solution is based on defrost status and fan speed

so my proposal to get Real Defrost state period shold look like :

    - platform: template
      heishamon_real_defrost:
        friendly_name: "Aquarea Real Defrost"
        value_template: >-
          {%- if is_state('sensor.aquarea_defrost_state','Active') and is_state('sensor.aquarea_fan_1_speed','0') -%}
            Active
          {%- else -%}
            Inactive
          {%- endif -%}

MiG-41, 2021-12-06:
But delta is natural indicator , when is positive HP is generating heat . when negative - consuming , and 0  then is turned off :)
Ofcourse maybe there is needd more precise temp sensors ( like paired ds18b20 conneced to Heishamon) , since these inside HP shows only intiger values.

gilder0, 2021-12-06:
yes you are right, fans speed and crossing of temperatures are in the same time, but i think it is much easier to use fan speed 0 as indicator connected to defrost status

rodaman, 2021-12-08:
I am writing here a question about COP but different. How to get average COP daily, monthly? It is difficult for me figure out average becose value is sometimes  '0" or very high while defrost.

gilder0, 2021-12-08:
COP = Energy Produced / Energy Consumed.

If you would like to calculate average COP for longer period you need have to SUM up above values over period.

As a first step you need to use : https://www.home-assistant.io/integrations/integration/ for each value separately.

Then you need to use https://www.home-assistant.io/integrations/utility_meter/ for each value and period separately.

After that create template sensors to make appropriate calculation of COP for a period.

This way you will be able to see the COP value for the current hour / day / month / year, but I my opinion will not be able to view the past values of these ratios. At least so far I have not found appropriate components that would be able to present calculations based on long-term statistics. 
Long-term statistics are a relatively new feature and maybe in future such components will show up.

rodaman, 2021-12-08:
I have tried that but I want to not include  "0" value and false very high  COP while defrosting. I am very bad at templating in HA. Any suggestions welcome.

Donpepe23, 2022-07-01:
Hi,

considering that water temps are correct, you can calculate the actual output: Delta water temp * 1,16 * flow rate (per h).

If this goes negativ, the heat pump takes energy from the house...  You just need a energy meter that sums it up...

Only problem: the water temps are quite inaccurate... flow rate is OK.

closed 2024-09-12 by Egyras