heishamon/HeishaMon #235: Wifi password field is too short.

Riemen, 2021-12-08:
Hello,

the wifi password field in the Heishamon input mask allows only input of 25 characters. A WPA-PSK/WPA2-PSK password may be 63 characters long.

My password was longer. It took me some time, to realize that the password was cutted while pasting into the field.

Now i temporary had to change my wifi password just for heishamon.

Please change the length of the Password field.

Best regards
Riemen

MiG-41, 2021-12-08:
Witch version you are using ? 
I remember there was modifications , to allowing to put standard generated HA Mqtt broker , with has 64 characters....

Riemen, 2021-12-08:
If I paste a password longer than 25 chars, the input field shows only 25 of these **** chars.
If the Password is shorter, the number of **** matches the number of entered chars.

My password was only alphanumeric and 63 chars long. No exotic special chars which could could have caused problems. But Heishamon refused to connect to my wifi.
With a shorter password, it connected without any problems.

trsqr, 2021-12-08:
in HeishaMon/webfunctions.h the size of the field is limited to 40:
  char wifi_password[40] = "";

MiG-41, 2021-12-08:
Ok , wifi password  , not mqtt password  , bad understood. 
So yes , till now that was enough . No one remember longer passwords :)

Riemen, 2021-12-08:
> in HeishaMon/webfunctions.h the size of the field is limited to 40: char wifi_password[40] = "";

I don't know how long it is internally. I only could see the *** chars are limited to 25 as described above.

But the problem still remains. Why limit the password length to 40 chars, if the specification allows 63?. This makes no sense to me.

trsqr, 2021-12-08:
Fairly sure no-one checked the actual max length specification allows. It's an easy fix to change from 40 to 63 though.

I can enter more than 25 characters in the Web UI though. You can check it easily using the web console of your browser to see what's actually submitted.

Riemen, 2021-12-18:
I Just changed my wireless password again. Didn't want to run my wifi with a ahort pw until this issue is fixed, so I tried to at least use a 40 chars pw.
This didn't work.
I tried 40 chars and 39 chars length. Both was too long. 30 chars length does work.

So it seems, even if the input field is 40 chars long, it seems the pw subsequently get's cut to a length somewhere between 30 and 38 chars.

IgorYbema, 2022-01-02:
I updated the code to fix this https://github.com/IgorYbema/HeishaMon/commit/3d3cabdaee3c175cfa7be94ffcb6abe007f638ce

will be in a future release

closed 2024-09-12 by Egyras