heishamon/HeishaMon #238: How much correct received data should i expect?

kr0815, 2021-12-11:
I installed a WH-MDC07J3E5 the last days - now trying to get heishamon working

I received some values the last days, but correct received data was always low - about 2-4%

Today i played with some settings in the heat pump - now heishamon stopped working completely

Should the correct received data be in a range from 1-5% or in a range of 98 - 100%?
Did Panasonic change something in the protocol

I use a premade PCB from TheHog

This is what i get:

`Received 194 bytes data
Received bad header. Ignoring this data!
Received more data than header suggests! Ignoring this as this is bad data.
data: 71 71 97 99 00 00 80 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 D0 7B F8 7E 1F 1F 79 79 8D 8D B0 
data: A3 7B BE DD FD F7 95 85 80 8F C5 94 9E 8F 8A 94 9E 85 8F 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 
data: 00 B9 E5 95 21 87 15 59 05 23 12 65 00 00 00 00 00 00 00 00 C2 D3 0B 34 65 B2 D3 0B 95 65 AD 00 
data: 00 83 A8 AD 32 32 B0 A3 32 32 32 B0 B0 AD D3 96 A8 7A 61 7C 37 9A 96 01 3B 44 00 00 31 19 93 97 
data: 48 01 3C 79 01 01 2C 00 00 15 00 00 01 00 
Received bad header. Ignoring this data!
data: 00 
Received bad header. Ignoring this data`

trsqr, 2021-12-11:
Should be close to 100%. I've got a J-gen 5kW split unit here and "Correct received data: 99.83%".

kr0815, 2021-12-11:
It is totally strange - now i received this, so wiring is correct

I can´t imagine that at 9600Baud running the data cable next to the power cable for 25 centimeters makes such problems?

`received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 55 00 01 00 00 00 00 00 00 00 00 59 15 11 55 55 15 55 55 56 19 00 00 
data: 00 00 00 00 00 00 B0 80 A3 80 B4 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B0 A3 7B 8F A3 A3 7B 8F 95 85 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 55 55 21 87 15 59 05 24 12 65 00 00 00 00 00 00 00 
data: 00 C2 D3 0B 34 65 B2 D3 0B 95 65 AB 00 00 83 A5 AB 32 32 B0 A3 32 32 32 B0 B0 AB CB 96 A6 7D 61 
data: 7E 37 9A 8D 01 3A 45 00 00 1F 18 4A 93 49 01 41 79 01 01 2E 00 00 15 00 00 01 00 00 01 00 00 08 
data: 01 0E 32 01 01 01 01 02 00 00 F5 
Checksum and header received ok!
received TOP1 Pump_Flow: 24.12
received TOP6 Main_Outlet_Temp: 43
received TOP8 Compressor_Freq: 68`

MiG-41, 2021-12-11:
If you have monoblock , and using 25cm cable i assume that HeishaMon is outside inside the metal cover of HP... What is your wifi signal strength? And if it is not above 65 try to improve it.

kr0815, 2021-12-11:
The Heishamon PCB is inside the House - but the Data Cable runs trough the same hole as the Power Cable

If i look at the "wrong" data - without understanding which Byte is what - it´s always the same - so i think it can´t be a distortion from the power cable?

MiG-41, 2021-12-11:
Ok , 25cm is the distance from power cable , not th length it self... 

You can try modbus in the middle connection as here:
https://github.com/IgorYbema/HeishaMon

Should be more resistant for intereferences on longer distances.

kr0815, 2021-12-11:
no, sorry, the cable is about 2,5 meters long
Heishamon PCB sits inside the house
Only where the cable goes trough the wall it runs in the same hole as the power cable

Even more strange - now i get fancy Power / S0 Readings - up to 40 KWh which is impossible at a 16A Fuse

i think something is wrong with the PCB...

kr0815, 2021-12-12:
in the end i build a new PCB- works like expected now

IgorYbema, 2022-01-02:
> in the end i build a new PCB- works like expected now

Can you contact me on the order confirm email I have sent you from the Tindie shop?

eigthyeight, 2022-01-30:
Instead of opening a new issue, I have the same issue.
Installed my Heishamon today with a 5m cable (from the tindy shop) (it is now in an enclosure next to the HP so 2m would also have been enough)

I get a Correct received data rate from 1 - 15%....
wifi is in the range of 100-66%, does not seem to make a difference

updated fw to 'HeishaMon.ino.d1-v2.2-iy-1' but no effect

data looks like

```
data: B4 
Received 116 bytes data
data: 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 
data: 99 A3 7B 8F A3 A3 7B 8F 98 85 7C 8F 8A 94 9E 8F 8A 94 9E 85 8F 8A 11 3D 78 C1 0B 7E 7C 0B 7B 7E 
data: 00 00 00 55 55 55 21 87 15 55 05 14 12 65 00 00 00 00 00 00 00 00 C2 D3 0B 35 65 B2 D3 0B 96 65 
data: 91 00 00 87 91 91 32 32 99 A3 32 32 32 99 99 91 88 88 8A 85 
Checksum received false!
Requesting new 1wire temperatures

```

any ideas to a solution?

IgorYbema, 2022-01-30:
Can you test with this development binary? https://github.com/IgorYbema/HeishaMon/actions/runs/1730932765 see artifacts
The bad data was calculated wrong. 1% was actually like 10-20%. 

Do you use s0 kWh meter? I had some customers with a specific kWh meter which caused a lot of noise on the GND of the heishamon causing these errors. They connected the negative side (GND side on S0 heishamon connector) to the metal chassis of the heatpump creating extra grounding. That helped a lot with these issue's.

If this all doesn't help, contact me back using the tindie contact.

geduxas, 2022-01-30:
> Do you use s0 kWh meter? I had some customers with a specific kWh meter which caused a lot of noise on the GND of the heishamon causing these errors. They connected the negative side (GND side on S0 [...]

Hmm.. could you explain More about that? I still can't read S0 from my meater. If you remember me, i have successfully read S0 only while i have attached oscilloscope on heishamon contact's.. maybe it's same grounding problem's? Do i need to ground negative contact on meter side?

IgorYbema, 2022-01-30:
Yes that is what I was told by these customers. But probably grounding the GND of the S0 contact (same cable) should work also. 
I believe that somehow there is a ground loop causing too much noise on the S0 connector.

But there are also some other kWh meters with different problems. It seems that not all kWh meters use same S0 logic and it is hard to make hardware for it that matches all brands.

mitchellengelage, 2022-03-29:
Having the same issue, 0-1% Correct data. Ran an extra wire from GND tot HP chassis, same result. Disconnected s0 meters altogether, ~30% correct data, I have  data in Home Assistant now, but is there any solution to use the s0 meters?  COP of 6 reported from the Pana is nice, but I would like some actual data ;) 
Used meters: DDS353

IgorYbema, 2022-03-29:
@mitchellengelage i'll contact you directly to check if you might have a bad pcb

mitchellengelage, 2022-04-05:
PCB has been fixed and I ran a wire from Gnd on s0 meter to actual earth point within the HP, which gave me ~80% correct data 👍

DL4JY, 2022-11-17:
Hello, 
I had a few 10.000 errors with bad headers per day. iI had the optional board emulation on 
However after all I took my oscilloscope I found the 5v from the Pana was very bad. Adding 2 220uf caps from both ends of the diode to ground plus a 2.2k pullup to 5v to RX solved the problems.

Mabe its worth looking a bit into voltage filtering for the next version .

cheerio

IgorYbema, 2022-11-17:
Which pcb version do you have? newer versions already have some improvement for this

DL4JY, 2022-11-18:
hello Igor,
it seems I have V1.0
[image]

please take into consderation that this was the first shot. And it worked. no reason to make it nice.

naseweiss01, 2023-01-18:
Hi,
with the above changes, I had still occasionally CRC & Bad header errors again. But very few.
I increased the value of the Capacitor after the diode to 2200uF.

Pleas note my wonderful artwork :-)

[image]

[image]

with that i have using the optional board emulation, 10m shielded CAT 6 Kabel & 30s MQTT interval no issues anymore. For days.

MiG-41, 2023-01-19:
@naseweiss01 ,in my opinion you have problems on the connection to HP, it is not solid enough (like pins not tight enough) and you red it as a bad power supply. After you unplug and plug again for somtime it is ok ,and you read it as a succesful modification...

naseweiss01, 2023-01-20:
> @naseweiss01 ,in my opinion you have problems on the connection to HP, it is not solid enough (like pins not tight enough) and you red it as a bad power supply. After you unplug and plug again for somtime it is ok ,and you read it as a succesful modification...

I can assure you, I analyzed the issue in depth. I used a digital Storage Oscilloscope at both ends.
To that comes that a friend has the same issue at H-Tcap,, too. 
I am not talking about a bad power supply, its about filtering,,support Capacitors  and data-line termination / pullup resistors . 
And I am admitting, I am at the limit of cable length without RS485 transmitters.
However, my unit also creates errors when directly mounted within the heatpump cabinet with a different cable - and I see the glitches on the  supply voltage. I am having a J-T-Cap.

To make it clear - I like the Heishamon, this is to make it better.

IgorYbema, 2023-01-20:
The pullup of the RX (TX from heishamon) is ok, but should not be needed as this is already on the panasonic side. However if it helps, it is ok to do.
The extra caps will help always to make a more stable power supply and if it helps for you, good find! In most cases it is not necessary but it sure can help for some situations. I see you have the opentherm v1.0 which had some issue's with tx/rx and maybe you solution helps in this case.

closed 2024-09-12 by Egyras