heishamon/HeishaMon #239: Many automations

Erikvers, 2021-12-23:
Is it nirmal that there are so many logs in home assistant? And can i make it less or thit it not harm my Pansonic?

is ser this kind of thinks 5 or 6 off it every 10 min:

Set heatshift selector has been triggered by mqtt topic panasonic_heat_pump/main/Z1_Heat_Request_Temp
23:25:25 - 9 minuten geleden

MiG-41, 2021-12-24:
So either you have changed in Heishamon  setting"How often all heatpump values are retransmitted to MQTT broker" ( normally will give you 2 triggers per 10 min , since it is set to 300s), or there is in automation aditional rule/thing , with update that topic more frequently.

Erikvers, 2021-12-24:
Treu it hapends every 5 min. 

But it can not harm - send to the other way? It only is sending from heatpump to home assistant? I read about the warning no to trigger the heatpump to much, that is why i aak this

MiG-41, 2021-12-24:
Yes, it is only from HP to HA via mqtt.
To check how often  setcommand is sending to Heishamon you have to check/log topic from /commands/ diectory in mqtt.

geduxas, 2021-12-24:
It's possible to go out from automations.. just need to write custom component :) i am now playing with mqtt thermostat and it's work's better than automations..

pasarn, 2021-12-24:
> It's possible to go out from automations.. just need to write custom component :) i am now playing with mqtt thermostat and it's work's better than automations.. 

Hi, can you share your mqtt thermostat config. I also test this method and can share my config tumorow.

pasarn, 2021-12-24:
- platform: mqtt
  name: Heizung
  modes:
    - "heat"
    - "off"
  temperature_state_topic: "panasonic_heat_pump1/main/Main_Target_Temp"
  current_temperature_topic: "panasonic_heat_pump1/main/Main_Outlet_Temp"
  away_mode_state_topic: "panasonic_heat_pump1/main/Holiday_Mode_State"
  away_mode_state_template : >-
    {%- if value == "0" -%}
      OFF
    {%- elif value == "1" -%}
      ON
    {%- endif -%}
  mode_state_topic: "panasonic_heat_pump1/main/Operating_Mode_State"
  mode_state_template: >-
    {%- if value == "0" -%}
      heat
    {%- elif value == "1" -%}
      cool
    {%- elif value == "2" -%}
      auto
    {%- elif value == "3" -%}
      heat
    {%- elif value == "4" -%}
      heat
    {%- elif value == "5" -%}
      cool
    {%- elif value == "6" -%}
      auto
    {%- endif -%}
  aux_state_topic: "panasonic_heat_pump1/main/Internal_Heater_State"
  aux_state_template: >-
    {%- if value == "0" -%}
      OFF
    {%- elif value == "1" -%}
      ON
    {%- endif -%}
  action_topic: "panasonic_heat_pump1/main/Compressor_Current"
  action_template: >-
    {%- if ( value | float ) > 0.0 -%}
      heating
    {%- elif ( value | float ) == 0.0 -%}
      idle
    {%- endif -%}
  value_template: "{{ value }}"

geduxas, 2021-12-24:
> * platform: mqtt
>   name: Heizung
>   modes:
>
>   * "heat"
>   * "off"
>     temperature_state_topic: "panasonic_heat_pump1/main/Main_Target_Temp"
>     current_temperature_topic: [...]

Wery Nice, it could be extended and commited in curent repository, i am on same way,  just couple modifications. Determinate heating mode and change heatin temperature vs curve setpoint. And show current mode heating/idle  (idle during DHW and defrost) what you think?

closed 2021-12-28 by Erikvers