heishamon/HeishaMon #241: Continuous change loop selector Slider

Erikvers, 2021-12-28:
Is it possible to put a 1 or 2 second delay here? Sometimes when I change the selector in Home Assistant from Heatshift or Tank Temp, it sometimes keeps oscillating back and forth...

**Think it would be best to pass the input from mqtt only if it is more than 1 sec?
Can you help me on the best way to adjust that?**

` #Fetches heatshift temperature from status and sets the selector accordingly
  - alias: Set heatshift selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Z1_Heat_Request_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_heatshift
        value: >-
          {{ "%.1f" % (trigger.payload | int) }}
  #Sets heatshift temperature to the selected value
  - alias: Set heatshift
    trigger:
      platform: state
      entity_id: input_number.heishamon_heatshift
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetZ1HeatRequestTemperature
        retain: false #IMPORTANT! ALWAYS set retain flag false for commands!
        payload_template: >-
          {{ "%.0f" % (states('input_number.heishamon_heatshift') | int) }}
  #Fetches tank target temperature from status and sets the selector accordingly
  - alias: Set tank target temperature selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/DHW_Target_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_tank_temp
        value: >-
          {{ "%.1f" % (trigger.payload | int) }}
  #Sets tank target temperature to the selected value
  - alias: Set tank target temperature
    trigger:
      platform: state
      entity_id: input_number.heishamon_tank_temp
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetDHWTemp
        retain: false #IMPORTANT! ALWAYS set retain flag false for commands!
        payload_template: >-
          {{ "%.0f" % (states('input_number.heishamon_tank_temp') | int) }}
`

MiG-41, 2021-12-28:
Move this part to automations.yaml , there it is possible to add delay.

Erikvers, 2021-12-28:
Thanks, and can you gif me a suggestion what i have to change? I am a Rookie :-)

miklomu, 2022-03-14:
I have similar issue, sometimes selectors are looping and only way to stop is to disable automation and then enable it again. Solution for this would be more than welcome.

closed 2024-09-12 by Egyras