heishamon/HeishaMon #247: Can't send command topics with IObroker

timaios2000, 2022-03-15:
The command topics don't appear when Heishamon connects to IObroker.

Do I need to set up the command topics manually? Should there be a folder named "commands"?

IgorYbema, 2022-03-15:
You need to set it up indeed. Commands are not received from heishamon but only too so that is why your mqtt broker doesn't have a 'folder' for it

timaios2000, 2022-03-15:
Thanks for your reply. Could you please give me a hint, how to set this up. Do I need a second instance of the mqtt broker, which sends commands to Heishamon?

IgorYbema, 2022-03-15:
A mqtt broker is just a message bus. Heishamon puts messages to it on the topics you see. From another client you post messages 'commands' to the broker. Heishamon listens to these commands and then performs the actions.
So your system which you are using to send commands just needs to post them to the broker. These are the commands: https://github.com/IgorYbema/HeishaMon/blob/main/MQTT-Topics.md#command-topics

closed 2024-09-12 by Egyras