heishamon/HeishaMon #248: Received a shorter datagram. Can't decode this yet.

draggOP, 2022-04-12:
I just connected a panasonic WH-MDC12H6E5 to a wemosd1 mini + a diy pcb following this design: https://easyeda.com/igor_6537/panasonic-cn-cnt-to-wemos
i'm using external power (not from the heatpump) and i'm on version 2.0
if it matters i'm using 2 conductors from a cat5 cable for TR and RX

I get no data in the dashboard and in the console i get this message over and over

Lost MQTT connection!
Heishamon stats: Uptime: 0 days 0 hours 1 minute 50 seconds ## Free memory: 78% 28176 bytes ## Wifi: 56% ## Mqtt reconnects: 0 ## Correct data: 100.00%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received 111 bytes data
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 
Checksum and header received ok!
Received a shorter datagram. Can't decode this yet.

Am i doing something wrong or is my heatpump not supported?

IgorYbema, 2022-04-12:
Somehow you are receiving the request datagram back. 
You only have the heishamon pcb connected or also the CZ-TAW1 official internet module?
Your heatpump should be supported.

draggOP, 2022-04-12:
I only have the heishamon pcb connected.

IgorYbema, 2022-04-12:
Then check for bad wiring. It looks like your TX and RX is shorted.

draggOP, 2022-04-12:
With TX and RX disconnected form the heishamon pcb i get a timeout error instead, so i think my pcb is not the issue.
There is no continuity either between the TX and RX wires coming from the heatpump.
I am usin a twisted pair from a cat5 cable as TX and RX is that ok?
I can also try to wire the heishamon pcb to the alternative header on the heatpump but that shouldn't make a difference since they are hardwired right?

IgorYbema, 2022-04-12:
Alternative header isn't necessary indeed.
Only thing I can think of now could be the twisted pair which is causing crosstalk somehow. Maybe only use one conductor only.
The answer should start with 71 c8 (c8 stands for the length to expect, 203 bytes).

draggOP, 2022-04-16:
It was the cable's fault, i am now using a shorter (~2m) taking ground and 5v from the heatpump, tx and rx are no longer on a twisted pair and everything is working flawlessly.
Thanks alot for the help

closed 2022-04-16 by draggOP