heishamon/HeishaMon #250: heishamon.yaml updated

edterbak, 2022-06-10:
HomeAssistant changed the designation of mqtt sensors and switches. 
This impacted **heishamon.yaml.** 
I have updated it already, and it is ready to use. working with HomeAssistant from release month 6-2022

text from HA release notes:

As an example, this is now deprecated:
```
sensor:
  - platform: "mqtt"
    name: "My sensor"
    state_topic: "some-state-topic"
```
The configuration needs to be updated to this format:
```
mqtt:
  sensor:
    - name: "My sensor"
      state_topic: "some-state-topic"
```

[heishamon.yaml.txt](https://github.com/Egyras/HeishaMon/files/8882679/heishamon.yaml.txt)

edterbak, 2022-06-10:
Note that you need to change the extension of the file. 
Github does not allow direct yaml.
File name should be: heishamon.yaml

edterbak, 2022-06-12:
Added a new version which corrects some bits.

edterbak, 2022-06-14:
> > Added a new version which corrects some bits.
> > it could do with some checking by someone :)
> 
> Nice work! Why not also migrate the legacy template sensors to their new format? I've added an (untested) example below.
> 
> ```
> template:
>   - sensor:
>       - name: "heishamon_w_production"
>         unit_of_measurement: 'W'
>         state: >-
>           {%- if states('sensor.aquarea_dhw_power_produced') != "0" -%}
>             {{ states('sensor.aquarea_dhw_power_produced') }}
>           {%- else -%}
>             {{ states('sensor.aquarea_power_produced') }}
>           {%- endif -%}
>       - name: "heishamon_w_consumption"
>         unit_of_measurement: 'W'
>         state: >-
>           {%- if states('sensor.aquarea_dhw_power_consumed') != "0" -%}
>             {{ states('sensor.aquarea_dhw_power_consumed') }}
>           {%- else -%}
>             {{ states('sensor.aquarea_power_consumed') }}
>           {%- endif -%}
>        - name: "heishamon_cop"
>          unit_of_measurement: "x"
>          state: >-
>           {%- if states('sensor.heishamon_w_production') != "Unknown" -%}
>             {%- if states('sensor.heishamon_w_consumption') > "0" -%}
>               {{ '%0.1f' % ((states('sensor.heishamon_w_production') | float ) / (states('sensor.heishamon_w_consumption') | float)) }}
>             {%- else -%}
>               0.0
>             {%- endif -%}
>           {%- else -%}
>             n/a
>           {%- endif -%}
> ```

Hi,
Yeah, I did not see this yet.
I have no experience handling the template sensors. Could you maybe have a go at it?
I will replace the one in my post with the updated one from you then *(with mentions to you)

:)

edterbak, 2022-06-14:
I gave it an attempt, 
Looking at your code, it is fairly similar. I have removed the - in the %-   -% codes, that is the only difference I can spot.

See attached file here.
[heishamon.yaml.txt](https://github.com/Egyras/HeishaMon/files/8904126/heishamon.yaml.txt)

I do see there is a potential issue that the value can be unknown, throwing an error. 
Does anyone know how to tackle that? 

thanks

MiG-41, 2022-06-15:
Beouse you dont have any sensor.heishamon_w_production and sensor.heishamon_w_production ? 
You should have something like sensor.aquarea_power_produced and sensor.aquarea_power_produced ( acc to #TOP15 and #TOP16) . This entities names can be checked i.e. in development menu.

edterbak, 2022-06-15:
This is a hint. I understand the issue, but I dont have experience fixing it. I am missing some good examples.

`ValueError: could not convert string to float: 'unknown'`

kampsj, 2022-06-15:
I guess I forgot that I have renamed the sensors in my personal yaml.

I will do some testing this weekend (and rename my sensors).

boutermw, 2022-06-18:
@edterbak I've updated with your heishamon.yaml on latest Home Assistant v 2022.6.6 and all working fine, thanks!

edterbak, 2022-06-19:
> @edterbak I've updated with your heishamon.yaml on latest Home Assistant v 2022.6.6 and all working fine, thanks!

Can  you check on the s0 meter if that is all ok?
I got feedback from one other that some naming was changes. 
I am having difficulty to understand how this change happend. Better to double check if there is an error here or not :)

kampsj, 2022-06-19:
@edterbak  

Your naming of the s0 meter is off.
`- name: S0 meter - Solar power Production` and `- name: S0 meter - Solar power Production Total`
versus 
`name: Aquarea Metered Power Consumption` and `Aquarea Metered Power Consumption Total`
for S0_1. Naming of S0_2 is also not identical, which is probably the cause.

edterbak, 2022-06-19:
> @edterbak
>
> Your naming of the s0 meter is off. `- name: S0 meter - Solar power Production` and `- name: S0 meter - Solar power Production Total` versus `name: Aquarea Metered Power Consumption` and [...]

Is this the only thing 'off'? If so, I can correct it. But if there is more, (and I still dont understand how this happend) I will throw it away and start over to be sure there are no other mistakes.

kampsj, 2022-06-19:
@edterbak I've checked your latest file against the latest version (https://github.com/Egyras/HeishaMon/blob/master/Integrations/Home%20Assistant/heishamon.yaml) and have made a few changes to sync both files.

Note that I've opted to keep the legacy templates as is.  
[heishamon.yaml.txt](https://github.com/Egyras/HeishaMon/files/8936178/heishamon.yaml.txt)

boutermw, 2022-06-21:
The naming was off, like kampsj stated. I mentioned this also on GoT. Now all working fine.

edterbak, 2022-06-23:
@kampsj  When I import  your latest .txt, I get errors.
I started over, with the choose to use of legacy templates now. Maybe later the newer version.

I think there is an indentation difference.

```
mqtt:
  sensor:
    - name: ....
    - name
    - name
....
....
  sensor:
    - platform: template
      sensors:
        heishamon_w_production:
```
The lower part with platform template is i
should be:

```
mqtt:
  sensor:
    - name: "xxx"
    - name: "yyy"
    - name: "zzz"
....
....
sensor:
  - platform: template
    sensors:
      heishamon_w_production:
```

The below version works for me, even after reboot. 
I dont know how you get your version working  😄 
[heishamon - 23-6-2022.yaml.txt](https://github.com/Egyras/HeishaMon/files/8968784/heishamon.-.23-6-2022.yaml.txt)

am I missing something or am I just ignorant ghehehee.. 

Hopefully I am actually helping here, instead of making a mess.

kampsj, 2022-06-23:
@edterbak You're right. There were a few (3) errors in my file. I've fixed them in the file below.

[heishamon.yaml.txt](https://github.com/Egyras/HeishaMon/files/8970127/heishamon.yaml.txt)

edterbak, 2022-06-24:
I confirm.
Our latest versions are now both identical. (give or take some extra lines in between) 
Both work.

But both are in old template format, which I dont bother with.

edterbak, 2022-06-24:
https://github.com/Egyras/HeishaMon/blob/c595ec074d8e9d8640ec10dc6a8535b1e31be226/Integrations/Home%20Assistant/heishamon.yaml

I see our work is outdated. 
pullrequest with updated heishamon.yaml already in there. Hahaha...

closed 2024-09-12 by Egyras