heishamon/HeishaMon #252: HeishaMon 2.0 losing WiFi connection regularly

ogbrugge, 2022-06-24:
My HeishaMon running 2.0 is regularly losing connection to my WiFi network, roughly two to three times per week, I have a Unify WiFi access point very close to the HeishaMon so it cannot be the signal strength being too weak.  Is there any way to debug the reason why HeishaMon is not realizing the connection is gone? Sometimes it spontaneously reboots, i can see that by the up time, is this to reconnect to the network?

IgorYbema, 2022-06-24:
Could you try this test build for v3.0 which also adds some wifi improvements?https://github.com/IgorYbema/HeishaMon/suites/6553213424/artifacts/244882468

ogbrugge, 2022-06-24:
I installed it and will let you know if this is more stable. Thank you for the quick response!

markuspulch, 2022-07-14:
Hi, I have the same issue but it looks the the build is not available anymore? ;-(

HighlyCompressedAir, 2024-09-11:
Fixed in #260 .
This issue can be closed.

closed 2024-09-12 by Egyras