heishamon/HeishaMon #256: Just a little typo for answer of "SetDHWHeatDelta"

Bogeyof, 2022-07-11:
If I set "SetDHWHeatDelta" the answer is "set floor heat delta to.."

Code
`unsigned int set_dhw_heat_delta(char *msg, unsigned char *cmd, char *log_msg) {
  unsigned int len = 0;
  String set_temperature_string(msg);

  byte request_temp = set_temperature_string.toInt() + 128;

  {
    char tmp[256] = { 0 };
    snprintf_P(tmp, 255, PSTR("set floor heat delta to %d"), request_temp - 128 );
    memcpy(log_msg, tmp, sizeof(tmp));
  }

  {
    memcpy_P(cmd, panasonicSendQuery, sizeof(panasonicSendQuery));
    cmd[99] = request_temp;
  }

  return sizeof(panasonicSendQuery);
}`

MiG-41, 2022-07-11:
Yes , in H it is the same range : from -12 to -2 deg....

Bogeyof, 2022-07-12:
Answer-Text is corrected in v3.0, may be also in v2.2. Both are a problem for me, cause JSON-Output ist different and I think it´s a bit messed up...

IgorYbema, 2022-07-12:
As noticed also in https://github.com/Egyras/HeishaMon/issues/257 I am fixing the json output now

closed 2024-09-12 by Egyras