heishamon/HeishaMon #257: REST API cant set 2 commands at one time

Bogeyof, 2022-07-12:
In Documentation, there is written the following:
`HTTP REST API: http://x.x.x.x/command?[topic]=[value]&[topic]=[value] (e.g.: http://x.x.x.x/command?SetQuietMode=3&SetZ1HeatRequestTemperature=21_`

If I try this like "http://192.168.xxx.xxx/command?SetDHWTemp=49&SetQuietMode=3", I get to answers from Webinterface:
`set DHW temperature to 49
set Quiet mode to 3`

But only the first parameter would be set after this command, the second would never change...

Bogeyof, 2022-07-12:
I will try it and report later...

Bogeyof, 2022-07-12:
I tried Firmware v3.0, still the same behavior, second param would not be set. 
Can´t use Firmware v3.0, cause Json-Output is messed up. Many Params are doubled, for example TOP12 "Operations_counter" and many more. So my reading routine failed...
Where can I get v2.0a for Download again. Could only find v2.0 or v2.1 and greater?

IgorYbema, 2022-07-12:
2.0a is only a temp fix for correcting the 'error' counter. You can use v2.0 for now.
Thanks for the feedback about the json output. Didn't see that before. I'll fix that right away

IgorYbema, 2022-07-12:
Also, while testing your report it seems to work for me:
panasonic_heat_pump/log Tue Jul 12 16:03:55 2022 (132049): set DHW temperature to 49
panasonic_heat_pump/log Tue Jul 12 16:03:55 2022 (132050): Already sending data. Buffering this send request
panasonic_heat_pump/log Tue Jul 12 16:03:55 2022 (132055): set Quiet mode to 1
panasonic_heat_pump/log Tue Jul 12 16:03:55 2022 (132060): Already sending data. Buffering this send request
...
panasonic_heat_pump/log Tue Jul 12 16:03:55 2022 (132199): Sending command from buffer
...
panasonic_heat_pump/log Tue Jul 12 16:03:56 2022 (132697): Sending command from buffer
...
panasonic_heat_pump/log Tue Jul 12 16:03:56 2022 (133188): received TOP9 DHW_Target_Temp: 49
...
panasonic_heat_pump/log Tue Jul 12 16:03:57 2022 (133688): received TOP18 Quiet_Mode_Level: 1

This is with v3.0

IgorYbema, 2022-07-12:
No, MQTT should be seperated from that issue. But yes the log I created is from the mqtt post.
However same log is also on the web interface - console tab. Can you check there?

Bogeyof, 2022-07-12:
Sorry, for me on the console tab there is nothing, only a black area. tried to enable hexdump log or MQTT log, but always only black area. MQTT log isn´t shown, may because if have no MQTT? Hexdump log also no output on console tab...

IgorYbema, 2022-07-12:
No that is independent. The console log should be visible. Weird. Maybe browser issue? Which browser are you using?
It is a websocket connection from the browser to heishamon which is displaying the logs.

Bogeyof, 2022-07-12:
Tested a minute ago with Chrome, Edge and Firefox, black area, no data...

Bogeyof, 2022-07-12:
Heishamon hangs in the basement, another WLAN is active there and routed via a port so that I can access the web interface. Possibly the problem is to be found here...

IgorYbema, 2022-07-12:
websocket it on same port as http (port 80) but if you are forwarding some other port (like 8080 or something) that could be the reason indeed

IgorYbema, 2022-07-12:
No I am more looking at this
Tue Jul 12 17:13:49 2022 (73746): Previous read data attempt failed due to timeout
Seems some reading issue. But need more log after that.

Bogeyof, 2022-07-12:
This time no param was changed:

Tue Jul 12 17:39:20 2022 (977090): Closing webserver client: 192.168.77.20:62884
Tue Jul 12 17:39:41 2022 (998083): New webserver client: 192.168.16.113:45424
Tue Jul 12 17:39:41 2022 (998107): set DHW temperature to 47
Tue Jul 12 17:39:41 2022 (998108): sent bytes: 111 including checksum value: 227 
Tue Jul 12 17:39:41 2022 (998110): set Quiet mode to 3
Tue Jul 12 17:39:41 2022 (998114): Already sending data. Buffering this send request
Tue Jul 12 17:39:41 2022 (998135): Received 203 bytes data
Tue Jul 12 17:39:41 2022 (998135): Checksum and header received ok!
Tue Jul 12 17:39:41 2022 (998145): received TOP55 Ipm_Temp: 26
Tue Jul 12 17:39:41 2022 (998154): Sending command from buffer
Tue Jul 12 17:39:41 2022 (998216): sent bytes: 111 including checksum value: 114 
Tue Jul 12 17:39:41 2022 (998233): Closing webserver client: 192.168.16.113:45424
Tue Jul 12 17:39:41 2022 (998680): New webserver client: 192.168.16.113:45422
Tue Jul 12 17:39:43 2022 (1000219): Previous read data attempt failed due to timeout!
Tue Jul 12 17:39:43 2022 (1000219): Received 0 bytes data
Tue Jul 12 17:39:50 2022 (1007127): New webserver client: 192.168.77.20:62900
Tue Jul 12 17:39:51 2022 (1008607): Closing webserver client: 192.168.77.20:62900
Tue Jul 12 17:40:03 2022 (1020029): Lost MQTT connection!
Tue Jul 12 17:40:03 2022 (1020029): Reconnecting to mqtt server ...
Tue Jul 12 17:40:03 2022 (1020032): Heishamon stats: Uptime: 0 days 0 hours 17 minutes 0 seconds ## Free memory: 74% ## Heap fragmentation: 10% ## Max free block: 26944 bytes ## Free heap: 29784 bytes ## Wifi: 56% (RSSI: -72) ## Mqtt reconnects: 0 ## Correct data: 90.91%
Tue Jul 12 17:40:03 2022 (1020055): Requesting new panasonic data
Tue Jul 12 17:40:03 2022 (1020059): sent bytes: 111 including checksum value: 18 
Tue Jul 12 17:40:03 2022 (1020490): Received 203 bytes data
Tue Jul 12 17:40:03 2022 (1020490): Checksum and header received ok!
Tue Jul 12 17:40:11 2022 (1028681): Timeout webserver client: 192.168.16.113:45422
Tue Jul 12 17:40:11 2022 (1028682): Closing webserver client: 192.168.16.113:45422
Tue Jul 12 17:40:12 2022 (1029659): New webserver client: 192.168.77.223:49702
Tue Jul 12 17:40:14 2022 (1031403): Closing webserver client: 192.168.77.223:49702
Tue Jul 12 17:40:21 2022 (1038643): New webserver client: 192.168.77.20:62917
Tue Jul 12 17:40:23 2022 (1040157): Closing webserver client: 192.168.77.20:62917

Bogeyof, 2022-07-12:
> Thanks. Are you sure the DHW=47 and quiet=3 where new values (they are different than the current values at the time)? If values are the same they are not displayed as changed.

Yes, value before was 48 and quiet mode 1.

Now I tried again with Computer off (it was also connected to heishamon, may be the cause for closing connectings very fast), tried again with Smartphone in basement, this time values are both changed. Values before was 48 and 0. Sending 2 combinations, first 48 / 1 and  second 47 / 1.  Values after second sending are 47 / 1:

Tue Jul 12 17:51:16 2022 (1693913): New webserver client: 192.168.16.113:45760
Tue Jul 12 17:51:17 2022 (1694337): Closing webserver client: 192.168.16.113:45760
Tue Jul 12 17:51:18 2022 (1695122): Closing webserver client: (IP unset):0
Tue Jul 12 17:51:27 2022 (1704070): New webserver client: 192.168.16.113:45766
Tue Jul 12 17:51:27 2022 (1704079): set DHW temperature to 48
Tue Jul 12 17:51:27 2022 (1704080): sent bytes: 111 including checksum value: 226 
Tue Jul 12 17:51:27 2022 (1704082): set Quiet mode to 1
Tue Jul 12 17:51:27 2022 (1704086): Already sending data. Buffering this send request
Tue Jul 12 17:51:28 2022 (1705857): Received 203 bytes data
Tue Jul 12 17:51:28 2022 (1705857): Checksum and header received ok!
Tue Jul 12 17:51:28 2022 (1705868): received TOP64 High_Pressure: 14.4
Tue Jul 12 17:51:28 2022 (1705879): Sending command from buffer
Tue Jul 12 17:51:28 2022 (1705879): sent bytes: 111 including checksum value: 130 
Tue Jul 12 17:51:28 2022 (1705910): Closing webserver client: 192.168.16.113:45766
Tue Jul 12 17:51:29 2022 (1706334): Received 203 bytes data
Tue Jul 12 17:51:29 2022 (1706334): Checksum and header received ok!
Tue Jul 12 17:51:48 2022 (1725066): New webserver client: 192.168.16.113:45772
Tue Jul 12 17:51:48 2022 (1725076): set DHW temperature to 47
Tue Jul 12 17:51:48 2022 (1725077): sent bytes: 111 including checksum value: 227 
Tue Jul 12 17:51:48 2022 (1725079): set Quiet mode to 2
Tue Jul 12 17:51:48 2022 (1725083): Already sending data. Buffering this send request
Tue Jul 12 17:51:48 2022 (1725123): Closing webserver client: 192.168.16.113:45772
Tue Jul 12 17:51:48 2022 (1725556): Received 203 bytes data
Tue Jul 12 17:51:48 2022 (1725556): Checksum and header received ok!
Tue Jul 12 17:51:48 2022 (1725561): received TOP18 Quiet_Mode_Level: 1
Tue Jul 12 17:51:48 2022 (1725577): Sending command from buffer
Tue Jul 12 17:51:48 2022 (1725578): sent bytes: 111 including checksum value: 122 
Tue Jul 12 17:51:50 2022 (1727580): Previous read data attempt failed due to timeout!
Tue Jul 12 17:51:50 2022 (1727581): Received 0 bytes data
Tue Jul 12 17:51:51 2022 (1728757): Lost MQTT connection!
Tue Jul 12 17:51:51 2022 (1728757): Reconnecting to mqtt server ...
Tue Jul 12 17:51:51 2022 (1728760): Heishamon stats: Uptime: 0 days 0 hours 28 minutes 48 seconds ## Free memory: 37% ## Heap fragmentation: 13% ## Max free block: 12984 bytes ## Free heap: 14824 bytes ## Wifi: 58% (RSSI: -71) ## Mqtt reconnects: 0 ## Correct data: 87.76%
Tue Jul 12 17:51:51 2022 (1728783): Requesting new panasonic data
Tue Jul 12 17:51:51 2022 (1728787): sent bytes: 111 including checksum value: 18 
Tue Jul 12 17:51:52 2022 (1729263): Received 203 bytes data
Tue Jul 12 17:51:52 2022 (1729264): Checksum and header received ok!
Tue Jul 12 17:51:52 2022 (1729267): received TOP9 DHW_Target_Temp: 47
Tue Jul 12 17:51:52 2022 (1729279): received TOP64 High_Pressure: 14.6

IgorYbema, 2022-07-12:
Hmm I notice that except to the set command there are no regular checks for the new values. Normally each 5 secs it will request new values (settings: How often new values are collected from heatpump - default 5 sec)). Did you change that maybe?

After a set command the next read will not display the change directly. It needs another request to see the changed value. And that is why you only see it after the next change.  And therefore there is a normal 5 sec refresh of the data like this:
Tue Jul 12 18:19:44 2022 (8280534): Requesting new panasonic data

closed 2024-09-12 by Egyras