heishamon/HeishaMon #264: Cannot import Json in Node Red

MikeyMan83, 2022-08-25:


IgorYbema, 2022-08-25:
You need to install the influx components into nodered also

MikeyMan83, 2022-08-25:
I did, although readme says its optional:

```
Prerequisite:

Domoticz with mqtt setup
node-red with mosquitto (optionally with https://flows.nodered.org/node/node-red-contrib-influxdb installed)
Working PCB with Heishamon firmware installed
```
Followed the link and executed the following command:
`npm install node-red-contrib-influxdb`

also added panasonic db:

[image]

IgorYbema, 2022-08-25:
Do you see the influx components in the left menu bar in nodered? Maybe they changed name a bit with newer versions

MikeyMan83, 2022-08-25:
i see these:

[image]

IgorYbema, 2022-08-25:
Not sure what is wrong. Think something with new version. Can you check in nodered forums?

MikeyMan83, 2022-08-25:
> Not sure what is wrong. Think something with new version. Can you check in nodered

Really dont know what to ask for, havent found any examples of this specific error tbh

MikeyMan83, 2022-08-25:
Did get this fixed through help on nodered forum.

Something was wrong with the install. Reinstalled without sudo. 

now getting another error though...

[image]

IgorYbema, 2022-08-25:
This flow is really old. Needs some updates but I don't remember who created it :)

MikeyMan83, 2022-08-25:
Oh ok... that doesn't help...

Its an issue with the json it seems, although similar flows give the same error:

[image]

MikeyMan83, 2022-08-25:
@IgorYbema any way you could have a look maybe? I'm not good with code :(

closed 2024-09-12 by Egyras